mysensors/MySensors #577: Softreboot loop with ATmega328P 8Mhz without bootloader

RalfJL, 2016-08-31:
Hi,

I am experimenting with a plain Atmega328p on 8Mhz as a mysensor and I found that when I send a I_REBOOT message the ATmega goes into a loop.

The reason is that the I_REBOOT is done via watchdog reset with a hwReboot with15ms timeout. This is simply not enough time to restart the ATmega and run enough mySensor code to set the watchdog again.

Can you please rewrite the macro to:
#define hwReboot() wdt_enable(WDTO_1S); while (1)

I know that 1 second is a mighty long time for an android, but a human can easily wait 1 second for the node to reboot.

1 second should also be enough time for a 1Mhz ATmega to do a reboot. The reason for using a 1Mhz ATmega is that it can go down to 1.8 V Vcc so running on 2 AA batteries it will make better use of the batteries. 8Mhz requires at least 2.7 V Vcc (according to datasheet)

Thanks

Ralf

tekka007, 2016-08-31:
@RalfJL Are you running optiboot or the stock (and buggy) bootloader?

RalfJL, 2016-09-02:
Well,

I know that the stock arduino bootloader is not working. I_REBOOT goes into a bootloop, a watchdog reboot (e.g. provoked with an infinite loop) goes into a bootloop as well. broken 

When I use a bootloader I use optiboot. That works always (I_REBOOT and real watchdog)

But what I am trying is a very minimal setup. So beside the ATmega328 an RFM69 and an OneWire temp sensor there should be nothing. No crystal, no voltage regulator, no nothing.
Even more basic should be the water sensor. One ATmega328P 1Mhz (internal clock) one RFM69 and two wires that look for water. (That would allow to go with two AA batteries for a very long time; down to 1.8 Volt, almost like Sensbender but more basic).
In these minimalistic setups, I do not need any bootloader. So the sketch is directly burned into the ATmega328P ( with the help of an Arduino uno as ISP) and the chip is than mounted into the sensor.

A regular watchdog reset works perfectly (tryed that with an infinit loop) but a I_REBOOT goes into a loop because the 15ms are not enough time to run enough mySensors code.
I agree that there are not many reasons you would like to reboot such a basic sensor node. But never the less I_REBOOT should work in theses cases as well.

cimba007, 2016-09-03:
Can u disable the watchdog in the "before"-function? It should be run before the mysensor codes kicks in. The watchdog can than be set as needed.

RalfJL, 2016-09-04:
Tried to disable the watchdog with wdt_disable();  in before() but did not work, which is unclear to me.
setting the watchdog to 1s in before did not work either
Most probably the node is still not fast enough to reach the function before() in time due to setup code.

It might be possible to go with a timeout of 30ms or 50ms and use before() to not get into a loop. But I would prefer to play it on the safe side.

cimba007, 2016-09-05:
What about increasing the WDT-timeout to the maximum of 8 seconds?

RalfJL, 2016-09-05:
8 seconds means that when you send an I_REBOOT message you will not have a "direct" reaction from the node.
1 second is perfect because it is long enough for the standalone-no bootloader Atmega to reboot but short enough that you get the impression of prompt reaction.

I am just proposing to increase the 15ms in the function hwReboot which is, as much as i found, only used by I_REBOOT message received by a controller.
I don't think it will have any side effects or cause any problems with the code

cimba007, 2016-09-05:
Now I understand your issue. So you will in general run into issues if you use sketches (without bootloader) that use a very short wdt-timeout as there is no time to disable it?

tekka007, 2016-09-05:
@RalfJL If you omit the bootloader in your setup, your application has to handle the watchdog as early as possible (even if not used, watchdog may enable during a brownout condition or runaway pointer leading to an eternal loop). That said, you have to reset WDRF and disable/reset the watchdog timer during early init (ideally .init3). IMHO increasing the timeout to 1s is not really a good solution.

RalfJL, 2016-09-06:
@tekka007
with the mySensors framework the earliest point where I can do something with watchdog in my sketch is the before() function and as i mentioned above I could not get it to work without going into a loop with a I_REBOOT call. A "normal" watchdog reset (forced by an infinite loop) is working with or without bootloader and with or without before() function beeing customized. So the regular watchdog seams to be set to higher value than 15ms.

I am talking about the following line:
#define hwReboot() wdt_enable(WDTO_15MS); while (1)
to be changed to
#define hwReboot() wdt_enable(WDTO_1S); while (1)

I agree it is not a very good solution, but as long as i can't solve that in before() my hands are tied.

RalfJL, 2016-09-06:
I might try the following sequence in before();
MCUSR &= ~(1<<WDRF);
WDTCSR |= (1<<WDCE) | (1<<WDE);
WDTCSR = 0x00;

There are reports that the wdt_disable() is not always working.
But it seems to be really tricky. I might also try that one in before().

  mcusr_mirror = MCUSR;
  MCUSR = 0;
  wdt_disable();

the according page suggests that wdt_disable() is not enough.

Let me try, before we change anything.

By the way, fuse is set to allow disable watchdog

henrikekblad, 2016-09-06:
There's also a preHwInit() in the dev-branch.

https://github.com/mysensors/MySensors/blob/development/core/MySensorsCore.cpp#L70

tekka007, 2016-09-06:
@RalfJL Yes, since you removed the bootloader, you have to reset MCUSR and disable/reset the watchdog ASAP:

This snippet will do the job (note the _.init3_ attribute):

```
#include <avr/wdt.h>

void wdt_init(void) __attribute__((naked)) __attribute__((section(".init3")));

void wdt_init(void) {
   MCUSR = 0;
   wdt_disable();
   return;
   }

```

RalfJL, 2016-09-07:
I will try and report.

But still I strongly believe that a sketch should be runnable unchanged on as many setups as possible. And actually with wdt_init() I will be doing something in before() that will be redone by the MySensors code a while later anyway.

So I still like you to consider to change
#define hwReboot() wdt_enable(WDTO_15MS); while (1)
to
#define hwReboot() wdt_enable(WDTO_1S); while (1)

Though not perfectly clean it would do the trick and wouldn't do any harm.
If you have a better solution in the framework, please let me know.
This "workaround" would help in not having to declare: "my sketch needs a bootloader other than the default arduino bootloader to work".
Or: "this code is there only if you plan to burn the bare sketch without bootloader"

So if we can avoid "surprises" for the developers, we should do. Not at all costs, but please still consider that change.

Thanks

RalfJL, 2016-09-08:
Code sequence:
MCUSR = 0;
wdt_disable(); 
works in before() whether you put it directly into before() or via wdt_init().
Do not use __attribute__((naked)) sketch will not start at all.
Tried it with 8Mhz internal.
Couldn't completely try it with 1Mhz because of a ping-pong timing problem, but that is a different story

cimba007, 2016-09-08:
Do I hear PING-PONG timing problem? You are welcome:

https://github.com/mysensors/MySensors/issues/578

Multiple suggestions for fix included

closed 2016-10-01 by tekka007