mysensors/MySensors #581: New internal message types

janchrillesen, 2016-09-12:
(also posted in the forum at https://forum.mysensors.org/topic/4836/new-internal-message-types/)

Since it's now possible to have sensors on the gateway and there seems to be a slight shift towards using more advanced hardware (Raspberry Pi, ESP8266, ARM based) I think it would be great to have some additional internal message types to report various internal data from both gateway and sensor nodes. So far I considered the following types

I_WIFI_RSSI - signal strenght from wifi network (dBm)
I_RADIO_RSSI - signal strengh from radio (dBm)
I_MEMORY - free memory (bytes)
I_UPTIME - uptime (seconds)
I_CPULOAD - CPU load (%)
I_VCC - supply voltage (volt)

RSSI would obvious only work for radios supporting RSSI (like the RFM69)

cimba007, 2016-09-12:
Regarding the RSSI .. remember that 50% of the supported RF modules (NRF24L01) doesn't have a very detailed RSSI indicator. Only >-64dBm I think.

I am using mycontroller and @jkandasa  was so nice to add a custom RSS-value.

[image] 

First: Total # of send messages
Second: # of lost messages even after retry
Third: total retry-count
Forth: Number of acks received without RPD=1 .. so basically with 382 and 24 I can read that ~7% of the acks were received with less then -64dBm

I track this numbers in my custom mysensor wrapper but getting the retry# from the nrf24 is very possible and easy ..

Having this "advanced statistics" I can keep much better tracking of the quality of my sensor locations.

https://github.com/mycontroller-org/mycontroller/issues/250

janchrillesen, 2016-09-12:
I am aware that only the RFM69 provides a proper RSSI indication. 

Having some statistics like you're showing would be really nice (even for radius that does not support RSSI reporting)  and I think it should be possible to add directly to an ESP8266 gateway - either through a webinterface, a small display connected via I2C or syslog messages. My last gateway is ESP8266 based and has a I2C bus for exactly that purpose (this is however offtopic for this specific issue!)

cimba007, 2016-09-12:
Luckyly mycontroller added this with currenlt the VAR_5 type .. but a custom type for general RSSI indication might be nice.

I would propose an
I_GENERIC_RSSI which might be a String:

    int s = snprintf(buff,buff_size,"rssi: %u|%u|%u|%u",_successful,_failure,_retry,_successful-_rpd);

The drawback of string is the messages might get very long and be prone to transmission errors .. a more packed representation might be better.