mysensors/MySensors #583: Example: Gateway W5100 MQTT Client does not work right

RubenKelevra, 2016-09-19:
We use the relatively new example from the dev branch, from 2-3 month ago and it worked very well. So we decided to upgrade, so we took the version from Saturday and tried our very small code on that.

Something in the network-code seems to be currently broken, were able to send packages but not able to receive them. So the TCP-SYN is send out but the SYN-ACK is never replied.

RubenKelevra, 2016-09-20:
Might be the same bug as here reported: https://github.com/mysensors/MySensors/issues/563

//cc @bitnukl

Yveaux, 2019-01-05:
Cannot be reproduced with 2.3.1.
Tested using:
- Arduino Nano
- nRF24L01+: E01-ML01D
- MySensors 2.3.1
- Arduino 1.8.8
- AVR boards 1.6.23
- Default GatewayESP8266MQTTClient.ino sketch

Decided to close.

closed 2019-01-05 by Yveaux