mysensors/MySensors #601: SPI.h include missing for W5100 MQTT Ethernet Gateway

henrikekblad, 2016-10-05:
Removal of SPI.h (#555 )include might have killed the W5100 MQTT Gateway according to reports on the forum.

https://forum.mysensors.org/topic/4802/building-a-mqtt-gateway/8

Wondering why our Jenkins builds doesn't catch this.

joshuacox, 2016-10-07:
same here building the GatewaySerial

I'm betting we're both forgetting some configuration step that needs to be better documented as this was all working when I programmed a gateway a month ago, or perhaps this is something to do with 2.0?

```
In file included from /home/arduino/sketchbook/libraries/MySensors/drivers/RF24/RF24.cpp:23:0,
                 from /home/arduino/sketchbook/libraries/MySensors/MySensors.h:295,
                 from GatewaySerial.ino:86:
/home/arduino/sketchbook/libraries/MySensors/drivers/RF24/RF24.h:37:18: fatal error: SPI.h: No such file or directory
  #include <SPI.h>
                  ^
compilation terminated.
```

adding `#include <SPI.h>` only leads to further issues BTW

```

In file included from /home/arduino/sketchbook/libraries/MySensors/MySensors.h:130:0,
                 from GatewaySerial.ino:88:
/home/arduino/sketchbook/libraries/MySensors/core/MyInclusionMode.cpp: In function ‘void inclusionModeSet(bool)’:
/home/arduino/sketchbook/libraries/MySensors/core/MyInclusionMode.cpp:45:37: error: ‘hwMillis’ was not declared in this scope
      _inclusionStartTime = hwMillis();
                                     ^
/home/arduino/sketchbook/libraries/MySensors/core/MyInclusionMode.cpp: In function ‘void inclusionProcess()’:
/home/arduino/sketchbook/libraries/MySensors/core/MyInclusionMode.cpp:58:33: error: ‘hwMillis’ was not declared in this scope
  if (_inclusionMode && hwMillis()-_inclusionStartTime>MY_INCLUSION_MODE_DURATION*1000L) {
                                 ^
In file included from /home/arduino/sketchbook/libraries/MySensors/MySensors.h:235:0,
                 from GatewaySerial.ino:88:
/home/arduino/sketchbook/libraries/MySensors/core/MyGatewayTransportSerial.cpp: In function ‘bool gatewayTransportSend(MyMessage&)’:
/home/arduino/sketchbook/libraries/MySensors/core/MyGatewayTransportSerial.cpp:35:2: error: ‘MY_SERIALDEVICE’ was not declared in this scope
  MY_SERIALDEVICE.print(protocolFormat(message));
  ^
/home/arduino/sketchbook/libraries/MySensors/core/MyGatewayTransportSerial.cpp: In function ‘bool gatewayTransportAvailable()’:
/home/arduino/sketchbook/libraries/MySensors/core/MyGatewayTransportSerial.cpp:49:9: error: ‘MY_SERIALDEVICE’ was not declared in this scope
  while (MY_SERIALDEVICE.available()) {
         ^
In file included from /home/arduino/sketchbook/libraries/MySensors/MySensors.h:28:0,
                 from GatewaySerial.ino:88:
/home/arduino/sketchbook/libraries/MySensors/core/MyTransport.cpp: In function ‘void stInitTransition()’:
/home/arduino/sketchbook/libraries/MySensors/core/MySensorsCore.h:88:52: error: ‘hwDebugPrint’ was not declared in this scope
  #define debug(x,...) hwDebugPrint(x, ##__VA_ARGS__) //!< debug
                                                    ^
/home/arduino/sketchbook/libraries/MySensors/core/MyTransport.h:150:33: note: in expansion of macro ‘debug’
  #define TRANSPORT_DEBUG(x,...) debug(x, ##__VA_ARGS__) //!< debug
                                 ^
/home/arduino/sketchbook/libraries/MySensors/core/MyTransport.cpp:56:2: note: in expansion of macro ‘TRANSPORT_DEBUG’
  TRANSPORT_DEBUG(PSTR("TSM:INIT\n"));
  ^
In file included from /home/arduino/sketchbook/libraries/MySensors/MySensors.h:287:0,
                 from GatewaySerial.ino:88:
/home/arduino/sketchbook/libraries/MySensors/core/MyTransport.cpp:73:82: error: ‘hwReadConfigBlock’ was not declared in this scope
  hwReadConfigBlock((void*)&_nc, (void*)EEPROM_NODE_ID_ADDRESS, sizeof(NodeConfig));
                                                                                  ^
In file included from /home/arduino/sketchbook/libraries/MySensors/MySensors.h:28:0,
                 from GatewaySerial.ino:88:
/home/arduino/sketchbook/libraries/MySensors/core/MyTransport.cpp: In function ‘void stInitUpdate()’:
/home/arduino/sketchbook/libraries/MySensors/core/MySensorsCore.h:88:52: error: ‘hwDebugPrint’ was not declared in this scope
  #define debug(x,...) hwDebugPrint(x, ##__VA_ARGS__) //!< debug
                                                    ^
/home/arduino/sketchbook/libraries/MySensors/core/MyTransport.h:150:33: note: in expansion of macro ‘debug’
  #define TRANSPORT_DEBUG(x,...) debug(x, ##__VA_ARGS__) //!< debug
                                 ^
/home/arduino/sketchbook/libraries/MySensors/core/MyTransport.cpp:82:3: note: in expansion of macro ‘TRANSPORT_DEBUG’
   TRANSPORT_DEBUG(PSTR("!TSM:INIT:TSP FAIL\n"));
   ^
/home/arduino/sketchbook/libraries/MySensors/core/MySensorsCore.h:88:52: error: ‘hwDebugPrint’ was not declared in this scope
  #define debug(x,...) hwDebugPrint(x, ##__VA_ARGS__) //!< debug
                                                    ^
[... 517 lines cut ...]
  hwWriteConfig(EEPROM_LOCAL_CONFIG_ADDRESS+pos, value);
                                                      ^
/home/arduino/sketchbook/libraries/MySensors/core/MySensorsCore.cpp: In function ‘uint8_t loadState(uint8_t)’:
/home/arduino/sketchbook/libraries/MySensors/core/MySensorsCore.cpp:428:53: error: ‘hwReadConfig’ was not declared in this scope
  return hwReadConfig(EEPROM_LOCAL_CONFIG_ADDRESS+pos);
                                                     ^
/home/arduino/sketchbook/libraries/MySensors/core/MySensorsCore.cpp: In function ‘void wait(long unsigned int)’:
/home/arduino/sketchbook/libraries/MySensors/core/MySensorsCore.cpp:433:33: error: ‘hwMillis’ was not declared in this scope
  unsigned long enter = hwMillis();
                                 ^
[... cut ...]
/home/arduino/sketchbook/libraries/MySensors/core/MySensorsCore.cpp: In function ‘int8_t _sleep(uint32_t, bool, uint8_t, uint8_t, uint8_t, uint8_t)’:
/home/arduino/sketchbook/libraries/MySensors/core/MySensorsCore.h:88:52: error: ‘hwDebugPrint’ was not declared in this scope
  #define debug(x,...) hwDebugPrint(x, ##__VA_ARGS__) //!< debug
                                                    ^
/home/arduino/sketchbook/libraries/MySensors/core/MySensorsCore.cpp:458:2: note: in expansion of macro ‘debug’
  debug(PSTR("MCO:SLP:MS=%lu,SMS=%d,I1=%d,M1=%d,I2=%d,M2=%d\n"), ms, smartSleep, interrupt1, mode1, interrupt2, mode2);
  ^

```

henrikekblad, 2016-10-07:
Which version of the IDE are you running? It actually compiles fine for 1.6.12 here.

joshuacox, 2016-10-07:
ya I realized it was an old version the IDE, and corrected that, the SPI.h missing is still relevant though

henrikekblad, 2016-10-07:
What do you mean? 
Are you still getting a compile error when not adding the SPI.h include?

joshuacox, 2016-10-07:
no I'm sorry, all good here, compiles fine once I add in the include, I only mean it is relevant as it should be fixed in the example code or whatever file needs that include

joshuacox, 2016-10-07:
scratch that too, after seeing your version (1.6.12) I tried the hourly build from arduino.cc and arduino.org (where I got the 1.7.11 from) and everything works fine without the SPI.h include added gatewayserial.  Sorry for the confusion.

henrikekblad, 2016-10-07:
Ok, thanks. Closing this issue now.

closed 2016-10-07 by henrikekblad