mysensors/MySensors #603: Refactor feature defines

henrikekblad, 2016-10-08:
Today we have some defines in MyConfig.h that can't be overridden in the sketch. There's also some "inverse" defines like MY_DISABLED_SERIAL that should be replaced by it's positive equivalent (I.e. MY_USE_SERIAL_FOR_DEBUG_MESSAGES). 

All configuration (and features) should have their defaults set in MyConfig.h like this:
```
#ifndef MY_XXX_FEATURE 
   #define MY_XXX_FEATURE true
#endif
```

In the code we should replace all 
```#if defined(MY_XXX_FEATURE)``` with ```#if MY_XXX_FEATURE==true```

aktentasche, 2017-02-24:
Correct me if I am wrong: the approach would be setting all commented out defines to "false" as in this feature is not enabled? Also,l this issue includes only "boolean" defines right?

mfalkvidd, 2018-02-18:
Could the next step for this be to get a list of the defines that are candidates for refactoring, so we can pick them off one by one?