mysensors/MySensors #615: Custom gateway (call receive() in GATEWAY mode)

StasiukDanylo, 2016-10-14:
Hi.
Have a problem to call receive() method in MY_GATEWAY_SERIAL mode.
I want to use data, received from my node directly on my gateway, but i can't receive data.
Method receive() does not call.
Have any options?

delinend, 2016-10-15:
I had same problem, and went to version 2.0.1-beta, and then it works. But now i can't remove the #define MY_DEBUG without the GW i stopping.

StasiukDanylo, 2016-10-15:
This problem is the current version?
I understood correctly? 
and no other options how to solve this problem? :(

delinend, 2016-10-15:
I use the development version 2.0.1-test, and it works for me. If I use the Master version 2.0.0, I had the same problem as you, that the recive() funktion i not working... Try the 2.0.1-test version.

StasiukDanylo, 2016-10-15:
Yes, in beta all work. (and disable debug too)
Why not work in stable?

delinend, 2016-10-15:
Super :-) But if I disable the MY_DEBUG in 2.0.1-beta, my GW is not working anymore :-(

tekka007, 2016-10-16:
@StasiukDanylo This is a known bug for 2.0.0 and has been fixed in 2.0.1

closed 2016-10-16 by tekka007