mysensors/MySensors #627: Signing in development branch?

CynthiaGreen, 2016-10-27:
I've been using the development branch since before the official release of 2.0.0. Today I jumped from a mid-July commit `929b001` to the latest commit `50eeb75`. Things were going swimmingly until I noticed my devices with signing enabled stopped accepting commands. They still present, but that seems to be where communication stops. 

Working backwards from the latest commit, I found `9304ceb` to be the last commit with which the devices with signing enabled accept commands. 

I looked over the commit messages, but nothing stands out. Is there an additional change needed to get signing working again, aside from my existing signing defines?

`
#define MY_SIGNING_SOFT
#define MY_SIGNING_REQUEST_SIGNATURES
`

I even tried re-applying the personalization sketch, but it didn't seem to fix the issue using future commits.

fallberg, 2016-10-28:
Personalization procedure have been updated to store secrets in EEPROM and not in sketch.
You have to update both gateway and sensor and redo personalization on both to enable them to communicate securely.

CynthiaGreen, 2016-10-29:
Yes, I have been using the EEPROM-based secrets for quite some time now. That is why I mentioned re-applying the personalization sketch. The commit that I was using was from after the release of MySensors 2.0.0. 

Oh, maybe are you saying the personalization methods have changed again since the 2.0.0 release?

fallberg, 2016-10-29:
No, they have not. Could you please provide some debugging information? The documentation for signing also have a troubleshooting section. The by far most common issue with signing is not related to signing at all actually. It is when messages are not successfully transmitted, due to rf issues or powering issues. It could be something else that has changed which affects the radio performance, and as signing uses full length transmissions it stresses the radio to the max. The debug log will show what is happening.

CynthiaGreen, 2016-10-29:
Unfortunately I only have debugging access to the gateway.

Process: Restart device with working signing; capture output on gateway; change commit on MySensors library; clean compiling caches; recompile device firmware; program new firmware to device (isp); capture output on gateway; paste here

```
--- 929b001; Signing Works
0;255;3;0;9;TSF:MSG:READ,10-10-255,s=255,c=3,t=7,pt=1,l=0,sg=0:102
0;255;3;0;9;TSF:MSG:BC
0;255;3;0;9;TSF:MSG:FPAR REQ,ID=10
0;255;3;0;9;TSF:PNG:SEND,TO=0
0;255;3;0;9;TSF:CKU:OK
0;255;3;0;9;TSF:MSG:GWL OK
0;255;3;0;9;TSF:MSG:SEND,0-0-10-10,s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=OK:0
0;255;3;0;9;TSF:MSG:READ,10-10-0,s=255,c=3,t=24,pt=1,l=1,sg=0:1
0;255;3;0;9;TSF:MSG:PINGED,ID=10,HP=1
0;255;3;0;9;!TSF:MSG:SEND,0-0-10-10,s=255,c=3,t=25,pt=1,l=1,sg=0,ft=0,st=NACK:1
0;255;3;0;9;TSF:MSG:READ,10-10-0,s=255,c=3,t=15,pt=6,l=2,sg=0:0101
0;255;3;0;9;TSF:MSG:SEND,0-0-10-10,s=255,c=3,t=15,pt=6,l=2,sg=0,ft=0,st=OK:0100
0;255;3;0;9;TSF:MSG:READ,10-10-0,s=255,c=0,t=17,pt=0,l=10,sg=0:2.0.1-beta
0;255;3;0;9;Sending message on topic: mygateway1-out/10/255/0/0/17
0;255;3;0;9;TSF:MSG:READ,10-10-0,s=255,c=3,t=6,pt=1,l=1,sg=0:0
0;255;3;0;9;Sending message on topic: mygateway1-out/10/255/3/0/6
0;255;3;0;9;TSF:MSG:READ,10-10-0,s=255,c=3,t=11,pt=0,l=20,sg=0:Device Ctl
0;255;3;0;9;Sending message on topic: mygateway1-out/10/255/3/0/11
0;255;3;0;9;TSF:MSG:READ,10-10-0,s=255,c=3,t=12,pt=0,l=12,sg=0:2.0.1-beta
0;255;3;0;9;Sending message on topic: mygateway1-out/10/255/3/0/12
0;255;3;0;9;TSF:MSG:READ,10-10-0,s=0,c=0,t=3,pt=0,l=9,sg=0:Device
0;255;3;0;9;Sending message on topic: mygateway1-out/10/0/0/0/3
0;255;3;0;9;TSF:MSG:READ,10-10-0,s=255,c=3,t=26,pt=1,l=1,sg=0:2
0;255;3;0;9;TSF:MSG:SEND,0-0-10-10,s=255,c=3,t=27,pt=1,l=1,sg=0,ft=0,st=OK:1
```

```
--- 50eeb75; Signing No Work
0;255;3;0;9;TSF:MSG:READ,10-10-255,s=255,c=3,t=7,pt=1,l=0,sg=0:7
0;255;3;0;9;TSF:MSG:BC
0;255;3;0;9;TSF:MSG:FPAR REQ,ID=10
0;255;3;0;9;TSF:PNG:SEND,TO=0
0;255;3;0;9;TSF:CKU:OK
0;255;3;0;9;TSF:MSG:GWL OK
0;255;3;0;9;TSF:MSG:SEND,0-0-10-10,s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=OK:0
0;255;3;0;9;TSF:MSG:READ,10-10-0,s=255,c=3,t=24,pt=1,l=1,sg=0:1
0;255;3;0;9;TSF:MSG:PINGED,ID=10,HP=1
0;255;3;0;9;!TSF:MSG:SEND,0-0-10-10,s=255,c=3,t=25,pt=1,l=1,sg=0,ft=0,st=NACK:1
0;255;3;0;9;TSF:MSG:READ,10-10-0,s=255,c=0,t=17,pt=0,l=10,sg=0:2.0.1-beta
0;255;3;0;9;Sending message on topic: mygateway1-out/10/255/0/0/17
0;255;3;0;9;TSF:MSG:READ,10-10-0,s=255,c=3,t=6,pt=1,l=1,sg=0:0
0;255;3;0;9;Sending message on topic: mygateway1-out/10/255/3/0/6
0;255;3;0;9;TSF:MSG:READ,10-10-0,s=255,c=3,t=11,pt=0,l=20,sg=0:Device Ctl
0;255;3;0;9;Sending message on topic: mygateway1-out/10/255/3/0/11
0;255;3;0;9;TSF:MSG:READ,10-10-0,s=255,c=3,t=12,pt=0,l=12,sg=0:2.0.1-beta
0;255;3;0;9;Sending message on topic: mygateway1-out/10/255/3/0/12
0;255;3;0;9;TSF:MSG:READ,10-10-0,s=0,c=0,t=3,pt=0,l=9,sg=0:Device
0;255;3;0;9;Sending message on topic: mygateway1-out/10/0/0/0/3
0;255;3;0;9;TSF:MSG:READ,10-10-0,s=255,c=3,t=26,pt=1,l=1,sg=0:2
0;255;3;0;9;TSF:MSG:SEND,0-0-10-10,s=255,c=3,t=27,pt=1,l=1,sg=0,ft=0,st=OK:1
```

fallberg, 2016-10-29:
Could you please also enable the flag for verbose signing debugging? I don't see a single signed message in either the working or non working case.

CynthiaGreen, 2016-10-29:
Updated previous post w/`#define MY_DEBUG_VERBOSE_SIGNING` on the gateway. 
Using a diff tool, it looks like the device isn't telling the gateway to enable or disable signing.

fallberg, 2016-10-29:
Yes indeed. And that explain why signing does not work. However, determining why is difficult to do without any debug information from the node.
What is shown however is that most likely the node has a configuration problem or a radio problem as it is not sending its signing preferences to the gateway (or the preferences fail to get through). There have been various radio optimizations done which may have had an effect on the handshaking.
But without knowing what your node attempts to do it is tricky to say the least to guess on the reason I am afraid. You have no possibility to set up a node with debugging possibility and replicate the setup?

CynthiaGreen, 2016-10-29:
I only have an isp and I do not think I can get debug info with that. I guess I need to order some kind of serial device for these purposes! Ack. 

For what it's worth, I can replicate it on all four of my nodes that signing _was_ working on.

Oh … oh! I uploaded the same sketch but w/`#define MY_DEBUG` and signing is working again. If I remove that line, signing stops working again. Phew. At least I have work around!

fallberg, 2016-10-30:
There have been reports of message collisions with signing presentation replies from gw colliding with presentation messages from nodes. A node which support signing haw a 2000ms delay to allow gw to send signing preferences. You could try to adjust that delay. If it helps by adding MY_DEBUG much point to timing related issues and that is the only one I've heard about for signing.

fallberg, 2016-10-30:
@CynthiaGreen I have made a pull request #630 that might or might not have an effect on your problem. You could have a go at testing it. It should hopefully not make things worse in any case :)
I am afraid is little more I can do to help you with the information available.
You could try to [bisect](https://git-scm.com/docs/git-bisect) your way to the specific commit that stops working and perhaps we can make conclusions based on that?

And for the record, reporting erroneous behaviour related to signing/security I never consider "bugging". It is important matters and all issues are treated seriously. :)

CynthiaGreen, 2016-11-01:
Thanks! Well I tried using your pull request, and modifying the 2s delay mentioned but neither alone seemed to improve things.

While poking around in MySigning.cpp I found the signerPresentation method and added an extra wait before the debug statement (as debugging was disabled). I tried various values for wait, but even wait(1) seemed to allow signing to work. 

I have not had an opportunity to test this on all my signing nodes, but I hope to tomorrow. I hope it was not just coincidence. 

```
diff --git i/core/MySigning.cpp w/core/MySigning.cpp
index 76ed69e..18b0c6a 100644
--- i/core/MySigning.cpp
+++ w/core/MySigning.cpp
@@ -140,6 +140,7 @@ void signerPresentation(MyMessage &msg, uint8_t destination) {
 
 #if defined(MY_SIGNING_REQUEST_SIGNATURES)
        msg.data[1] |= SIGNING_PRESENTATION_REQUIRE_SIGNATURES;
+      wait(1);
        SIGN_DEBUG(PSTR("Signing required\n"));
 #endif
 #if defined(MY_SIGNING_NODE_WHITELISTING)
```

fallberg, 2016-11-01:
I have a strong suspicion that you have a power supply issue. This might seem odd as it is software dependent, but as it is, optimizations have been made to the radio stack which have resulted in faster code. That leads to messages being transmitted more frequently and that in turn loads the supply harder. Do you have good radio supply decoupling and stabel voltages to the radio on your node? I think your node tries to send its signing presentation in close succession to its other presentation and the voltage is not stable enough to get the signing presentation through, so the GW never realize that the node require signing. So a delay before the signing presentation would allow voltage to settle, but the problem is essentially the power supply to the radio which is expected to be fast enough to tackle this. Or that's my theory anyway. :)

CynthiaGreen, 2016-11-01:
That is something I'll have to test, thank you for pointing that out. I do have a 10uF capacitor next to input from the regulated power supply, and a 10uF capacitor right next to the radio but not on the radio as I see many users do with 4.7uF capacitor. Would you suggest something else?

I <3 improvements, but I fear for your support issue queue if these software improvements unearth what-appear-to-be broken features to the user :(

fallberg, 2016-11-02:
I have made a troubleshooting section in the signing documentation that should cover all known issues users have with signing. The reason I have not just sent you there is because you don't have debug output from your node. So I can't "prove" anything. But I am pretty sure that you get NACKs on transmissions from your node and that is not a signing issue. Normally, users have access to the debug output and therefore detect any issues quite fast and there are many forum threads discussing radio problems and how to solve them.

CynthiaGreen, 2016-11-03:
I should have a USB serial device in a day or two, not sure why I ever started with just isp. I'll post debug output when I do!

In the while I'll take a look at the guide - have not needed to until now. Thanks for mentioning it.

CynthiaGreen, 2016-11-08:
Well that seemed like a waste of time. :(

With any kind of debug on, the status of nearly all messages is OK from the device. I should have realized that before wasting time debugging all day. 

I'm going to try increasing the capacitors. I noticed that while some of the MySensors documtatoin say 4.7-47uF, others strictly say 47uF. I will try increasing mine from 10uF.

I will close this for now as you probably (correctly) diagnosed the hardware issue due to software changes.

closed 2016-11-08 by CynthiaGreen