mysensors/MySensors #631: MQTT topic prefix containing a slash doesn't parse

frizzy, 2016-11-02:
I would like to use a MY_MQTT_SUBSCRIBE_TOPIC_PREFIX value containing slashes in the topic:

```MY_MQTT_SUBSCRIBE_TOPIC_PREFIX "mysensors/1/in"```

The client subscribes correctly, however protocolMQTTParse() currently seems to split the topic by "/" and compares the first element with MY_MQTT_SUBSCRIBE_TOPIC_PREFIX. It would probably be better to first trim the topic by MY_MQTT_SUBSCRIBE_TOPIC_PREFIX before splitting.

Thanks!

bblacey, 2016-11-02:
What happens if you prepend the topic with a leading "/"?  For example: "/mysensors/1/in"

frizzy, 2016-11-02:
I just tried that just to be sure, but looking at the actual code it wouldn't work. The function splits the the whole topic string by "/" and then checks if the first element is the same as the subscriber topic prefix. Prepending with a "/" would likely have element 0 as an empty string and that would not match the configured prefix.

I don't like the idea of being forced to use a leading slash for topics as this is not a requirement for MQTT. It is also discouraged [here](http://www.hivemq.com/blog/mqtt-essentials-part-5-mqtt-topics-best-practices).

bblacey, 2016-11-02:
Agreed - was probing to see if there was a quick workaround.  Care to submit a pull-request to address the issue?

henrikekblad, 2016-11-24:
Fixed in #633

closed 2016-11-24 by henrikekblad