mysensors/MySensors #641: Implement check connection and update status

pepeEL, 2016-11-08:
Hi
I use on MySensors in Domoticz relays to switches light. One of the problem is...
When power is down ok all relays is OFF or ON (what i set in sketch).  This ok ok. But when client mysensors lost connection with gateway and then i change status on relays by phisical switch , then client connect to gateway it no update status in Domoticz. Still show status before lost connection with gateway. It is big problem... because when power down client start faster then start Domoticz on Raspberry with Gateway USB.

henrikekblad, 2016-11-24:
Your node needs to send updates regularly to make this happen. There is no automatic send magic going on in the node.

closed 2016-11-24 by henrikekblad

pepeEL, 2016-11-24:
But how do it ?

pepeEL, 2016-11-24:
Please help me how i can do it ?