mysensors/MySensors #642: Node crashes when using MY_REPEATER_FUNCTION

rayvenwalker, 2016-11-08:
Using the "#define MY_REPEATER_FUNCTION" causes the node to crash when it receives a SET command from another node that also is requesting ack.

Sending node uses this code to send the set command.
'void toggleCPMState() {
    if (CPMState[4]) {
    CPMState[4] = false;
  } else {
    CPMState[4] = true;
  }
  msgCPM.setDestination(9);
  send(msgCPM.set((bool)CPMState[4]),true);
}'

Receiving node uses the code below.

`#include <DHT.h>
#include <DHT_U.h>

#define VERSION "0.1.6.4"
#define sDebug true

#define MY_RADIO_NRF24
//#define MY_RF24_PA_LEVEL RF24_PA_HIGH
//#define MY_REPEATER_FEATURE
#define MY_NODE_ID 9
#define MY_PARENT_NODE_ID 13
#define MY_PARENT_NODE_IS_STATIC

#include <SPI.h>
#include <MySensors.h>
#include <math.h>
#include <Bounce2.h>

#define RELAY_1  5  // Arduino Digital I/O pin number for first relay (second on pin+1 etc)
#define NUMBER_OF_RELAYS 4 // Total number of attached relays
#define RELAY_ON 0  // GPIO value to write to turn on attached relay
#define RELAY_OFF 1 // GPIO value to write to turn off attached relay
bool RelayState[4] = {0,0,0,0};

#define Th1PIN 0
#define CHILD_ID_TH1 0
float pad = 9850;// balance/pad resistor value, set this to
float thermr = 10000;
float lasttemp = 0 ;

#define CHILD_ID_HUM 5
#define CHILD_ID_TEMP 6
#define HUMIDITY_SENSOR_DIGITAL_PIN 4
unsigned long SLEEP_TIME = 30000; // Sleep time between reads (in milliseconds)
unsigned long old_time;

#define CHILD_ID_FLOAT 5
#define FLOAT_PIN  3  // Arduino Digital I/O pin for button/reed switch

Bounce debouncer = Bounce();
int oldValue = -1;

#define DHTTYPE DHT11   // DHT 11
DHT dht(HUMIDITY_SENSOR_DIGITAL_PIN, DHTTYPE);

float lastTemp;
float lastHum;
boolean metric = true;

MyMessage msgTh1(CHILD_ID_TH1, V_TEMP);
MyMessage msgHum(CHILD_ID_HUM, V_HUM);
MyMessage msgTemp(CHILD_ID_TEMP, V_TEMP);
MyMessage msgFloat(CHILD_ID_FLOAT, V_TRIPPED);

void before() {
  for (int sensor = 1, pin = RELAY_1; sensor <= NUMBER_OF_RELAYS; sensor++, pin++) {
    // Then set relay pins in output mode
    pinMode(pin, OUTPUT);
    // Set relay to last known state (using eeprom storage)
    digitalWrite(pin, loadState(sensor) ? RELAY_ON : RELAY_OFF);
    RelayState[sensor] = (loadState(sensor) ? RELAY_ON : RELAY_OFF);
  }
}

void setup()
{
  // Setup the button
  pinMode(FLOAT_PIN, INPUT);
  // Activate internal pull-up
  digitalWrite(FLOAT_PIN, HIGH);

  // After setting up the button, setup debouncer
  debouncer.attach(FLOAT_PIN);
  debouncer.interval(8);
}

void presentation()  {
  sendSketchInfo("Pump", VERSION);
  present(CHILD_ID_TH1, S_TEMP);
  for (int sensor = 1, pin = RELAY_1; sensor <= NUMBER_OF_RELAYS; sensor++, pin++) {
    // Register all sensors to gw (they will be created as child devices)
    present(sensor, S_BINARY);
  }
  present(CHILD_ID_HUM, S_HUM);
  present(CHILD_ID_TEMP, S_TEMP);
  present(CHILD_ID_FLOAT, S_DOOR);

}

void loop()
{

  debouncer.update();
  // Get the update value
  int value = debouncer.read();

  if (value != oldValue) {
    // Send in the new value
    send(msgFloat.set(value == HIGH ? 1 : 0));
    oldValue = value;
  }
  if ((millis() - old_time) > SLEEP_TIME)
  {
    sendHeartbeat();
    DHTLoop();
    Thermistor(Th1PIN);
    old_time = millis();
    }
}

void Thermistor(int Pin) {
  int RawADC;
  RawADC = analogRead(Pin);
  long Resistance;
  float Temp;  // Dual-Purpose variable to save space.

  Resistance = pad / ((1024.0 / RawADC) - 1);
  Temp = log(Resistance); // Saving the Log(resistance) so not to calculate  it 4 times later
  Temp = 1 / (0.001129148 + (0.000234125 * Temp) + (0.0000000876741 * Temp * Temp * Temp));
  Temp = Temp - 273.15;  // Convert Kelvin to Celsius
  //Temp = (Temp * 9.0) / 5.0 + 32.0;                 // Convert to Fahrenheit
  send(msgTh1.set(Temp, 1));
}

void DHTLoop() {
  //Serial.print("DHT Loop  ");
  float temperature = dht.readTemperature();
  if (isnan(temperature)) {
    //Serial.println("Failed reading temperature from DHT");
    send(msgTemp.set(212, 1));
  } else if (temperature != lastTemp) {
    lastTemp = temperature;
    //    if (!metric) {
    //      temperature = dht.toFahrenheit(temperature);
    //    }
    send(msgTemp.set(temperature, 1));
  }

  float humidity = dht.readHumidity();
  if (isnan(humidity)) {
    //Serial.println("Failed reading humidity from DHT");
    send(msgHum.set(99, 1));
  } else if (humidity != lastHum) {
    lastHum = humidity;
    send(msgHum.set(humidity, 1));
  }
}

void receive(const MyMessage &message) {
  int msgcmd = mGetCommand(message);
  if (sDebug) {
    Serial.print("mGetCommand = ");
    Serial.println(msgcmd);
    Serial.print("Sender = ");
    Serial.println(message.sender);
    Serial.print("Incoming change for sensor:");
    Serial.print(message.sensor);
    Serial.print(", New status: ");
    Serial.println(message.getBool());
  }
  if (message.type == V_STATUS) {
    // Command type 1 is a set command
    if (msgcmd == 1) {
      digitalWrite(message.sensor - 1 + RELAY_1, message.getBool() ? RELAY_ON : RELAY_OFF);
      RelayState[message.sensor] = message.getBool();
      saveState(message.sensor, message.getBool());
    }
    // Command type 2 is a request for the sensor state
    if (msgcmd == 2) {
      MyMessage msg(message.sensor, V_STATUS);
      msg.setDestination(message.sender);
      msg.setType(V_STATUS);
      msg.set(RelayState[message.sensor]);
      send(msg);
    }
  }
}`

tekka007, 2016-11-08:
Can you share additional information on you issue, such as
- HW description (type, power supply, radio power supply)
- network desctiption
- Arduino & board def version
- MySensors version
- Full debug log of all involved nodes

rayvenwalker, 2016-11-09:
I cant supply a debug of the crashing node.  Not and keep it in the
location it is in connected to its sensors.
Using a 'TheSimple" Arduino Pro Mini 5v 16Mhz for both nodes (sending and
receiving)
Mysensors 2.0

I could possibly send the debug log for the sending node which is how i
realized the node was restarting.  Because is presented it;s sketch info
right after i sent the command.

I will reply back later today with additional info.

rayvenwalker, 2016-11-11:
The arduino that this was running on was bad.  I'm closing the issue.

closed 2016-11-11 by rayvenwalker