mysensors/MySensors #656: MySensors EEPROM save/load Position

sghazagh, 2016-11-24:
Hi,
I have created a sensor which working fine with my Controller.
I also have implemented webserver on my board (Wemos D1 mini Pro) to have control on sensor over the web as well.
So it gives me option to access to the webserver on the board directly to control the sensor if for some reason controller was offline.

For my webserver, I made it dynamic and saving SSID, Password and static IP, Gateway , Subnet Mask addresses into the eeprom start from position 0 and uses up to address 205.

I also need to save and load MySensor relay state on eeprom. So, i have set the position to save on location 206 (one byte after my webserver settings).
But it always writes on position other than what I ask so it mess up my settings saved on other locations.

Can you please double check why the "saveState(206, message.getBool());" does not save the status on address that I have specified?

Note: I also use "loadState(206);" to retrieve the status. It also reads it from same position that saveSate was used. (So it's working but my webserver settings does get mess and will not have proper values if I use saveState function)

Hope I could explained well!
Many thanks,

henrikekblad, 2016-11-24:
Are you using master or development branch?
Could you post the sketch?

sghazagh, 2016-11-24:
I'm using the one install with Arduino library manager. It's version 2.0 do should be the latest stake version.
My code odds do lots of stuff, but to simplify that, address  0 to 205 is used by my code and can restore dara properly with no issue.
SSID saved from 0 to 32.
When I saveState my relay status for instance on pos 206, only this part of my data (0 to 32) get messy. The rest can be read ok.
But why does position 206 saves something on this area, I don't understand?

sghazagh, 2016-11-24:
Just an update, I shifted the address to write my stuff from 10 instead of 0, and all are ok.
So it defently has something to do with address 0~10 when you use saveState even the position set somewhere else!

worth to have a look and see if it's valid!

tekka007, 2016-11-24:
MySensors uses the first ~500 EEPROM bytes to store internal data - you can use either

`load/saveState(pos,byte) `

which stores data outside of the MySensors range (effectively at EEPROM position _EEPROM_LOCAL_CONFIG_ADDRESS_ + pos), 

For direct EEPROM RW commands, make sure to always include the offset:

`EEPROM_LOCAL_CONFIG_ADDRESS`

elar-systems, 2016-11-24:
Thanks tekka007,
Can you please give me a short example how to use "EEPROM_LOCAL_CONFIG_ADDRESS"?
As per "https://github.com/mysensors/MySensors/blob/development/core/MyEepromAddresses.h"
It depends on other variables as well.
for my stuff, I can shift them by start writing from byte 501 afterward.
how can I use this offset in my code? should I define it or it already has value by MySensors library initiation?

tekka007, 2016-11-25:
yes, it is defined by the library - here is an example:
`EEPROM.write(EEPROM_LOCAL_CONFIG_ADDRESS + pos, 100);
`

tekka007, 2016-11-27:
@elar-systems Any feedback on that issue or can I close it?

elar-systems, 2016-11-27:
Hello my friend,
It's still not clear what address we can use to be safe for storing data!
In my case, I shifted the data to start from byte 10 instead of 0 and it works for now!
But I think you either better to clarify that what address is safe to start writing data or have a look and see why the saveState(pos, data) with pos set to 106 for example , saves the data somewhere between byte 0 to 10~

You mentioned in last post to use: "EEPROM.write(EEPROM_LOCAL_CONFIG_ADDRESS + pos, 100);"
But it's not clear to me that should I set the "EEPROM_LOCAL_CONFIG_ADDRESS" by myself or not?
This variable depends to all other variables before it listed in MyEepromAddress.h" file.
And it already address whole 256 byte of Arduino eeproms.
Mine is WemosD1 Mini Pro, so it has 4096 of EEPROM. It might not be an issue for boards like Wemos D1 Mini with large size of EEPROM, but it will make problem for boards with less EEPROM size.
Please advise how I can find the safe FREE area to start writing the variables, and if you think the case is resolved, it's up to you then to close it!

Many thanks

tekka007, 2016-11-27:
EEPROM_LOCAL_CONFIG_ADDRESS is set by the library - you can simply use it as offset as described above. Alternatively, use load/saveState() from the MySensors API that does all the internal mapping.

These two commands write to the exact same EEPROM position:

`EEPROM.write(EEPROM_LOCAL_CONFIG_ADDRESS + 10, 123);
`

and

`saveState(10, 123);`

elar-systems, 2016-11-27:
Then that exactly is the problem.
I have used saveState(206, xxx);
Then it messed up my saved data on bytes between 0~10!
My data was written between 0 to 205. And I had used one byte after to be used by MySensore saveState.
And it's mess up my saved area at the beginning of the eeprom address (somewhere between 0 to 100)

This was my question from the beginning: Why the MySensors writes something on area 0~10 even when I mention to save the data specifically at Byte 206 when I use saveState(206,xxx);   ???

tekka007, 2016-11-27:
Looking at the code, the EEPROM addresses cannot wrap around with pos 206. Can you upload your sketch and provide a full EEPROM dump - something sounds not right here...

elar-systems, 2016-11-27:
That's part of the code writes two variables of my program (it saves more variables after but not a big concern):

```
Serial.println("writing eeprom ssid:");
      for (int i = 0; i < qssid.length(); ++i) {
        EEPROM.write(i, qssid[i]);
        Serial.print("Wrote: ");
        Serial.println(qssid[i]);
      }
      Serial.println("writing eeprom pass:");
      for (int i = 0; i < qpass.length(); ++i) {
        EEPROM.write(32 + i, qpass[i]);
        Serial.print("Wrote: ");
        Serial.println(qpass[i]);
      }
EEPROM.commit();
```
and here is part of the code reads the data:
```
 Serial.println("Reading SSID from EEPROM");
  String essid;
  for (int i = 0; i < 32; ++i) {
    essid += char(EEPROM.read(i));
  }
  Serial.print("SSID: ");
  Serial.println(essid);
  
  Serial.println("Reading Password from EEPROM");
  String epass = "";
  for (int i = 32; i < 96; ++i) {
    epass += char(EEPROM.read(i));
  }
  Serial.print("PASS: ");
Serial.println(epass);
```

They are all ok if I don't use saveState somewhere in my code.
Otherwise, the reading part shows funny charachter for saved SSID! (bytes 0 to 31)

tekka007, 2016-11-27:
ok, MySensors uses EEPROM addresses 0 - EEPROM_LOCAL_CONFIG_ADDRESS (which is ~413) for internal data - this region should not be overwritten with user data.

Please update your code to:

```
Serial.println("writing eeprom ssid:");
      for (int i = 0; i < qssid.length(); ++i) {
        EEPROM.write(EEPROM_LOCAL_CONFIG_ADDRESS + i, qssid[i]);
        Serial.print("Wrote: ");
        Serial.println(qssid[i]);
      }
      Serial.println("writing eeprom pass:");
      for (int i = 0; i < qpass.length(); ++i) {
        EEPROM.write(EEPROM_LOCAL_CONFIG_ADDRESS + 32 + i, qpass[i]);
        Serial.print("Wrote: ");
        Serial.println(qpass[i]);
      }
EEPROM.commit();
```

and 

```
Serial.println("Reading SSID from EEPROM");
  String essid;
  for (int i = 0; i < 32; ++i) {
    essid += char(EEPROM.read(EEPROM_LOCAL_CONFIG_ADDRESS + i));
  }
  Serial.print("SSID: ");
  Serial.println(essid);
  
  Serial.println("Reading Password from EEPROM");
  String epass = "";
  for (int i = 32; i < 96; ++i) {
    epass += char(EEPROM.read(EEPROM_LOCAL_CONFIG_ADDRESS + i));
  }
  Serial.print("PASS: ");
Serial.println(epass);
```

The first free position to be used for save/loadState() is 96 (position 95 is the last char of your password).

Does that make sense?

elar-systems, 2016-11-27:
of course it makes sense mate, because even when I used start point of byte 10, before knowing I can use "EEPROM_LOCAL_CONFIG_ADDRESS", it was working as well.
So I thinbk your solution will be the proper one. Instead of fixed 10 byte value, I use "EEPROM_LOCAL_CONFIG_ADDRESS" to be added in front to shift my data.
I cannot test the code till 8-10 hours later.
Will try it and will update you here. It should be OK anyway.

Thanks for your help and clarification,

sghazagh, 2016-11-28:
All good body,
thanks for helping me.
I think this can be closed for good ;)

closed 2016-11-28 by tekka007