mysensors/MySensors #658: GatewaySerialRS485 sends 0;255;3;0;5;0 to controller endlessly.

rayvenwalker, 2016-11-24:
referencing comment from a mycontroller issue:
https://github.com/mycontroller-org/mycontroller/issues/312#issuecomment-262795830

I have noticed several times the gateway 'locking up' endlessly sending '''0;255;3;0;5;0''' to the controller.

In the code for the gateway I have remarked the inclusion
mode feature.

'''//#define MY_INCLUSION_MODE_FEATURE'''

in file:

https://github.com/rayvenwalker/StillBot-mysensors/blob/master/GatewayRS485/GatewayRS485.ino

rayvenwalker, 2016-11-24:
All the sensors and the gateway are using the MYSBootloader.

henrikekblad, 2016-11-24:
That sounds very strange. If MY_INCLUSION_MODE_FEATURE isn't defined, the code that sends "0;255;3;0;5;0" gets removed entirely.

rayvenwalker, 2016-11-24:
Could this have anything to do with the MYSBootloader or the watchdog timer?

henrikekblad, 2016-11-24:
I doubt it. Does it behave the same with default optiboot?

rayvenwalker, 2016-11-24:
No, I was having erratic issues with all of my rs485 nodes and I updated all the bootloaders to the latest MYSBootlader at the same time.  I added a watchdog timer to the sketch because I was having to reset the gateway manually a few times a day.

tekka007, 2016-11-24:
I'm a little bit puzzled why you are using MYSBootloader on rs485 nodes/gw. MYSBootloader is only supporting RF24 radios...

rayvenwalker, 2016-11-24:
Hmm.....  I had been trying to use OTA with rs485 witch @hek mentioned
should work.

https://forum.mysensors.org/topic/4909/wireless-programming/12

This might be a bigger question but I thought in all my reading I saw that
the MSYBootloader was a more reliable bootloader for using with the
mysensors library.  Is it not necessary when using RS485?

tekka007, 2016-11-24:
Yes, in your case you should use a transport-agnostic bootloader (Dualoptiboot) and external flash; 
Here you'll find more information about this topic and the differences between both bootloaders: https://www.mysensors.org/about/ota

rayvenwalker, 2016-11-25:
Thanks for the heads up.  And then for the gateway the stock firmware that
ships with the Arduino IDE?

tbowmo, 2016-11-25:
Use the stock bootloader for the gateway.. The OTA features in mysensors is only for the nodes.

rayvenwalker, 2016-11-25:
So that's the one that comes with the IDE sometimes labled optiboot?
Thanks for the info.  Could this be what caused the behavior I experienced?

rayvenwalker, 2016-11-26:
I have updated the bootloader on the gateway and sensor nodes to the optiboot that comes with the arduino ide.  I have removed the watchdog code from the gateway sketch. I am experiencing gateway lockups every few hours now.  

I could probably tap into the serial pins on the gateway and record the traffic.   Would this be helpful.

tekka007, 2016-12-03:
Any feedback on this issue, i.e. debug logs, etc.?

rayvenwalker, 2016-12-04:
Still occurring.  I have to reset the gateway every few hours.  I can post
whatever logs you would like...   Currently I only have logs for the
MyController.

rayvenwalker, 2016-12-04:
I could possibly capture the traffic from the gateway to the computer,
would this be helpful?

tekka007, 2016-12-05:
Any log supporting your case would be helpful (ideally the MySensors GW debug log, node log) and also your GW sketch. Also, can you rule out power issues and/or voltage fluctuations?

rayvenwalker, 2016-12-05:
I have changed the platform (from rpi to pcduino) and arduino that I am
using.  The code is all published to
https://github.com/rayvenwalker/StillBot-mysensors currently the only
sensors on the rs458 gateway are the controller and the elementpower
sketches.  The GatwayRS485 is the sketch on the gateway.  I removed the
watchdog code that is present in the version on github so that I would know
if it was still crashing or not.  Their is a log file attached to the
initial bug report I made when this error caused my controller(
mycontroller.org) to crash.  In this bug report you'll notice that in the
post "
https://github.com/mycontroller-org/mycontroller/issues/312#issuecomment-262795830"
@jkandasa described what he saw in the mycontroller logs.  I'll turn the
mycontroller server off tonight and add the MY_DEBUG flag to the sketch.
Then save a log of what the gateway sends, is this what you meant by "MySensors
GW debug log"?

tekka007, 2016-12-05:
@rayvenwalker
> I'll turn the mycontroller server off tonight and add the MY_DEBUG flag to the sketch.
> Then save a log of what the gateway sends, is this what you meant by "MySensors
> GW debug log"?

Yes, that's it.

rayvenwalker, 2016-12-07:
The gateway just crashed again.  It's been up since my last post though.  I had it connected to my controller at the time but I was capturing the gateway traffic... here is what I've got for the last few messages.  I'll add that on this network their is no NODEID 39 or 2.

200;0;1;0;4;995
200;1;1;0;0;19.75
202;0;1;0;39;0.0
202;1;1;0;38;116.5
39;0.0
2;1;1;0;3;255;3;0;22;3499177153
0;255;3;0;22;3500976448
0;255;3;0;22;3502775482
0;255;3;0;22;3504574651

rayvenwalker, 2016-12-08:
Got another crash this morning.  I forgot to unplug from the controller and enable MY_DEBUG so all i have is the gateway side of the serial traffic.

202;1;1;0;38;116.6
202;2;1;0;17;0.0
202;3;1;0;3;0.0
202;5;1;0;16;0
202;6;1;0;17;0.0
202;7;1;0;39;32.0
200;4;1;0;0;18.50
200;5;1;0;0;17.781
200;9;1;0;2;0
200;10;1;0;2;0
200;11;1;0;2;0
202;8;2;0;39;
202;0;1;0;39;0.0
202;1;1;0;38;119.8
0;255;3;0;22;19520776
0;255;3;0;22;21320045
0;255;3;0;22;23119251
0;255;3;0;22;24918656
0;255;3;0;22;26717594
0;255;3;0;22;28517069
0;255;3;0;22;30316167
0;255;3;0;22;32115358
0;255;3;0;22;33914686
0;255;3;0;22;35713960

tekka007, 2016-12-09:
This doesn't help much, we really need to see the MY_DEBUG output for further troubleshooting.

rayvenwalker, 2016-12-10:
OK, here is the output from last night.

0;255;3;0;9;TSP:MSG:READ 202-202-0 s=2,c=1,t=17,pt=7,l=5,sg=0:0.0
202;2;1;0;17;0.0
0;255;3;0;9;TSP:MSG:READ 202-202-0 s=3,c=1,t=3,pt=7,l=5,sg=0:0.0
202;3;1;0;3;0.0
0;255;3;0;9;TSP:MSG:READ 202-202-0 s=5,c=1,t=16,pt=1,l=1,sg=0:0
202;5;1;0;16;0
0;255;3;0;9;TSP:MSG:READ 202-202-0 s=6,c=1,t=17,pt=7,l=5,sg=0:0.0
202;6;1;0;17;0.0
0;255;3;0;9;TSP:MSG:READ 202-202-0 s=7,c=1,t=39,pt=7,l=5,sg=0:32.0
202;7;1;0;39;32.0
0;255;3;0;9;TSP:MSG:READ 200-200-0 s=0,c=1,t=4,pt=2,l=2,sg=0:1009
200;0;1;0;4;1009
0;255;3;0;9;TSP:MSG:READ 202-202-0 s=8,c=2,t=39,pt=0,l=0,sg=0:
202;8;2;0;39;
0;255;3;0;9;TSP:MSG:READ 200-200-0 s=1,c=1,t=0,pt=7,l=5,sg=0:16.50
200;1;1;0;0;16.50
0;255;3;0;9;TSP:MSG:READ 200-200-0 s=4,c=1,t=0,pt=7,l=5,sg=0:15.50
200;4;1;0;0;15.50
0;255;3;0;9;TSP:MSG:READ 200-200-0 s=7,c=1,t=47,pt=0,l=7,sg=0:T03 Err
0;255;3;0;9;TSP:MSG:ACK msg
0;255;3;0;9;TSP:MSG:SEND 0-0-200-200
s=7,c=1,t=47,pt=0,l=7,sg=0,ft=0,st=ok:T03 Err
200;7;1;0;47;T03 Err
0;255;3;0;9;TSP:MSG:READ 11-9-0 s=4,c=3,t=41,pt=0,l=0,sg=0:
0;255;3;0;9;!TSP:MSG:PVER mismatch
0;255;3;0;9;TSP:MSG:READ 202-202-0 s=0,c=1,t=39,pt=7,l=5,sg=0:0.0
202;0;1;0;39;0.0
0;255;3;0;9;TSP:MSG:READ 202-202-0 s=1,c=1,t=38,pt=7,l=5,sg=0:114.4
202;1;1;0;38;114.4
0;255;3;0;9;TSP:MSG:READ 202-202-0 s=2,c=1,t=17,pt=7,l=5,sg=0:0.0
202;2;1;0;17;0.0
0;255;3;0;9;TSP:MSG:READ 202-202-0 s=3,c=1,t=3,pt=7,l=5,sg=0:0.0
202;3;1;0;3;0.0
0;255;3;0;9;TSP:MSG:READ 202-202-0 s=5,c=1,t=16,pt=1,l=1,sg=0:0
202;5;1;0;16;0
0;255;3;0;9;TSP:MSG:READ 202-202-0 s=6,c=1,t=17,pt=7,l=5,sg=0:0.0
202;6;1;0;17;0.0
0;255;3;0;9;TSP:MSG:READ 202-202-0 s=7,c=1,t=39,pt=7,l=5,sg=0:32.0
202;7;1;0;39;32.0
0;255;3;0;9;TSP:MSG:READ 200-200-0 s=0,c=1,t=4,pt=2,l=2,sg=0:1009
200;0;1;0;4;1009
0;255;3;0;9;TSP:MSG:READ 202-202-0 s=8,c=2,t=39,pt=0,l=0,sg=0:
202;8;2;0;39;
0;255;3;0;9;TSP:MSG:READ 200-200-0 s=1,c=1,t=0,pt=7,l=5,sg=0:16.00
200;1;1;0;0;16.00
0;255;3;0;9;TSP:MSG:READ 200-200-0 s=4,c=1,t=0,pt=7,l=5,sg=0:15.75
200;4;1;0;0;15.75
0;255;3;0;9;TSP:MSG:READ 200-200-0 s=7,c=1,t=47,pt=0,l=7,sg=0:T03 Err
0;255;3;0;9;TSP:MSG:ACK msg
0;255;3;0;9;TSP:MSG:SEND 0-0-200-200
s=7,c=1,t=47,pt=0,l=7,sg=0,ft=0,st=ok:T03 Err
200;7;1;0;47;T03 Err
THIS IS WHERE I RESTARTED THE NODE BECAUSE IT WAS UNRESPONSIVE
0;255;3;0;9;Starting gateway (RSNGA-, 2.0.0)
0;255;3;0;9;TSM:INIT
0;255;3;0;9;TSM:RADIO:OK
0;255;3;0;9;TSM:GW MODE
0;255;3;0;9;TSM:READY
0;255;3;0;14;Gateway startup complete.
0;255;0;0;18;2.0.0
0;255;3;0;9;No registration required
0;255;3;0;9;Init complete, id=0, parent=0, distance=0, registration=1
0;255;3;0;9;TSP:MSG:READ 200-200-255 s=255,c=3,t=7,pt=0,l=0,sg=0:
0;255;3;0;9;TSP:MSG:BC
0;255;3;0;9;TSP:MSG:FPAR REQ (sender=200)
0;255;3;0;9;TSP:CHKUPL:OK (FLDCTRL)
0;255;3;0;9;TSP:MSG:GWL OK
0;255;3;0;9;TSP:MSG:SEND 0-0-200-200
s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=ok:0
0;255;3;0;9;TSP:MSG:READ 200-200-0 s=255,c=3,t=24,pt=1,l=1,sg=0:1
0;255;3;0;9;TSP:MSG:PINGED (ID=200, hops=1)
0;255;3;0;9;TSP:MSG:SEND 0-0-200-200
s=255,c=3,t=25,pt=1,l=1,sg=0,ft=0,st=ok:1

rayvenwalker, 2016-12-11:
I have noticed since I lowered the number of nodes, i.e. reduced the
traffic by over 50%.  The frequency of lockups has reduced to once a day.

tekka007, 2016-12-11:
Can you add a incremental counter in loop() and print the number every minute or so to see if the GW is freezing?

tekka007, 2016-12-11:
actually, you also seem to have transmission issues:

```
0;255;3;0;9;!TSP:MSG:PVER mismatch
```

Does the GW crash regularly after such a message?

henrikekblad, 2016-12-28:
Closed due to inactivity.

closed 2016-12-28 by henrikekblad

Elektroman2, 2017-11-03:
I have also a similar issue: 
Message from gateway received: 0;255;3;0;5;0
11:08:29.094 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;1
11:09:22.380 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;9;162000135 TSM:READY:NWD REQ
11:09:22.461 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;9;162000151 TSF:MSG:SEND,0-0-255-255,s=255,c=3,t=20,pt=0,l=0,sg=0,ft=0,st=OK:
11:09:23.044 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;9;162000799 TSF:MSG:READ,7-7-0,s=255,c=3,t=21,pt=1,l=1,sg=0:0
11:09:23.142 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 7;255;3;0;21;0
11:09:25.432 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;9;162003180 TSF:SAN:OK
11:09:29.023 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;0
11:09:29.090 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;1
11:09:37.852 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;9;162015605 TSF:MSG:READ,0-0-0,s=1,c=1,t=16,pt=0,l=1,sg=0:1
11:09:37.950 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;1;1;0;16;1
11:09:38.017 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;9;162015634 TSF:MSG:READ,0-0-12,s=2,c=1,t=64,pt=7,l=5,sg=0:24.70
11:09:38.119 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;9;162015634 !TSF:MSG:PVER,0!=2
11:10:29.020 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;0
11:10:29.088 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;1
11:11:29.015 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;0
11:11:29.091 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;1
11:12:29.023 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;0
11:12:29.090 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;1
11:12:37.965 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;9;162195736 TSF:MSG:READ,0-0-0,s=1,c=1,t=16,pt=0,l=1,sg=0:1
11:12:38.062 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;1;1;0;16;1
11:13:29.016 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;0
11:13:29.086 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;1
11:14:29.025 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;0
11:14:29.094 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;1
11:15:29.020 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;0
11:15:29.089 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;1
11:15:38.107 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;9;162375899 TSF:MSG:READ,0-0-12,s=2,c=1,t=64,pt=7,l=5,sg=0:24.70
11:15:38.208 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;9;162375900 !TSF:MSG:PVER,0!=2
11:16:29.013 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;0
11:16:29.082 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;1
11:17:29.006 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;0
11:17:29.075 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;1
11:18:29.009 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;0
11:18:29.078 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;1
11:18:38.194 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;9;162556002 TSF:MSG:READ,0-0-0,s=1,c=1,t=16,pt=0,l=1,sg=0:1
11:18:38.292 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;1;1;0;16;1
11:19:29.000 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;0
11:19:29.069 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;1
11:20:28.998 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;0
11:20:29.067 [DEBUG] [orsAbstractConnection$MySensorsReader] - Message from gateway received: 0;255;3;0;5;1

Controller: openHAB 2.2.0 Build #1065 (Debian 9.2 kernel:4.9.0-4-amd64)
Binding: org.openhab.binding.mysensors-2.2.0-SNAPSHOT.jar
Gateway: serial, STM32duino (STM32F103)
Transport: RS485
Nodes:  STM32duino (STM32F103)
INCLUSION currently enabled. (Maybe I should disable it? )
If any other log need just tell me what needed.

Thanks in advice, Marcell