mysensors/MySensors #665: digitalPinToInterrupt function is missing with using a MoteinoMega AVR1284P

trlafleur, 2016-11-27:
In RFM95.ccp file at line ~ 160, their is a reference for digitalPinToInterrupt(), that is missing when using the MoteinoMega 1284P processor.  Remove this for now, as listed below solved the compile issue.

~~~
	// IRQ
	hwPinMode(MY_RFM95_IRQ_PIN, INPUT);
	#if defined (SPI_HAS_TRANSACTION) && !defined (ESP8266) 
		_SPI.usingInterrupt((MY_RFM95_IRQ_PIN));
		//_SPI.usingInterrupt(digitalPinToInterrupt(MY_RFM95_IRQ_PIN));
	#endif
	attachInterrupt((MY_RFM95_IRQ_PIN), RFM95_interruptHandler, RISING);
	//attachInterrupt(digitalPinToInterrupt(MY_RFM95_IRQ_PIN), RFM95_interruptHandler, RISING);
~~~

bblacey, 2016-11-27:
@trlafleur - these changes as well as those citied in #664 would be best submitted as a pull request against the development branch.  Submitting a pull request will cause the MySensors CI server to build MySensors and run regression tests.  Do you plan to submit these as a [pull request](url)?

trlafleur, 2016-11-27:
From pass experience, my development environment is NOT setup up to do this
correctly with the MySensor team...

So I will reference the issues that I find, but will leave it to other to
integrate into the code base.

tekka007, 2016-11-27:
digitalPinToInterrupt() is missing from the pin definitions MoteinoMega/pins_arduino.h 

See also: https://github.com/LowPowerLab/Moteino/issues/2

closed 2016-11-29 by tekka007