mysensors/MySensors #672: Develop commit-msg hook to catch "bad commit messages" before git commit

bblacey, 2016-11-28:
Currently git commit messages are validated by Jenkins to ensure compliance with the MySensors' standards, a derivative of [Chris Beam's Seven Rules to writing good commit messages](http://chris.beams.io/posts/git-commit/).

fallberg, 2016-11-28:
I think we can start with having some container for generic/shared scripts usable by both hooks and jenkins. I thought about this for the astyle and cppcheck handling, but they are not really identical, and the settings for both are already in separate files. Jenkins will look at the entire repo while the hooks only concern themselvs with the changes in a specific commit.
But parsing a commit message and validating a set of rules is a distinct functionality that can be put in a standalone script, perferrably (for jenkins sake) one that can return a set of codes depending on what rule was violated so Jenkis can send proper notifications. The commit hook could use the same codes to give more specific details on why a particular commit message was rejected.

bblacey, 2016-12-07:
The scope of this work entails two phases.  

1. Develop tool bundle framework/runtime and port existing astyle and cppcheck tools to bundles
2. Investigate and implement commit-msg as a re-usable tool bundle

The first phase is feature complete and refactors the existing tools into a lightweight bundle framework that packages the tools into bundles that encapsulate the command-line options and configuration files wrapped in git command aliases offering a simple command-line interface to consistently run the tools across modalities (i.e. local development, continuous integration, etc.).  So far the framework has only been tested on GNU bash, version 3.2.57(1)-release (x86_64-apple-darwin16) running on macOS.  Cross-platform testing and bug fixing is pending but the framework is working sufficiently to proceed with phase 2.

The bundle framework development is available for review here https://github.com/bblacey/MySensors/tree/commit-msg-hook/.tools

bblacey, 2016-12-07:
For phase 2, verifying commit messages clearly isn’t rocket science but there is a lot of prior art that we may be able to leverage offering varying levels of configurability. However much of what seems to be available strictly comprises commit-msg scripts that are geared towards interactive commit message validation errors and editing during the commit process.  For example, [git-good commit ](https://github.com/tommarshall/git-good-commit) looks great, is pure bash (i.e. no additional prerequisites) but would need to be slightly modified to work non-interactively and the rules are hardwired into the script (consider this a nice improvement over what we have now).

One fairly promising option is [gitlint](http://jorisroovers.github.io/gitlint/) that offers decent configurability, a broad set of [rules](http://jorisroovers.github.io/gitlint/rules/), is a command-line tool, includes a commit-msg hook but requires python as a prerequisite.  It looks like it covers the main Chris Beam rules except verifying use of imperative mood in the subject line. I have [prototyped gitlint on this branch](https://github.com/bblacey/MySensors/tree/gitlint-test/.tools).

In addition, here are some [results of running gitlint against the MySensors commit history](https://gist.github.com/bblacey/b531103841dc79a1432c46d0107f4841) run with the default behavior but[ configured to match the rules codified in the Jenkins scripts](https://github.com/bblacey/MySensors/blob/gitlint-test/.tools/gitlint/config/commitlintsettings.cfg).  This resulted in quite a bit of noise in the results due to commit message bodies either missing or being too short so I ran it again with it configured to `ignore=body-is-missing, body-min-length` to increase the signal to noise for examination - [see the results ](https://gist.github.com/bblacey/c9d871d6a618f32860f160778dc89412).  gitlint looks good if we can live with python as a prerequisite and lack of verification of imperative mood in subjects.

Next I’m going to experiment with another popular ruby script [fit-commit](https://github.com/m1foley/fit-commit) to see how it stacks up because it seems to be fairly popular.  I haven't looked at it yet and might find that it isn't well-suite to running from the command line...

At the end of the day, if integrating an existing tool into our stack offers low enough barriers relative to the value, then we will do it, otherwise concede to a roll-our-own bash script because again, this isn't rocket science.

bblacey, 2016-12-11:
Phase I completed, tested and submitted under pull request #685.  Hope to find some time to work phase II this week, beyond my last gitlint update.

mfalkvidd, 2018-01-25:
Should we close this? If not, what is the next step?

fallberg, 2018-02-18:
I will try to implement some lightweight scripted verification. I am not a huge fan of introducing more external tool dependencies to the bundle.

PR #1067 (2018-03-06): Add git commit message hook to the mystools bundle
The message hook validates that the c ommit message follows
the MySensors requirements.
More specifically, it checks that
- Commit message does not exceed a line length of 72 characters
- Commit title start with an uppercase character
- Commit title does not end with a period ('.')

Fixes #672
commit 3f2d2921 (2018-03-06): Add git commit message hook to the mystools bundle (#1067)

The message hook validates that the c ommit message follows
the MySensors requirements.
More specifically, it checks that
- Commit message does not exceed a line length of 72 characters
- Commit title start with an uppercase character
- Commit title does not end with a period ('.')

Fixes #672
closed 2018-03-06 by bblacey