mysensors/MySensors #675: Ability to set node ID at run time..

trlafleur, 2016-11-30:
At this points, node ID is delivered automatically if you have a controller that supports this, or via setting it as a static value at compile time...

If one is using MQTT, there is no ability to set automatically..

 Enhancement request is to allow setting of node ID at runtime... we need to ability to read setting (jumpers) on the MySensor device to determine its node ID.

tekka007, 2016-11-30:
This is implemented. The function you are looking for is `transportAssignNodeID(newNodeID)`

trlafleur, 2016-12-01:
I did a test....

I set:
~~~
#define MY_NODE_ID            10 
~~~
In setup I set:
~~~
transportAssignNodeID(15);
~~~
As you can see from the log's below, node ID stays at 10....
I do NOT see any change to ID = 15
~~~
0 MCO:BGN:INIT NODE,CP=RLNNA--,VER=2.1.0-beta
4 TSM:INIT
5 TSF:WUR:MS=0
6 RFM95:INIT
17 RFM95:PTX:LEVEL=23
19 TSM:INIT:TSP OK
21 TSM:INIT:STATID=10
23 TSF:SID:OK,ID=10
24 TSM:FPAR
25 RFM95:SWR:SEND TO=255,RETRY=0
77 TSF:MSG:SEND,10-10-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
637 RFM95:RCV:SEND ACK
640 RFM95:SAC:SEND ACK to=0,RSSI=123
645 TSF:MSG:READ,0-0-10,s=255,c=3,t=8,pt=1,l=1,sg=0:0
650 TSF:MSG:FPAR PREF
652 TSF:MSG:FPAR OK,ID=0,D=1
655 TSM:FPAR:OK
656 TSM:ID
657 TSM:ID:OK
659 TSM:UPL
660 RFM95:SWR:SEND TO=0,RETRY=0
1235 !RFM95:SWR:NACK
1236 RFM95:PTX:LEVEL=23
1240 RFM95:SWR:SEND TO=0,RETRY=1
1343 RFM95:SWR:ACK FROM=0,SEQ=3,RSSI=-15,SNR=9
1347 RFM95:ATC:ADJ TXL,cR=-15,tR=-60,TXL=22
1351 TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=OK:1
1946 RFM95:RCV:SEND ACK
1948 RFM95:SAC:SEND ACK to=0,RSSI=123
1954 TSF:MSG:READ,0-0-10,s=255,c=3,t=25,pt=1,l=1,sg=0:1
1959 TSF:MSG:PONG RECV,HP=1
1961 TSM:UPL:OK
1964 TSM:READY:ID=10,PAR=0,DIS=1
1967 RFM95:SWR:SEND TO=0,RETRY=0
2544 !RFM95:SWR:NACK
2545 RFM95:PTX:LEVEL=22
2548 RFM95:SWR:SEND TO=0,RETRY=1
2652 RFM95:SWR:ACK FROM=0,SEQ=6,RSSI=-14,SNR=10
2656 RFM95:ATC:ADJ TXL,cR=-14,tR=-60,TXL=21
2660 TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=15,pt=6,l=2,sg=0,ft=0,st=OK:0100
3255 RFM95:RCV:SEND ACK
3258 RFM95:SAC:SEND ACK to=0,RSSI=123
3263 TSF:MSG:READ,0-0-10,s=255,c=3,t=15,pt=6,l=2,sg=0:0100
3268 RFM95:SWR:SEND TO=0,RETRY=0
3868 !RFM95:SWR:NACK
3869 RFM95:PTX:LEVEL=21
3872 RFM95:SWR:SEND TO=0,RETRY=1
3991 RFM95:SWR:ACK FROM=0,SEQ=9,RSSI=-16,SNR=9
3995 RFM95:ATC:ADJ TXL,cR=-16,tR=-60,TXL=20
3999 TSF:MSG:SEND,10-10-0-0,s=255,c=0,t=17,pt=0,l=10,sg=0,ft=0,st=OK:2.1.0-beta
4006 RFM95:SWR:SEND TO=0,RETRY=0
4110 RFM95:SWR:ACK FROM=0,SEQ=10,RSSI=-15,SNR=9
4114 RFM95:ATC:ADJ TXL,cR=-15,tR=-60,TXL=19
4118 TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=6,pt=1,l=1,sg=0,ft=0,st=OK:0
6124 RFM95:SWR:SEND TO=0,RETRY=0
6243 RFM95:SWR:ACK FROM=0,SEQ=11,RSSI=-16,SNR=10
6247 RFM95:ATC:ADJ TXL,cR=-16,tR=-60,TXL=18
6251 TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=11,pt=0,l=11,sg=0,ft=0,st=OK:Water Meter
6258 RFM95:SWR:SEND TO=0,RETRY=0
6367 RFM95:SWR:ACK FROM=0,SEQ=12,RSSI=-16,SNR=9
6372 RFM95:ATC:ADJ TXL,cR=-16,tR=-60,TXL=17
6375 TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=12,pt=0,l=3,sg=0,ft=0,st=OK:1.1
6382 RFM95:SWR:SEND TO=0,RETRY=0
6486 RFM95:SWR:ACK FROM=0,SEQ=13,RSSI=-15,SNR=9
6490 RFM95:ATC:ADJ TXL,cR=-15,tR=-60,TXL=16
6494 TSF:MSG:SEND,10-10-0-0,s=1,c=0,t=21,pt=0,l=0,sg=0,ft=0,st=OK:
6500 MCO:REG:REQ
6502 RFM95:SWR:SEND TO=0,RETRY=0
6605 RFM95:SWR:ACK FROM=0,SEQ=14,RSSI=-15,SNR=9
6609 RFM95:ATC:ADJ TXL,cR=-15,tR=-60,TXL=15
6614 TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=26,pt=1,l=1,sg=0,ft=0,st=OK:2
7208 RFM95:RCV:SEND ACK
7212 RFM95:SAC:SEND ACK to=0,RSSI=122
7217 TSF:MSG:READ,0-0-10,s=255,c=3,t=27,pt=1,l=1,sg=0:1
7222 MCO:PIM:NODE REG=1
7224 MCO:BGN:STP
In setup  <------------------we enter setup here, as you can see above we are using id ID=10
7226 MCO:BGN:INIT OK,TSP=1
374986 TSF:MSG:READ,0-0-255,s=255,c=3,t=20,pt=0,l=0,sg=0:
374991 TSF:MSG:BC
375202 RFM95:SWR:SEND TO=0,RETRY=0
375307 RFM95:SWR:ACK FROM=0,SEQ=16,RSSI=-23,SNR=9
375311 RFM95:ATC:ADJ TXL,cR=-23,tR=-60,TXL=14
375315 TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=21,pt=1,l=1,sg=0,ft=0,st=OK:0
975411 TSF:MSG:READ,0-0-255,s=255,c=3,t=20,pt=0,l=0,sg=0:
975416 TSF:MSG:BC
975987 RFM95:SWR:SEND TO=0,RETRY=0
976092 RFM95:SWR:ACK FROM=0,SEQ=17,RSSI=-22,SNR=9
976096 RFM95:ATC:ADJ TXL,cR=-22,tR=-60,TXL=13
976100 TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=21,pt=1,l=1,sg=0,ft=0,st=OK:0
1575835 TSF:MSG:READ,0-0-255,s=255,c=3,t=20,pt=0,l=0,sg=0:
1575840 TSF:MSG:BC
1576772 RFM95:SWR:SEND TO=0,RETRY=0
1576877 RFM95:SWR:ACK FROM=0,SEQ=18,RSSI=-22,SNR=9
1576882 RFM95:ATC:ADJ TXL,cR=-22,tR=-60,TXL=12
1576886 TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=21,pt=1,l=1,sg=0,ft=0,st=OK:0
2176260 TSF:MSG:READ,0-0-255,s=255,c=3,t=20,pt=0,l=0,sg=0:
2176265 TSF:MSG:BC
2176534 RFM95:SWR:SEND TO=0,RETRY=0
2176638 RFM95:SWR:ACK FROM=0,SEQ=19,RSSI=-22,SNR=9
2176643 RFM95:ATC:ADJ TXL,cR=-22,tR=-60,TXL=11
2176647 TSF:MSG:SEND,10-10-0-0,s=255,c=3,t=21,pt=1,l=1,sg=0,ft=0,st=OK:0
2776685 TSF:MSG:READ,0-0-255,s=255,c=3,t=20,pt=0,l=0,sg=0:
2776690 TSF:MSG:BC
2777320 RFM95:SWR:SEND TO=0,RETRY=0

~~~

trlafleur, 2016-12-01:
This is now resolved, one can use the before() function to set node ID...
It was a setup issue on my part... sorry for the confusion...

tekka007, 2016-12-01:
at this point, try using transportAssignNodeID() in before() and report back. Ideally, you also set MY_NODE_ID to AUTO.

closed 2016-12-03 by tekka007