mysensors/MySensors #677: Wrong packet length with enabled encryption

fbeek, 2016-12-02:
Hello,

when i use the latest development version on my two node and enabled encryption i get the following error on the gateway: 

0;255;3;0;9;!TSF:MSG:LEN,16!=7 

Both nodes have the correct AES key (i checked it in with a debug statement) and before the update (last version was from may) everything worked fine.

If i disable the encryption everything is ok.  There i no power issue, because they where running without problems until now and i checked it twice. Signing is working also.

Regards 
fbeek

fallberg, 2016-12-02:
And you have used the newer personalization routine so that the AES key is stored in EEPROM?

fallberg, 2016-12-02:
(I am guessing you have done that if you have signing working, which require the same procedure)

fbeek, 2016-12-02:
Hey.

Yes i used the newer one to personalize the nodes. The key where in the Memory. I added a serial output function to print the keys. They are the same on both devices.

fallberg, 2016-12-02:
I have not used encryption more than testing it once for nrf24 and it worked back then when I moved the keys to eeprom. Which radio are you using?

fbeek, 2016-12-02:
I use two normal NRF24, the ones without pa and lna.

fallberg, 2016-12-03:
Ok, and no relay nodes used?

fallberg, 2016-12-03:
Could you please try to add some debug to determine if the message sent has been encrypted at all or if it even attempts to decrypt it at the receiving end?

fbeek, 2016-12-03:
Good Morning,

there is no relay node in use. As far as I debugged it yesterday, the package gets encrypted on the sender side and decrypted on gateway Side, both with the correct key.

fallberg, 2016-12-03:
We have a clue to what's wrong. Expect some  ore details during the day.

fbeek, 2016-12-03:
Thanks for your help.I stay tuned.

fallberg, 2016-12-03:
@tekka007 has found a problem in the RF24 driver. It's being looked at. Thanks for reporting.

fbeek, 2016-12-03:
Ok. Thanks for your great work on this project!

PR #680 (2016-12-06): Fix payload length verification
- decrypted messages are not processed due a payload length verification error.
- Fixes #677
closed 2016-12-06 by fallberg