mysensors/MySensors #697: No MQTT messages published with current development/beta release

DerTomm, 2016-12-18:
Hi everyone,

as the current PlatformIO beta release of MySensors library (2.1.0-beta) contains a bug (which prevents my firmware from being linked) I choose the latest Github development branch release instead.
Unfortunately using this version my ESP8266 gateway/MQTT client does not publish any MQTT messages anymore. It just connects to the broker and does not send anything afterwards (not even the sketch version).

As workaround I randomly chose an older development commit (4be733e1e63aebc5d036de1a1f73d094708dc1 from October 31st) now and with this version MQTT publishing works fine. In the meantime there were some MQTT related commits - maybe one of them broke the publishing.

tekka007, 2016-12-18:
@DerTomm I'm running the GatewayESP8266MQTTClient sketch with the latest dev branch, Arduino IDE 1.6.13, ESP8266 board defs 2.3.0 and Mosquitto broker - everything up and running. 
Please provide details on your setup, versions and debug log for further troubleshooting.

DerTomm, 2016-12-18:
This is my sketch:

```
#include <Arduino.h>
#include <EEPROM.h>
#include <SPI.h>

ADC_MODE (ADC_TOUT);  // set ADC to "external mode" - commented out ADC_MODE(ADC_VCC) in core/MyHwESP8266.cpp instead

// Enable debug prints to serial monitor
#define MY_DEBUG

// Use a bit lower baudrate for serial prints on ESP8266 than default in MyConfig.h
#define MY_BAUD_RATE 115200

#define MY_NODE_ID 3

#define MY_GATEWAY_MQTT_CLIENT
#define MY_GATEWAY_ESP8266

// Set this node's subscribe and publish topic prefix
#define MY_MQTT_PUBLISH_TOPIC_PREFIX "espOutdoorTemp-out"
#define MY_MQTT_SUBSCRIBE_TOPIC_PREFIX "espOutdoorTemp-in"

// Set MQTT client id
#define MY_MQTT_CLIENT_ID "espOutdoorTemp"

// Set WIFI SSID and password
#define MY_ESP8266_SSID "$WIFISSID"
#define MY_ESP8266_PASSWORD "$WIFIKEY"

// Enable MY_IP_ADDRESS here if you want a static ip address (no DHCP)
#define MY_IP_ADDRESS 192,168,64,150

// If using static ip you need to define Gateway and Subnet address as well
#define MY_IP_GATEWAY_ADDRESS 192,168,64,253
#define MY_IP_SUBNET_ADDRESS 255,255,255,0

// MQTT broker ip address.  
#define MY_CONTROLLER_IP_ADDRESS 192, 168, 64, 253

// The MQTT broker port to to open 
#define MY_PORT 1883      

#include <ESP8266WiFi.h>
#include <MySensors.h>
#include <OneWire.h>
#include <DallasTemperature.h>
#include "PiduinoCommons/Battery/LGICR18650B42600.h"

#define ONE_WIRE_BUS 14

OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature sensors(&oneWire);

#define CHILD_ID_TEMPERATURE        0
MyMessage msgTemperature(CHILD_ID_TEMPERATURE, V_TEMP);

#define BATTERY_VOLTAGE_ADC_PIN A0
#define BATTERY_VOLTAGE_FACTOR 3.73
LGICR18650B42600 battery;

unsigned long sleeptimeS = 30;

/**************************************************************************************************************
 *
 */
double getBatteryVoltage() {

  wifi_set_sleep_type (NONE_SLEEP_T);  // disable radio sleeping for more accurate ADC readings
  delay(50);

  int countAdc = system_adc_read();
  return ((countAdc * 1.1) / 1023.0) * BATTERY_VOLTAGE_FACTOR;
}

/**************************************************************************************************************
 *
 */
void setup() {

  pinMode(BATTERY_VOLTAGE_ADC_PIN, INPUT);
  sensors.begin();
}

/**************************************************************************************************************
 *
 */
void presentation() {

  // Send the Sketch Version Information to the Gateway
  sendSketchInfo("ESP8266OutdoorTempSensor", "1.0");

  // Register all sensors to gw (they will be created as child devices)
  present(CHILD_ID_TEMPERATURE, S_TEMP);
}

/**************************************************************************************************************
 *
 */
void loop() {

  double voltage = getBatteryVoltage();
  int stateOfCharge = battery.getStateOfCharge(voltage);
  sendBatteryLevel(stateOfCharge);

  sensors.requestTemperatures(); // Send the command to get temperatures
  delay(50);
  send(msgTemperature.set(sensors.getTempCByIndex(0), 1));

  ESP.deepSleep(sleeptimeS * 1000 * 1000, RFMode::RF_DEFAULT);
}
```

I am using the current PlatformIO release 3.2.1 as development environment and provide the MySensors Github development snapshot as library in the "lib_extra_dirs" folder.

As said the built firmware works fine with the 10/31 snapshot but does not publish any MQTT messages with todays dev branch snapshot.

tekka007, 2016-12-18:
@DerTomm A gateway cannot sleep:
`ESP.deepSleep(sleeptimeS * 1000 * 1000, RFMode::RF_DEFAULT);`

DerTomm, 2016-12-18:
In this case the "gateway" is only a sensor node with one DS18B20 temperature sensor waking up every 5 minutes (30s in the example above) to get the current temperature and send this value to the broker.
As far as I see there is no dedicated way of defining a pure "ESP8266MQTTClientSensorNode" yet.

Besides this - the question I am asking me is: Why is the old snapshot working properly and the current one not?

tekka007, 2016-12-18:
@DerTomm Please post the serial debug log (for the 10/31 snapshot and the latest dev), this may give additional hints.

DerTomm, 2016-12-18:
Yes, I'll post the node's MY_DEBUG and Mosquitto's log output for both versions tomorrow evening.

DerTomm, 2016-12-19:
As promised, here are the debug outputs.

Let's begin with the **current development snapshot (2016/12/19)**:

_ESP8266 sensor node output:_
```
0;255;3;0;9;MCO:BGN:INIT GW,CP=R-NGE--,VER=2.1.0-beta
scandone
f 0, ....scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt

connected with FoobarNet, channel 3
ip:192.168.64.150,mask:255.255.255.0,gw:192.168.64.253
.IP: 192.168.64.150

0;255;3;0;9;MCO:BGN:STP
0;255;3;0;9;MCO:BGN:INIT OK,TSP=NA
IP: 192.168.64.150

0;255;3;0;9;Attempting MQTT connection...
0;255;3;0;9;MQTT connected
sleep disable
state: 5 -> 0 (0)
rm 0
del if0
usl

/*
ESP.deepSleep() for 10 seconds with following unit wakeup/reset
*/

0;255;3;0;9;MCO:BGN:INIT GW,CP=R-NGE--,VER=2.1.0-beta
scandone
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt

connected with FoobarNet, channel 3
ip:192.168.64.150,mask:255.255.255.0,gw:192.168.64.253
.IP: 192.168.64.150

0;255;3;0;9;MCO:BGN:STP
0;255;3;0;9;MCO:BGN:INIT OK,TSP=NA
IP: 192.168.64.150

0;255;3;0;9;Attempting MQTT connection...
0;255;3;0;9;MQTT connected
sleep disable
state: 5 -> 0 (0)
rm 0
del if0
usl

/*
and so on...
*/
```

_Mosquitto MQTT broker output:_
```
1482172948: mosquitto version 1.4.8 (build date Fri, 19 Feb 2016 12:03:16 +0100) starting
[... 113 lines cut ...]
1482173703:     espOutdoorTemp-in/+/+/+/+/+ (QoS 0)
1482173703: espOutdoorTemp 0 espOutdoorTemp-in/+/+/+/+/+
1482173703: Sending SUBACK to espOutdoorTemp
1482173703: Received PUBLISH from espOutdoorTemp (d0, q0, r0, m0, 'espOutdoorTemp-out/0/255/3/0/0', ... (2 bytes))
1482173704: Received PUBLISH from espOutdoorTemp (d0, q0, r0, m0, 'espOutdoorTemp-out/0/0/1/0/0', ... (4 bytes))
1482173725: Client espOutdoorTemp has exceeded timeout, disconnecting.
1482173725: Socket error on client espOutdoorTemp, disconnecting.
1482173734: New connection from 192.168.64.150 on port 1883.
1482173734: New client connected from 192.168.64.150 as espOutdoorTemp (c1, k15).
1482173734: Sending CONNACK to espOutdoorTemp (0, 0)
1482173734: Received PUBLISH from espOutdoorTemp (d0, q0, r0, m0, 'espOutdoorTemp-out/0/255/0/0/17', ... (10 bytes))
1482173734: Received PUBLISH from espOutdoorTemp (d0, q0, r0, m0, 'espOutdoorTemp-out/0/255/3/0/11', ... (24 bytes))
1482173734: Received PUBLISH from espOutdoorTemp (d0, q0, r0, m0, 'espOutdoorTemp-out/0/255/3/0/12', ... (3 bytes))
1482173734: Received PUBLISH from espOutdoorTemp (d0, q0, r0, m0, 'espOutdoorTemp-out/0/0/0/0/6', ... (0 bytes))
1482173734: Received SUBSCRIBE from espOutdoorTemp
1482173734:     espOutdoorTemp-in/+/+/+/+/+ (QoS 0)
1482173734: espOutdoorTemp 0 espOutdoorTemp-in/+/+/+/+/+
1482173734: Sending SUBACK to espOutdoorTemp
1482173734: Received PUBLISH from espOutdoorTemp (d0, q0, r0, m0, 'espOutdoorTemp-out/0/255/3/0/0', ... (2 bytes))
1482173735: Received PUBLISH from espOutdoorTemp (d0, q0, r0, m0, 'espOutdoorTemp-out/0/0/1/0/0', ... (4 bytes))
1482173756: Client espOutdoorTemp has exceeded timeout, disconnecting.
1482173756: Socket error on client espOutdoorTemp, disconnecting.
```

The test setup and the code is exactly the same in both runs. The only difference is the used MySensors library.

tekka007, 2016-12-19:
@DerTomm Thanks for your feedback, bug has been isolated and PR in preparation.

closed 2016-12-19 by henrikekblad

DerTomm, 2016-12-20:
Thanks for the prompt analysis and fix - great support for this also great project :-)

PR #698 (2016-12-19): Fix bug preventing GW-attached sensors to send messages
Bug: GW-attached sensors cannot send messages if no radio-network defined.
- Fixes #697