mysensors/MySensors #700: Data transport size

trlafleur, 2016-12-22:
"The maximum payload size is 25 bytes! The NRF24L01+ has a maximum of 32 bytes. The MySensors library (version 2.0) uses 7 bytes for the message header"

This should be determined by the radio or transport capability, not by the lowest available size.
This should be setable in the configuration file to set max size for the network based on the radio transport used.

A future option might be to allow a limited number "multiple packets to be sent per message"

sle118, 2016-12-22:
I was just about to create a similar issue and @trlafleur  best me to it. 

Messages would not be limited to 25 characters; the transport layer should implement a rudimentary protocol which breaks messages in smaller packets, similar to how tcpip does it. I think having a size limit might be a good idea on Arduino as resources are limited and buffers could overflow.

My requirement is to be able to implement an IR blaster that can send and receive full length pronto codes from the controller rather than playing back stored codes. The site below has a complete API which would be easy to pull from/contribute to. 

[http://irdb.tk](url)

Bigger payloads would open up the door to much more than this!

fallberg, 2016-12-23:
Sounds like a great idea! Please feel free to file a pull request with this implemented, but please also mind that message signing needs to handle this variable payload size as well.