mysensors/MySensors #715: SAMD WDT and Sleep support

trlafleur, 2017-01-01:
As the SAMD processors are gaining major use and support, adding the missing support for the WDT and Sleep functions in MySensor would be very helpful.

The current SAMD IDE board core code ver 1.6.11, support for sleep-wakeup is incomplete. Their is an issue with how to sleep the processor and allow edge trigger of interrupt to wake the processor. Level interrupts work fine...

There are some note's on how to make this work in the thread:
~~~
https://github.com/arduino/ArduinoCore-samd/issues/142#issuecomment-223061918
~~~

Thanks

d00616, 2017-08-26:
@mfalkvidd The https://github.com/arduino-libraries/ArduinoLowPower supports low power modes for SAMD.

sapdemon, 2018-10-31:
Some more information on low power
http://www.rocketscream.com/blog/docs-item/using-rtc-in-low-power-mode-on-mini-ultra-pro/#tab-id-2
It uses RTCZero Arduino lib