mysensors/MySensors #719: Add support for SPIDEV

eyesoftsys, 2017-01-04:
I'm trying to build MySensor Gateway on Banana Pi Pro and failed with :

./MySensors.h:258:2: error: #error No support for nRF24 radio on this platform
 #error No support for nRF24 radio on this platform
 #error No support for nRF24 radio on this platform
  ^
In file included from ./drivers/RF24/RF24.cpp:23:0,
                 from ./MySensors.h:294,
                 from examples_linux/mysgw.cpp:74:
./drivers/RF24/RF24.h:52:17: fatal error: SPI.h: No such file or directory
 #include <SPI.h>
                 ^
compilation terminated.
Makefile:98: recipe for target 'build/examples_linux/mysgw.o' failed
make: *** [build/examples_linux/mysgw.o] Error 1.

i see that RF24.h -> Based on maniacbug's RF24 library, copyright (C) 2011 J. Coliz <maniacbug@ymail.com>

when i compile https://github.com/TMRh20/RF24 i get :
root@bananapi:~/RF24# ./configure
[SECTION] Detecting arm compilation environment.
  [OK] arm-linux-gnueabihf-gcc detected.
  [OK] arm-linux-gnueabihf-g++ detected.
[SECTION] Detecting target machine.
[OK] machine detected: SoC=A20, Type=unknown, CPU=armv7l.
[SECTION] Detecting DRIVER
  [OK] DRIVER **detected:SPIDEV**.
[SECTION] Detecting OS.
  [INFO] OS detected:LINUX.
[SECTION] Preparing configuration.
[SECTION] Saving configuration.
[SECTION] Cleaning previous builds.
[OK] Finished.

arm-linux-gnueabihf-g++ -fPIC -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -Ofast -Wall -pthread  -c RF24.cpp
arm-linux-gnueabihf-g++ -fPIC -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -Ofast -Wall -pthread  -c utility/SPIDEV/spi.cpp
arm-linux-gnueabihf-g++ -fPIC -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -Ofast -Wall -pthread  -c utility/SPIDEV/gpio.cpp
arm-linux-gnueabihf-gcc -fPIC  -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -Ofast -Wall -pthread  -c utility/SPIDEV/compatibility.c
arm-linux-gnueabihf-g++ -fPIC -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -Ofast -Wall -pthread  -c utility/SPIDEV/interrupt.c
[Linking]
arm-linux-gnueabihf-gcc -pthread -shared -Wl,-soname,librf24.so.1 -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -Ofast -Wall -pthread  -o librf24.so.1.2.0 RF24.o spi.o gpio.o compatibility.o interrupt.o

thank you

aktentasche, 2017-01-04:
Hi, @marceloaqno has done a fork to incorporate SPIDEV: https://github.com/marceloaqno/MySensors/commit/8ae05ac79b5b603754350fe0ac8b96b0ad2ab9b7

Stay tuned I guess?

eyesoftsys, 2017-01-04:
yes, i see here some work https://github.com/marceloaqno/MySensors/tree/spidev thanks to  marceloaqno.

git clone https://github.com/marceloaqno/MySensors.git marceloaqno-spidev
git checkout spidev
Branch spidev set up to track remote branch spidev from origin.
Switched to a new branch 'spidev'

./configure --my-gateway=ethernet  --my-transport=nrf24 --my-rf24-irq-pin=15 --my-rf24-channel=119
[SECTION] Detecting target machine.
  [OK] machine detected: SoC=A20, Type=unknown, CPU=armv7l.
[SECTION] Detecting GPIO driver.
  [OK] GPIO driver detected:SPIDEV.
[SECTION] Detecting init system.
  [OK] init system detected: sysvinit.
[SECTION] Saving configuration.
[SECTION] Cleaning previous builds.
[OK] Finished.

 make
gcc -MT build/drivers/Linux/log.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV -DMY_RF24_CHANNEL=119 -DMY_RX_MESSAGE_BUFFER_FEATURE -DMY_RF24_IRQ_PIN=15  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/log.c -o build/drivers/Linux/log.o
g++ -MT build/drivers/Linux/noniso.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV -DMY_RF24_CHANNEL=119 -DMY_RX_MESSAGE_BUFFER_FEATURE -DMY_RF24_IRQ_PIN=15  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/noniso.cpp -o build/drivers/Linux/noniso.o
g++ -MT build/drivers/Linux/Print.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV -DMY_RF24_CHANNEL=119 -DMY_RX_MESSAGE_BUFFER_FEATURE -DMY_RF24_IRQ_PIN=15  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/Print.cpp -o build/drivers/Linux/Print.o
g++ -MT build/drivers/Linux/EthernetClient.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV -DMY_RF24_CHANNEL=119 -DMY_RX_MESSAGE_BUFFER_FEATURE -DMY_RF24_IRQ_PIN=15  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/EthernetClient.cpp -o build/drivers/Linux/EthernetClient.o
g++ -MT build/drivers/Linux/SerialPort.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV -DMY_RF24_CHANNEL=119 -DMY_RX_MESSAGE_BUFFER_FEATURE -DMY_RF24_IRQ_PIN=15  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/SerialPort.cpp -o build/drivers/Linux/SerialPort.o
g++ -MT build/drivers/Linux/Stream.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV -DMY_RF24_CHANNEL=119 -DMY_RX_MESSAGE_BUFFER_FEATURE -DMY_RF24_IRQ_PIN=15  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/Stream.cpp -o build/drivers/Linux/Stream.o
In file included from drivers/Linux/Stream.cpp:24:0:
./drivers/Linux/Arduino.h:17:18: fatal error: GPIO.h: No such file or directory
 #include "GPIO.h"
                  ^
compilation terminated.
Makefile:114: recipe for target 'build/drivers/Linux/Stream.o' failed
make: *** [build/drivers/Linux/Stream.o] Error 1

marceloaqno, 2017-01-04:
It's still a work in progress, but with the latest version (https://github.com/marceloaqno/MySensors/tree/spidev) I was able to use SPIDEV in the RPi, I had to set the CE and CSN manually to the gpio number (not the physical number), you might need to do this in BananaPi.

If someone wants to test, this is how:
examples_linux/mysgw.cpp
#define MY_RF24_CS_PIN 8  // Choose the appropriate pin for your setup
#define MY_RF24_CE_PIN 25 // Choose the appropriate pin for your setup

./configure --driver=SPIDEV
make
sudo ./bin/mysgw -d

eyesoftsys, 2017-01-04:
tell me if it's on to output here, for now on your version 1.4.1-1243-g67de16a on Banana Pi

./configure --driver=SPIDEV
[SECTION] Detecting target machine.
  [OK] machine detected: SoC=A20, Type=unknown, CPU=armv7l.
[SECTION] Detecting init system.
  [OK] init system detected: sysvinit.
[SECTION] Saving configuration.
[SECTION] Cleaning previous builds.
[OK] Finished.

gcc -MT build/drivers/Linux/log.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/log.c -o build/drivers/Linux/log.o
g++ -MT build/drivers/Linux/noniso.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/noniso.cpp -o build/drivers/Linux/noniso.o
g++ -MT build/drivers/Linux/Print.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/Print.cpp -o build/drivers/Linux/Print.o
g++ -MT build/drivers/Linux/EthernetClient.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/EthernetClient.cpp -o build/drivers/Linux/EthernetClient.o
g++ -MT build/drivers/Linux/SerialPort.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/SerialPort.cpp -o build/drivers/Linux/SerialPort.o
g++ -MT build/drivers/Linux/Stream.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/Stream.cpp -o build/drivers/Linux/Stream.o
g++ -MT build/drivers/Linux/IPAddress.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/IPAddress.cpp -o build/drivers/Linux/IPAddress.o
g++ -MT build/drivers/Linux/compatibility.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/compatibility.cpp -o build/drivers/Linux/compatibility.o
g++ -MT build/drivers/Linux/SoftEeprom.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/SoftEeprom.cpp -o build/drivers/Linux/SoftEeprom.o
g++ -MT build/drivers/Linux/EthernetServer.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/EthernetServer.cpp -o build/drivers/Linux/EthernetServer.o
g++ -MT build/examples_linux/mysgw.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c examples_linux/mysgw.cpp -o build/examples_linux/mysgw.o
In file included from ./MySensors.h:72:0,
                 from examples_linux/mysgw.cpp:75:
./core/MyHwLinuxGeneric.cpp:124:11: error: Not supported on linux-generic
 void hwDigitalWrite(uint8_t pin, uint8_t value)
           ^
./core/MyHwLinuxGeneric.cpp:129:11: error: Not supported on linux-generic
 int hwDigitalRead(uint8_t pin)
           ^
./core/MyHwLinuxGeneric.cpp:134:11: error: Not supported on linux-generic
 void hwPinMode(uint8_t pin, uint8_t mode)
           ^
In file included from ./MySensors.h:294:0,
                 from examples_linux/mysgw.cpp:75:
./drivers/RF24/RF24.cpp:39:11: error: Not supported on linux-generic
  hwDigitalWrite(MY_RF24_CS_PIN, level);
           ^
./drivers/RF24/RF24.cpp:44:11: error: Not supported on linux-generic
  hwDigitalWrite(MY_RF24_CE_PIN, level);
           ^
./drivers/RF24/RF24.cpp:388:11: error: Not supported on linux-generic
  hwPinMode(MY_RF24_CE_PIN,OUTPUT);
           ^
./drivers/RF24/RF24.cpp:389:11: error: Not supported on linux-generic
  hwPinMode(MY_RF24_CS_PIN,OUTPUT);
           ^
In file included from ./MySensors.h:72:0,
                 from examples_linux/mysgw.cpp:75:
./core/MyHwLinuxGeneric.cpp:125:1: error: expected unqualified-id before ‘{’ token
 {
 ^
./core/MyHwLinuxGeneric.cpp:130:1: error: expected unqualified-id before ‘{’ token
 {
 ^
./core/MyHwLinuxGeneric.cpp:135:1: error: expected unqualified-id before ‘{’ token
 {
 ^
In file included from ./MySensors.h:294:0,
                 from examples_linux/mysgw.cpp:75:
./drivers/RF24/RF24.cpp:37:32: warning: unused parameter ‘level’ [-Wunused-parameter]
 LOCAL void RF24_csn(const bool level)
                                ^
./drivers/RF24/RF24.cpp:42:31: warning: unused parameter ‘level’ [-Wunused-parameter]
 LOCAL void RF24_ce(const bool level)
                               ^
Makefile:114: recipe for target 'build/examples_linux/mysgw.o' failed
make: *** [build/examples_linux/mysgw.o] Error 1

marceloaqno, 2017-01-05:
sorry, it should compile now.

eyesoftsys, 2017-01-05:
./configure  --driver=SPIDEV
[SECTION] Detecting target machine.
  [OK] machine detected: SoC=A20, Type=unknown, CPU=armv7l.
[SECTION] Detecting init system.
  [OK] init system detected: sysvinit.
[SECTION] Saving configuration.
[SECTION] Cleaning previous builds.
[OK] Finished.

gcc -MT build/drivers/Linux/log.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV -DLINUX_GPIO_CAPABLE  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/log.c -o build/drivers/Linux/log.o
g++ -MT build/drivers/Linux/noniso.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV -DLINUX_GPIO_CAPABLE  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/noniso.cpp -o build/drivers/Linux/noniso.o
g++ -MT build/drivers/Linux/Print.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV -DLINUX_GPIO_CAPABLE  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/Print.cpp -o build/drivers/Linux/Print.o
g++ -MT build/drivers/Linux/EthernetClient.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV -DLINUX_GPIO_CAPABLE  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/EthernetClient.cpp -o build/drivers/Linux/EthernetClient.o
g++ -MT build/drivers/Linux/SerialPort.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV -DLINUX_GPIO_CAPABLE  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/SerialPort.cpp -o build/drivers/Linux/SerialPort.o
g++ -MT build/drivers/Linux/Stream.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV -DLINUX_GPIO_CAPABLE  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/Stream.cpp -o build/drivers/Linux/Stream.o
g++ -MT build/drivers/Linux/IPAddress.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV -DLINUX_GPIO_CAPABLE  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/IPAddress.cpp -o build/drivers/Linux/IPAddress.o
g++ -MT build/drivers/Linux/compatibility.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV -DLINUX_GPIO_CAPABLE  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/compatibility.cpp -o build/drivers/Linux/compatibility.o
g++ -MT build/drivers/Linux/SoftEeprom.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV -DLINUX_GPIO_CAPABLE  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/SoftEeprom.cpp -o build/drivers/Linux/SoftEeprom.o
g++ -MT build/drivers/Linux/EthernetServer.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV -DLINUX_GPIO_CAPABLE  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/Linux/EthernetServer.cpp -o build/drivers/Linux/EthernetServer.o
g++ -MT build/examples_linux/mysgw.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV -DLINUX_GPIO_CAPABLE  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c examples_linux/mysgw.cpp -o build/examples_linux/mysgw.o
examples_linux/mysgw.cpp:75:0: warning: "MY_RF24_CS_PIN" redefined
 #define MY_RF24_CS_PIN 24
 ^
In file included from ./core/MySensorsCore.h:71:0,
                 from ./MySensors.h:28,
                 from examples_linux/mysgw.cpp:74:
./MyConfig.h:514:0: note: this is the location of the previous definition
 #define MY_RF24_CS_PIN 10
 ^
examples_linux/mysgw.cpp:76:0: warning: "MY_RF24_CE_PIN" redefined
 #define MY_RF24_CE_PIN 22
 ^
In file included from ./core/MySensorsCore.h:71:0,
                 from ./MySensors.h:28,
                 from examples_linux/mysgw.cpp:74:
./MyConfig.h:498:0: note: this is the location of the previous definition
 #define MY_RF24_CE_PIN 9
 ^
g++ -MT build/drivers/SPIDEV/GPIO.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV -DLINUX_GPIO_CAPABLE  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/SPIDEV/GPIO.cpp -o build/drivers/SPIDEV/GPIO.o
g++ -MT build/drivers/SPIDEV/SPI.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG -DLINUX_DRIVER_SPIDEV -DLINUX_GPIO_CAPABLE  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/SPIDEV -c drivers/SPIDEV/SPI.cpp -o build/drivers/SPIDEV/SPI.o
[... cut ...]
after i delete from examples_linux/mysgw.cpp MY_RF24_CE_PIN and MY_RF24_CS_PIN  lines and modify MyConfig.h with MY_RF24_CE_PIN and MY_RF24_CS_PIN   with right values i have no compilation errors.

mysgw: Starting gateway...
mysgw: Protocol version - 2.2.0-beta
mysgw: MCO:BGN:INIT GW,CP=RNNG---,VER=2.2.0-beta
mysgw: TSM:INIT
mysgw: TSF:WUR:MS=0
mysgw: !TSM:INIT:TSP FAIL
mysgw: TSM:FAIL:CNT=1
mysgw: TSM:FAIL:PDT
mysgw: TSM:FAIL:RE-INIT
mysgw: TSM:INIT
mysgw: !TSM:INIT:TSP FAIL
mysgw: TSM:FAIL:CNT=3
mysgw: TSM:FAIL:PDT

marceloaqno, 2017-01-14:
#734

marceloaqno, 2017-01-14:
#734

eyesoftsys, 2017-01-15:
thank you  marceloaqno

closed 2017-01-15 by marceloaqno