mysensors/MySensors #720: RF24 module not working with release 2.1.0 and 2.2.0-development

DerTomm, 2017-01-05:
Hi,

today I updated MySensors in my serial gateway from an older 2.1.0-beta release to the final 2.1.0 release. But then I encountered a problem. My serial gateway does not start anymore.

This is the output with release 2.1.0:
```
0;255;3;0;9;MCO:BGN:INIT GW,CP=RNNGA--,VER=2.1.0
0;255;3;0;9;TSM:INIT
0;255;3;0;9;TSF:WUR:MS=0
0;255;3;0;9;!TSM:INIT:TSP FAIL
0;255;3;0;9;TSM:FAIL:CNT=1
0;255;3;0;9;TSM:FAIL:PDT
0;255;3;0;9;TSM:FAIL:RE-INIT
0;255;3;0;9;TSM:INIT
0;255;3;0;9;!TSM:INIT:TSP FAIL
0;255;3;0;9;TSM:FAIL:CNT=2
0;255;3;0;9;TSM:FAIL:PDT
0;255;3;0;9;MCO:BGN:INIT GW,CP=RNNGA--,VER=2.1.0
0;255;3;0;9;TSM:INIT
0;255;3;0;9;TSF:WUR:MS=0
0;255;3;0;9;!TSM:INIT:TSP FAIL
0;255;3;0;9;TSM:FAIL:CNT=1
0;255;3;0;9;TSM:FAIL:PDT
0;255;3;0;9;TSM:FAIL:RE-INIT
0;255;3;0;9;TSM:INIT
0;255;3;0;9;!TSM:INIT:TSP FAIL
0;255;3;0;9;TSM:FAIL:CNT=2
0;255;3;0;9;TSM:FAIL:PDT
(and so on - bootloop)
```
The behaviour is completely different with the older 2.0.0 final release and 2.1.0 beta-releases:
```
0;255;3;0;9;Starting gateway (RNNGA-, 2.0.0)
0;255;3;0;9;TSM:INIT
0;255;3;0;9;TSM:RADIO:OK
0;255;3;0;9;TSM:GW MODE
0;255;3;0;9;TSM:READY
0;255;3;0;14;Gateway startup complete.
0;255;0;0;18;2.0.0
0;255;3;0;11;MySensors SerialGateway a
0;255;3;0;12;1.5
0;0;0;0;6;
0;1;0;0;1;
0;2;0;0;26;
0;61;0;0;23;
0;62;0;0;23;
0;255;3;0;9;No registration required
0;255;3;0;9;Init complete, id=0, parent=0, distance=0, registration=1
0;0;1;0;0;19.9
0;1;1;0;16;1
0;255;3;0;9;TSP:MSG:READ 1-1-255 s=255,c=3,t=7,pt=0,l=0,sg=0:
0;255;3;0;9;TSP:MSG:BC
0;255;3;0;9;TSP:MSG:FPAR REQ (sender=1)
0;255;3;0;9;TSP:CHKUPL:OK
0;255;3;0;9;TSP:MSG:GWL OK
0;255;3;0;9;TSP:MSG:SEND 0-0-1-1 s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=ok:0
0;255;3;0;9;TSP:MSG:READ 1-1-0 s=255,c=3,t=24,pt=1,l=1,sg=0:1
0;255;3;0;9;TSP:MSG:PINGED (ID=1, hops=1)
0;255;3;0;9;TSP:MSG:SEND 0-0-1-1 s=255,c=3,t=25,pt=1,l=1,sg=0,ft=0,st=ok:1
...
```

My sketch is derived from SerialGateway example with two presentations and these additional config definements:
```
#define MY_RF24_CE_PIN  18
#define MY_RF24_CS_PIN  19
#define MY_RF24_CHANNEL  83
```

The used environment:
uC: ATmega 1284p with Arduino bootloader
RF24: Chinese nRF24L01+ module
Compiler: PlatformIO v3.2.1 and Arduino 1.6.13 (same behaviour with both)

tekka007, 2017-01-05:
@DerTomm Please post your sketch and resubmit both debug logs with MY_DEBUG_VERBOSE_RF24 enabled.

DerTomm, 2017-01-05:
Debug output with 2.0.0-final release and exactly the same sketch:

```
0;255;3;0;9;Starting gateway (RNNGA-, 2.0.0)
0;255;3;0;9;TSM:INIT
0;255;3;0;9;RF24:write register, reg=0, value=14
0;255;3;0;9;RF24:write register, reg=3, value=3
0;255;3;0;9;RF24:write register, reg=4, value=95
0;255;3;0;9;RF24:write register, reg=5, value=83
0;255;3;0;9;RF24:write register, reg=6, value=39
0;255;3;0;9;RF24:read register, reg=6, value=39
0;255;3;0;9;RF24:read register, reg=5, value=83
0;255;3;0;9;RF24:write register, reg=16, value=115
0;255;3;0;9;RF24:write register, reg=29, value=6
0;255;3;0;9;RF24:write register, reg=2, value=2
0;255;3;0;9;RF24:write register, reg=1, value=0
0;255;3;0;9;RF24:write register, reg=28, value=3
0;255;3;0;9;RF24:flushRX
0;255;3;0;9;RF24:flushTX
0;255;3;0;9;RF24:write register, reg=7, value=112
0;255;3;0;9;TSM:RADIO:OK
0;255;3;0;9;TSM:GW MODE
0;255;3;0;9;RF24:write register, reg=2, value=3
0;255;3;0;9;RF24:write register, reg=1, value=1
0;255;3;0;9;RF24:start listening
0;255;3;0;9;RF24:write register, reg=0, value=15
0;255;3;0;9;RF24:write register, reg=10, value=0
0;255;3;0;9;TSM:READY
0;255;3;0;14;Gateway startup complete.
0;255;0;0;18;2.0.0
0;255;3;0;9;No registration required
0;255;3;0;9;Init complete, id=0, parent=0, distance=0, registration=1
0;255;3;0;9;RF24:read message, len=7
0;255;3;0;9;RF24:write register, reg=7, value=64
0;255;3;0;9;TSP:MSG:READ 1-1-255 s=255,c=3,t=7,pt=0,l=0,sg=0:
0;255;3;0;9;TSP:MSG:BC
0;255;3;0;9;TSP:MSG:FPAR REQ (sender=1)
0;255;3;0;9;TSP:CHKUPL:OK (FLDCTRL)
0;255;3;0;9;TSP:MSG:GWL OK
0;255;3;0;9;RF24:stop listening
0;255;3;0;9;RF24:write register, reg=0, value=14
0;255;3;0;9;RF24:open writing pipe, recipient=1
0;255;3;0;9;RF24:write register, reg=10, value=1
0;255;3;0;9;RF24:write register, reg=16, value=1
0;255;3;0;9;RF24:send message to 1, len=8
0;255;3;0;9;RF24:flushTX
0;255;3;0;9;RF24:write register, reg=7, value=48
0;255;3;0;9;RF24:start listening
0;255;3;0;9;RF24:write register, reg=0, value=15
0;255;3;0;9;RF24:write register, reg=10, value=0
0;255;3;0;9;TSP:MSG:SEND 0-0-1-1 s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=ok:0
0;255;3;0;9;RF24:read message, len=8
0;255;3;0;9;RF24:write register, reg=7, value=64
0;255;3;0;9;TSP:MSG:READ 1-1-0 s=255,c=3,t=24,pt=1,l=1,sg=0:1
0;255;3;0;9;TSP:MSG:PINGED (ID=1, hops=1)
0;255;3;0;9;RF24:stop listening
0;255;3;0;9;RF24:write register, reg=0, value=14
0;255;3;0;9;RF24:open writing pipe, recipient=1
0;255;3;0;9;RF24:write register, reg=10, value=1
0;255;3;0;9;RF24:write register, reg=16, value=1
0;255;3;0;9;RF24:send message to 1, len=8
0;255;3;0;9;RF24:flushTX
0;255;3;0;9;RF24:write register, reg=7, value=48
0;255;3;0;9;RF24:start listening
0;255;3;0;9;RF24:write register, reg=0, value=15
0;255;3;0;9;RF24:write register, reg=10, value=0
0;255;3;0;9;TSP:MSG:SEND 0-0-1-1 s=255,c=3,t=25,pt=1,l=1,sg=0,ft=0,st=ok:1
```

tekka007, 2017-01-05:
@DerTomm Thanks, can you try this:

Substitute these three lines to:
https://github.com/mysensors/MySensors/blob/master/core/MyHwATMega328.h#L62-L64

```
#define hwDigitalWrite(__pin, __value) digitalWrite(__pin, __value)
#define hwDigitalRead(__pin) digitalRead(__pin)
#define hwPinMode(__pin, __value) pinMode(__pin, __value)
```
and test again?

DerTomm, 2017-01-05:
No problem :-)

Looks a little bit better - but then it seems to be caught in a loop:
Output:

```
0;255;3;0;9;MCO:BGN:INIT GW,CP=RNNGA--,VER=2.2.0-beta
0;255;3;0;9;TSM:INIT
0;255;3;0;9;TSF:WUR:MS=0
0;255;3;0;9;RF24:write register, reg=0, value=14
0;255;3;0;9;RF24:write register, reg=3, value=3
0;255;3;0;9;RF24:write register, reg=4, value=95
0;255;3;0;9;RF24:write register, reg=5, value=83
0;255;3;0;9;RF24:write register, reg=6, value=39
0;255;3;0;9;RF24:write register, reg=16, value=115
0;255;3;0;9;RF24:write register, reg=29, value=6
0;255;3;0;9;RF24:read register, reg=6, value=39
0;255;3;0;9;RF24:read register, reg=5, value=83
0;255;3;0;9;RF24:write register, reg=2, value=2
0;255;3;0;9;RF24:write register, reg=1, value=0
0;255;3;0;9;RF24:write register, reg=28, value=3
0;255;3;0;9;RF24:flushRX
0;255;3;0;9;RF24:flushTX
0;255;3;0;9;RF24:write register, reg=7, value=112
0;255;3;0;9;TSM:INIT:TSP OK
0;255;3;0;9;TSM:INIT:GW MODE
0;255;3;0;9;RF24:write register, reg=2, value=3
0;255;3;0;9;RF24:write register, reg=1, value=1
0;255;3;0;9;RF24:STRT LIS
0;255;3;0;9;RF24:write register, reg=0, value=15
0;255;3;0;9;RF24:write register, reg=10, value=0
0;255;3;0;9;TSM:READY:ID=0,PAR=0,DIS=0
0;255;3;0;9;MCO:REG:NOT NEEDED
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;14;Gateway startup complete.
0;255;0;0;18;2.2.0-beta
0;255;3;0;9;MCO:BGN:STP
0;255;3;0;9;MCO:BGN:INIT OK,TSP=1
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
0;255;3;0;9;RF24:read register, reg=23, value=17
<last line appears over and over again>
```

tekka007, 2017-01-05:
@DerTomm ok, I'll prepare a PR and would be glad if you could test it and provide feedback.
The "loop" is the register polling for incoming messages, nothing unusal :)

DerTomm, 2017-01-05:
Sounds great - up to then I use the patched lib as hotfix. Thanks for this great support!

tekka007, 2017-01-05:
@DerTomm Did you already manage to test this PR?

DerTomm, 2017-01-06:
Unfortunately I had to finish the gateway yesterday for a weekend test - so the next chance to test the PR is beginning of next week. I'm going to test it as soon as I build the next firmware revision for my gateway.

DerTomm, 2017-01-08:
Today I could test the PR. As it seems the change does not fix the problem. Here is the log output after applying the PR:

```
0;255;3;0;9;MCO:BGN:INIT GW,CP=RNNGA--,VER=2.1.0
0;255;3;0;9;TSM:INIT
0;255;3;0;9;TSF:WUR:MS=0
0;255;3;0;9;RF24:write register, reg=0, value=14
0;255;3;0;9;RF24:write register, reg=3, value=3
0;255;3;0;9;RF24:write register, reg=4, value=95
0;255;3;0;9;RF24:write register, reg=5, value=83
0;255;3;0;9;RF24:write register, reg=6, value=39
0;255;3;0;9;RF24:write register, reg=16, value=115
0;255;3;0;9;RF24:write register, reg=29, value=6
0;255;3;0;9;RF24:read register, reg=6, value=255
0;255;3;0;9;RF24:read register, reg=5, value=255
0;255;3;0;9;!RF24:INIT:SANCHK FAIL
0;255;3;0;9;!TSM:INIT:TSP FAIL
0;255;3;0;9;TSM:FAIL:CNT=1
0;255;3;0;9;TSM:FAIL:PDT
0;255;3;0;9;RF24:write register, reg=0, value=12
0;255;3;0;9;RF24:PD
0;255;3;0;9;TSM:FAIL:RE-INIT
0;255;3;0;9;TSM:INIT
0;255;3;0;9;RF24:write register, reg=0, value=14
0;255;3;0;9;RF24:write register, reg=3, value=3
0;255;3;0;9;RF24:write register, reg=4, value=95
0;255;3;0;9;RF24:write register, reg=5, value=83
0;255;3;0;9;RF24:write register, reg=6, value=39
0;255;3;0;9;RF24:write register, reg=16, value=115
0;255;3;0;9;RF24:write register, reg=29, value=6
0;255;3;0;9;RF24:read register, reg=6, value=255
0;255;3;0;9;RF24:read register, reg=5, value=255
0;255;3;0;9;!RF24:INIT:SANCHK FAIL
0;255;3;0;9;!TSM:INIT:TSP FAIL
0;255;3;0;9;TSM:FAIL:CNT=2
0;255;3;0;9;TSM:FAIL:PDT
0;255;3;0;9;RF24:write register, reg=0, value=12
0;255;3;0;9;RF24:PD
...
```

tekka007, 2017-01-08:
@DerTomm What board settings are you using to compile?

DerTomm, 2017-01-08:
This is the content of platformio.ini:

```
[env:bb1284p8m]
platform = atmelavr
framework = arduino
board = bb1284p8m
lib_extra_dirs = ../libs
upload_port = /dev/ttyS1
upload_speed = 57600
```

Ah - reading these lines I got an idea: It might be an important fact that the uC is running with the internal 8 MHz clock...

tekka007, 2017-01-10:
@DerTomm Updated the PR to use builtin pin functions for ATMega644 / ATmega1284

DerTomm, 2017-01-11:
The latest code changes in the PR solve the problem. The bootup of the serial gateway with attached RF24 radio module works flawlessly now. Thanks!

PR #739 (2017-01-15): AVR: Restrict fast pin function use
- Remove fast pin functions for several archs to reduce maintenance burden
- Fix ATmega168 compatibility
- Fixes #720 
- This supersedes PR #721
closed 2017-01-11 by DerTomm