mysensors/MySensors #724: New set/req type V_TOGGLE

sbrinkmann, 2017-01-07:
I plan to install ceiling lights which can be turned on or off through a push button. This button is only enabled while it is pressed, so the light toggles everytime the switch is pressed. I would like to handle this event properly, but currently the serial api doesn't provide a proper set/req type for that. It would be very helpful when you could add the set/req type **V_TOGGLE**. The type could be used for child sensors of the type _S_BINARY_ or _S_DIMMER_. But I think there are plenty further use cases where actuators could be controlled by toggling there current state. I know I could re-use V_STATUS or V_TRIPPED, but the first type turn actuators on or off and the second one is on while the child sensor action is active, like door is open or motion control is triggered. So they're not suitable for my project.

Thanks and best regards

LemarinelNet, 2017-02-24:
I'm also interrested for this new type.
I would prefer it to be named **V_ACTION**, that is more generic and not only designed for ON/OFF devices.
It would allow it to be used i.e. for window shades/blinds, lights (as you said), locks, IR Sender (Action would mean "send default IR code...", etc...

aktentasche, 2017-02-24:
+1 for this one, it potentially can make the code easier and more readable.

aktentasche, 2017-02-24:
Well, technically you toggle between "going up" and "going down". However, I do agree that it should be usable in other scenarios too. V_xxx could be of integer type, where the integer identifies the type of action to execute as decided by the user.

Can some of the developers confirm that this would potentially only need an extension of the type enums (and the consequences of doing this) in  [MyMessage.h](https://github.com/mysensors/MySensors/blob/development/core/MyMessage.h)?

henrikekblad, 2017-02-24:
Yes, it's only a matter of adding it in the enum. And specifying how it should be interpreted for controller developers.

But I think it sounds a lot like S_SCENE_CONTROLLER  / V_SCENE_ON /V_SCENE_OFF

LemarinelNet, 2017-02-24:
I don't agree. It's not only a matter of on/off: when my blind is moving
the action will stop it...
And nonetheless, the sender of the message doesn't have to know the status
of the target node and decide to send an ON or Off message: it just sends
an Action message.

henrikekblad, 2017-02-24:
Yes, agree, a bit different. So which S-types do you see supporting this V_ACTION? Would the payload have any meaning to the controller?

LemarinelNet, 2017-02-24:
For my cases, the s_dimmer, s_binary ans s_cover will be used.

On the s_dimmer the payload may contain an additionnal byte: increment dim
value, or just toggle...