mysensors/MySensors #725: mysensors crashes when compiled with platformio

crabmanX, 2017-01-07:
when compiling a mysensors sketch with platformio, the code crashes once sleep() is called. compiling the same code with arduino IDE 1.8.0 works fine.

platformio version 3.2.1
arduino uno
nothing else connected to the arduino, no radio, no sensors, etc.

platformio.ini
```
[env:uno]
platform = atmelavr
board = uno
framework = arduino
upload_port = /dev/ttyACM0
lib_deps =
    MySensors@2.1.0
```

src/main.cpp
```
// Enable debug prints to serial monitor
#define MY_DEBUG

// Enable serial gateway
#define MY_GATEWAY_SERIAL

#include <Arduino.h>
#include <MySensors.h>

void presentation()
{
}

void setup()
{
    pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
    digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
    sleep(1000);                       // wait for a second
    digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
    sleep(1000);                       // wait for a second
}
```

serial output
```
³¢á4Ÿ_¿iŽèÞ7¶ÓÞ
JåÛ@Š¡mqùÊ_óiJ5R§ŸK/o¿I5dö|^nIÚ7·ÓIHÙDŠµË8t§Î¢¹ùÓ°dr*5oó8È/1X»§Î¢ù)KAåÉºñ¢ÐèÞî¶µŒ³8áIj#R×¯á4Ÿ_¿iŽèÞ7¶ÓÞ
JåÛ@Š¡mqùÊ_óiJ5R§ŸK/o¿I5dö|^nIÚ7·ÓIHÙDŠ0;255;3;0;9;MCO:BGN:INIT GW,CP=R-NGA--,VER=2.1.0
0;255;3;0;14;Gateway startup complete.
0;255;0;0;17;2.1.0
0;255;3;0;9;MCO:BGN:STP
0;255;3;0;9;MCO:REG:NOT NEEDED
0;255;3;0;9;MCO:BGN:INIT OK,TSP=NA
0;255;3;0;9;MCO:SLP:MS=1000,SMS=0,I1=255,M1=255,I2=255,M2=255
0;255;3;0;9;MCO:BGN:INIT GW,CP=R-NGA--,VER=2.1.0
0;255;3;0;14;Gateway startup complete.
0;255;0;0;17;2.1.0
0;255;3;0;9;MCO:BGN:STP
0;255;3;0;9;MCO:REG:NOT NEEDED
0;255;3;0;9;MCO:BGN:INIT OK,TSP=NA
0;255;3;0;9;MCO:SLP:MS=1000,SMS=0,I1=255,M1=255,I2=255,M2=255
0;255;3;0;9;MCO:BGN:INIT GW,CP=R-NGA--,VER=2.1.0
0;255;3;0;14;Gateway startup complete.
0;255;0;0;17;2.1.0
0;255;3;0;9;MCO:BGN:STP
0;255;3;0;9;MCO:REG:NOT NEEDED
0;255;3;0;9;MCO:BGN:INIT OK,TSP=NA
0;255;3;0;9;MCO:SLP:MS=1000,SMS=0,I1=255,M1=255,I2=255,M2=255
0;255;3;0;9;MCO:BGN:INIT GW,CP=R-NGA--,VER=2.1.0
```

I am not sure if this is a MySensors bug or a problem in platformio.

tekka007, 2017-01-07:
@crabmanX Sounds like the code optimization bug present in gcc < 6.1. It has been fixed in 6.3 and back-ported to 5.5, but will take some time until integrated in the AVR toolchain.
(see here for reference: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=77326)

Could you add this to your sketch and report if that changes anything?

```
void indication(const indication_t)
{
}
```

crabmanX, 2017-01-07:
@tekka007 yep, adding that function makes it work as expected. I indeed configured the arduino IDE to use the toolchain from my OS (archlinux), which ships avr-gcc 6.3. 

Thanks a lot!

closed 2017-01-07 by crabmanX

tekka007, 2017-01-07:
@crabmanX Perfect, thanks for the feedback