mysensors/MySensors #732: Making id request more robust

henrikekblad, 2017-01-13:
**Today:** When several nodes requests id at the same time. They will all get the same id-reponse; thus picking up and start using the same id.

**Proposal:** Implement a new I_ID_REQUEST_V2 which adds a good random number (possible?) in the payload.  When receiving a  I_ID_RESPONSE_V2 it contains both the random number and the new node id. The node checks that the random number matches the one sent.

tekka007, 2017-01-13:
Yes, that's something I previously thought about which can be implemented with a little modification to be backwards compatible: the random number can be placed in the sensor id field sent from the controller:

Node: I_ID_REQUEST, addressed to BC, payload: random uint8
Controller: I_ID_RESPONSE, addressed to BC sensor id = previous random number, payload = new ID

henrikekblad, 2017-01-13:
Ah.. smart! 

I wonder if the controllers preserve sensorId from the id-request (in the response)... that would allow us to implement it without controller even knowing...

MartinHjelmare, 2017-01-13:
Home Assistant / pymysensors will work with this as well.

closed 2017-03-28 by henrikekblad