mysensors/MySensors #733: Error Handling

RWoerz, 2017-01-14:
I'm trying to figure out a way to send error messages from my solar heater project to me in the form of a text message or email. Is there already a way to do that or does that have to be added to MySensors or Vera. We need to way to handle errors in the for of an error message i.e. "Your flux capacitor is out of alignment and could wipe out New York".

henrikekblad, 2017-01-18:
Please use the support forum.

closed 2017-01-18 by henrikekblad

RWoerz, 2017-01-18:
Thank you I will it's just that I saw a similar question on the support
forum and the person was told to post it to Git so I started there. I was
surprised when I didn't find an easy way to report and handle errors. Of
course we're also limited by the controller we chose.

I'm currently using a Vera Edge UI7 which I'm about to throw in the trash.
It's down almost more than it's up. A few minutes of downtime in a week
might be OK but hours a day isn't and so far Vera hasn't been any help.

Thanks for all your work.