mysensors/MySensors #747: MySensors build failed

korvad, 2017-01-18:
When I trying to build mysensors gateway through make I get the following:
[user@hostMySensors]$ ./configure 
[SECTION] Detecting target machine.
[OK] machine detected: SoC=unknown, Type=unknown, CPU=x86_64.
[FAILED] unknown init system
[SECTION] Saving configuration.
[SECTION] Cleaning previous builds.
[OK] Finished.

[user@hostMySensors]$ make
gcc -MT build/drivers/Linux/log.o -MMD -MP -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -c drivers/Linux/log.c -o build/drivers/Linux/log.o
g++ -MT build/drivers/Linux/compatibility.o -MMD -MP -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -c drivers/Linux/compatibility.cpp -o build/drivers/Linux/compatibility.o
g++ -MT build/drivers/Linux/EthernetClient.o -MMD -MP -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -c drivers/Linux/EthernetClient.cpp -o build/drivers/Linux/EthernetClient.o
g++ -MT build/drivers/Linux/EthernetServer.o -MMD -MP -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -c drivers/Linux/EthernetServer.cpp -o build/drivers/Linux/EthernetServer.o
g++ -MT build/drivers/Linux/IPAddress.o -MMD -MP -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -c drivers/Linux/IPAddress.cpp -o build/drivers/Linux/IPAddress.o
g++ -MT build/drivers/Linux/noniso.o -MMD -MP -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -c drivers/Linux/noniso.cpp -o build/drivers/Linux/noniso.o
g++ -MT build/drivers/Linux/Print.o -MMD -MP -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -c drivers/Linux/Print.cpp -o build/drivers/Linux/Print.o
g++ -MT build/drivers/Linux/SerialPort.o -MMD -MP -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -c drivers/Linux/SerialPort.cpp -o build/drivers/Linux/SerialPort.o
g++ -MT build/drivers/Linux/SoftEeprom.o -MMD -MP -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -c drivers/Linux/SoftEeprom.cpp -o build/drivers/Linux/SoftEeprom.o
g++ -MT build/drivers/Linux/Stream.o -MMD -MP -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -c drivers/Linux/Stream.cpp -o build/drivers/Linux/Stream.o
g++ -MT build/examples_linux/mysgw.o -MMD -MP -DMY_RADIO_NRF24 -DMY_GATEWAY_LINUX -DMY_DEBUG  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -c examples_linux/mysgw.cpp -o build/examples_linux/mysgw.o
In file included from examples_linux/mysgw.cpp:74:0:
./MySensors.h:258:2: error: #error No support for nRF24 radio on this platform
 #error No support for nRF24 radio on this platform
  ^
In file included from ./MySensors.h:294:0,
                 from examples_linux/mysgw.cpp:74:
./drivers/RF24/RF24.cpp:39:11: error: Not supported on linux-generic
  hwDigitalWrite(MY_RF24_CS_PIN, level);
           ^
./drivers/RF24/RF24.cpp:44:11: error: Not supported on linux-generic
  hwDigitalWrite(MY_RF24_CE_PIN, level);
           ^
./drivers/RF24/RF24.cpp:388:11: error: Not supported on linux-generic
  hwPinMode(MY_RF24_CE_PIN,OUTPUT);
           ^
./drivers/RF24/RF24.cpp:389:11: error: Not supported on linux-generic
  hwPinMode(MY_RF24_CS_PIN,OUTPUT);
           ^
In file included from ./drivers/RF24/RF24.cpp:23:0,
                 from ./MySensors.h:294,
                 from examples_linux/mysgw.cpp:74:
./drivers/RF24/RF24.h:54:8: error: ‘HardwareSPI’ does not name a type
 extern HardwareSPI SPI;
        ^
In file included from ./MySensors.h:294:0,
                 from examples_linux/mysgw.cpp:74:
./drivers/RF24/RF24.cpp:37:12: warning: unused parameter ‘level’ [-Wunused-parameter]
 LOCAL void RF24_csn(const bool level)
            ^
./drivers/RF24/RF24.cpp:42:12: warning: unused parameter ‘level’ [-Wunused-parameter]
 LOCAL void RF24_ce(const bool level)
            ^
In file included from ./drivers/RF24/RF24.cpp:23:0,
                 from ./MySensors.h:294,
                 from examples_linux/mysgw.cpp:74:
./drivers/RF24/RF24.cpp: In function ‘uint8_t RF24_spiMultiByteTransfer(uint8_t, uint8_t*, uint8_t, bool)’:
./drivers/RF24/RF24.h:58:14: error: ‘SPI’ was not declared in this scope
 #define _SPI SPI
              ^
./drivers/RF24/RF24.cpp:53:2: note: in expansion of macro ‘_SPI’
  _SPI.beginTransaction(SPISettings(MY_RF24_SPI_MAX_SPEED, MY_RF24_SPI_DATA_ORDER,
  ^
./drivers/RF24/RF24.h:33:32: error: ‘MSBFIRST’ was not declared in this scope
 #define MY_RF24_SPI_DATA_ORDER MSBFIRST
                                ^
./drivers/RF24/RF24.cpp:53:59: note: in expansion of macro ‘MY_RF24_SPI_DATA_ORDER’
  _SPI.beginTransaction(SPISettings(MY_RF24_SPI_MAX_SPEED, MY_RF24_SPI_DATA_ORDER,
                                                           ^
./drivers/RF24/RF24.h:34:31: error: ‘SPI_MODE0’ was not declared in this scope
 #define MY_RF24_SPI_DATA_MODE SPI_MODE0
                               ^
./drivers/RF24/RF24.cpp:54:36: note: in expansion of macro ‘MY_RF24_SPI_DATA_MODE’
                                    MY_RF24_SPI_DATA_MODE));
                                    ^
In file included from ./MySensors.h:294:0,
                 from examples_linux/mysgw.cpp:74:
./drivers/RF24/RF24.cpp:54:57: error: ‘SPISettings’ was not declared in this scope
                                    MY_RF24_SPI_DATA_MODE));
                                                         ^
./drivers/RF24/RF24.cpp:56:11: error: ‘LOW’ was not declared in this scope
  RF24_csn(LOW);
           ^
./drivers/RF24/RF24.cpp:58:22: error: ‘delayMicroseconds’ was not declared in this scope
  delayMicroseconds(10);
                      ^
./drivers/RF24/RF24.cpp:99:11: error: ‘HIGH’ was not declared in this scope
  RF24_csn(HIGH);
           ^
./drivers/RF24/RF24.cpp: In function ‘void RF24_startListening()’:
./drivers/RF24/RF24.cpp:235:10: error: ‘HIGH’ was not declared in this scope
  RF24_ce(HIGH);
          ^
./drivers/RF24/RF24.cpp: In function ‘void RF24_stopListening()’:
./drivers/RF24/RF24.cpp:241:10: error: ‘LOW’ was not declared in this scope
  RF24_ce(LOW);
          ^
./drivers/RF24/RF24.cpp:243:23: error: ‘delayMicroseconds’ was not declared in this scope
  delayMicroseconds(130);
                       ^
./drivers/RF24/RF24.cpp: In function ‘void RF24_powerDown()’:
./drivers/RF24/RF24.cpp:251:10: error: ‘LOW’ was not declared in this scope
  RF24_ce(LOW);
          ^
./drivers/RF24/RF24.cpp: In function ‘bool RF24_sendMessage(uint8_t, const void*, uint8_t)’:
./drivers/RF24/RF24.cpp:269:10: error: ‘HIGH’ was not declared in this scope
  RF24_ce(HIGH);
          ^
./drivers/RF24/RF24.cpp:276:10: error: ‘LOW’ was not declared in this scope
  RF24_ce(LOW);
          ^
In file included from ./drivers/RF24/RF24.cpp:23:0,
                 from ./MySensors.h:294,
                 from examples_linux/mysgw.cpp:74:
./drivers/RF24/RF24.cpp: In function ‘bool RF24_initialize()’:
./drivers/RF24/RF24.h:58:14: error: ‘SPI’ was not declared in this scope
 #define _SPI SPI
              ^
./drivers/RF24/RF24.cpp:394:2: note: in expansion of macro ‘_SPI’
  _SPI.begin();
  ^
In file included from ./MySensors.h:294:0,
                 from examples_linux/mysgw.cpp:74:
./drivers/RF24/RF24.cpp:395:10: error: ‘LOW’ was not declared in this scope
  RF24_ce(LOW);
          ^
./drivers/RF24/RF24.cpp:396:11: error: ‘HIGH’ was not declared in this scope
  RF24_csn(HIGH);
           ^
make: *** [build/examples_linux/mysgw.o] Error 1

My environment is:
[user@hostMySensors]$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.5 LTS
Release:	14.04
Codename:	trusty

[user@hostMySensors]$ uname -a
Linux host 3.13.0-101-generic #148-Ubuntu SMP Thu Oct 20 22:08:32 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

[user@hostMySensors]$ gcc --version
gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

marceloaqno, 2017-01-18:
You need to choose another transport option, since rf24 is not supported on your platform.
Run ./configure -h for a complete list of options.

korvad, 2017-01-18:
@marceloaqno, sorry but it's not clear to me what do you mean. Let me clarify what I would like to do. I will plan to build MySensors gateway on Raspberry Pi as described here: https://www.mysensors.org/build/raspberry
Unfortunately my Raspberry has not delivered yet so I can't try to build MySensors on target platform. So I just try to make build steps on my host to check that I can build gateway. But as I wrote earlier I have no success. For now I have arduino uno connected to my host and I try to build Serial Gateway like this: (as it described at https://www.mysensors.org/build/raspberry)
./configure --my-gateway=serial --my-serial-port=/dev/ttyACM0 --my-serial-baudrate=115200
make
But the result is the same:
...
g++ -MT build/examples_linux/mysgw.o -MMD -MP -DMY_RADIO_NRF24 -DMY_GATEWAY_SERIAL -DMY_DEBUG -DMY_BAUD_RATE=115200 -DMY_LINUX_SERIAL_PORT=\"/dev/ttyACM0\"  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -c examples_linux/mysgw.cpp -o build/examples_linux/mysgw.o
In file included from examples_linux/mysgw.cpp:74:0:
./MySensors.h:258:2: error: #error No support for nRF24 radio on this platform
 #error No support for nRF24 radio on this platform
  ^
In file included from ./MySensors.h:294:0,
                 from examples_linux/mysgw.cpp:74:
./drivers/RF24/RF24.cpp:39:11: error: Not supported on linux-generic
  hwDigitalWrite(MY_RF24_CS_PIN, level);
           ^
./drivers/RF24/RF24.cpp:44:11: error: Not supported on linux-generic
  hwDigitalWrite(MY_RF24_CE_PIN, level);
           ^
./drivers/RF24/RF24.cpp:388:11: error: Not supported on linux-generic
  hwPinMode(MY_RF24_CE_PIN,OUTPUT);
           ^
./drivers/RF24/RF24.cpp:389:11: error: Not supported on linux-generic
  hwPinMode(MY_RF24_CS_PIN,OUTPUT);
           ^
In file included from ./drivers/RF24/RF24.cpp:23:0,
                 from ./MySensors.h:294,
                 from examples_linux/mysgw.cpp:74:
./drivers/RF24/RF24.h:54:8: error: ‘HardwareSPI’ does not name a type
 extern HardwareSPI SPI;
        ^
In file included from ./MySensors.h:294:0,
                 from examples_linux/mysgw.cpp:74:
./drivers/RF24/RF24.cpp:37:12: warning: unused parameter ‘level’ [-Wunused-parameter]
 LOCAL void RF24_csn(const bool level)
            ^
./drivers/RF24/RF24.cpp:42:12: warning: unused parameter ‘level’ [-Wunused-parameter]
 LOCAL void RF24_ce(const bool level)
            ^
In file included from ./drivers/RF24/RF24.cpp:23:0,
                 from ./MySensors.h:294,
                 from examples_linux/mysgw.cpp:74:
./drivers/RF24/RF24.cpp: In function ‘uint8_t RF24_spiMultiByteTransfer(uint8_t, uint8_t*, uint8_t, bool)’:
./drivers/RF24/RF24.h:58:14: error: ‘SPI’ was not declared in this scope
 #define _SPI SPI
              ^
./drivers/RF24/RF24.cpp:53:2: note: in expansion of macro ‘_SPI’
  _SPI.beginTransaction(SPISettings(MY_RF24_SPI_MAX_SPEED, MY_RF24_SPI_DATA_ORDER,
  ^
./drivers/RF24/RF24.h:33:32: error: ‘MSBFIRST’ was not declared in this scope
 #define MY_RF24_SPI_DATA_ORDER MSBFIRST
                                ^
./drivers/RF24/RF24.cpp:53:59: note: in expansion of macro ‘MY_RF24_SPI_DATA_ORDER’
  _SPI.beginTransaction(SPISettings(MY_RF24_SPI_MAX_SPEED, MY_RF24_SPI_DATA_ORDER,
                                                           ^
./drivers/RF24/RF24.h:34:31: error: ‘SPI_MODE0’ was not declared in this scope
 #define MY_RF24_SPI_DATA_MODE SPI_MODE0
                               ^
./drivers/RF24/RF24.cpp:54:36: note: in expansion of macro ‘MY_RF24_SPI_DATA_MODE’
                                    MY_RF24_SPI_DATA_MODE));
                                    ^
In file included from ./MySensors.h:294:0,
                 from examples_linux/mysgw.cpp:74:
./drivers/RF24/RF24.cpp:54:57: error: ‘SPISettings’ was not declared in this scope
                                    MY_RF24_SPI_DATA_MODE));
                                                         ^
./drivers/RF24/RF24.cpp:56:11: error: ‘LOW’ was not declared in this scope
  RF24_csn(LOW);
           ^
./drivers/RF24/RF24.cpp:58:22: error: ‘delayMicroseconds’ was not declared in this scope
[... 25 lines cut ...]
  RF24_ce(LOW);
          ^
In file included from ./drivers/RF24/RF24.cpp:23:0,
                 from ./MySensors.h:294,
                 from examples_linux/mysgw.cpp:74:
./drivers/RF24/RF24.cpp: In function ‘bool RF24_initialize()’:
./drivers/RF24/RF24.h:58:14: error: ‘SPI’ was not declared in this scope
 #define _SPI SPI
              ^
./drivers/RF24/RF24.cpp:394:2: note: in expansion of macro ‘_SPI’
  _SPI.begin();
  ^
In file included from ./MySensors.h:294:0,
                 from examples_linux/mysgw.cpp:74:
./drivers/RF24/RF24.cpp:395:10: error: ‘LOW’ was not declared in this scope
  RF24_ce(LOW);
          ^
./drivers/RF24/RF24.cpp:396:11: error: ‘HIGH’ was not declared in this scope
  RF24_csn(HIGH);
           ^
make: *** [build/examples_linux/mysgw.o] Error 1

I suppose maybe I'm doing it wrong. 
In this case point what I need to build gateway success please.
Thx.

henrikekblad, 2017-01-18:
Please use the support forum for these discussions.

closed 2017-01-18 by henrikekblad