mysensors/MySensors #751: getting error TSM:RADIO:FAIL

bkrajendra, 2017-01-18:
I'm new to MySensor platform.
Ive setup my node as per instructions
Im using Arduino Nano 5v (with 3.3v out) and NRF24l01 basic module.
on Arduino 1.6.13
I've tried using multiple options release 2.00, 2.1.1, latest Dev branch.
I'm getting this 
```
0;255;3;0;9;Starting gateway (RNNGA-, 2.0.0)
0;255;3;0;9;TSM:INIT
0;255;3;0;9;!TSM:RADIO:FAIL
0;255;3;0;9;!TSM:FAILURE
0;255;3;0;9;TSM:PDT
0;255;3;0;9;TSM:INIT
0;255;3;0;9;!TSM:RADIO:FAIL
0;255;3;0;9;!TSM:FAILURE
0;255;3;0;9;TSM:PDT
```

henrikekblad, 2017-01-18:
Please use the support forum.

closed 2017-01-18 by henrikekblad

bkrajendra, 2017-01-19:
Hey thanks a lot,
But I just fixed it.
This due to little confusion. Some of the tutorial on website says to use port 7,8 for CE, CSN
and there some which says to use 9, 10

I just swapped pins and it started working.