mysensors/MySensors #753: RFM95 with TCXO

trlafleur, 2017-01-19:
There is a variant of the RFM95 with a temperature control oscillator... It allow for better stability using SF 9 to 12. It requires some small changes to the configuration of the radio to enable it..

Changes needed have added to the RadioHead library (in 1.6.7). You just need a call to enableTCXO() after init().

I have tested it using a RocketScream M0 board with my version of MySensor 2.1 driver for the RFM95.

tekka007, 2017-01-19:
@trlafleur Thanks for your input. Would you mind submitting a PR implementing these changes? This would be helpful considering the limited bandwidth we all have - if you need assistance/help in generating a PR please reach out to the MySensors team.

trlafleur, 2017-01-19:
My development setup is NOT compatible with your PR system...

tekka007, 2017-01-19:
@trlafleur as stated previously, if you need help or assistance, please reach out to the team and I'm sure we'll find a mutual solution.

trlafleur, 2017-02-01:
The following changes were made to support an RFM95 with an TCXO in the default MySensor driver for the RFM95.
This was tested on a RocketStream M0 with an RFM95T with a TCXO.

IDE 1.8.0
SAMD M0 support package 1.6.11
MySensor 2.1 released

In your project, add the define below to enable TCXO support:
~~~
#define MY_RF95_TCXO                    // If using an RFM95 with a TCXO
~~~

In RFM95registers.h added at line: ~202
~~~
// RF95_REG_4B_TCXO 0x4b
#define RF95_TCXO_TCXO_INPUT_ON                 0x10 //!< TCXO_TCXO_INPUT_ON
~~~

In RFM95.h at line: ~337
~~~
/** 
 * @brief Enable TCXO mode
 *   Call this immediately after init(), to force your radio to use an external 
 *   frequency source, such as a Temperature Compensated Crystal Oscillator (TCXO).
 *   See the comments in the main documentation about the sensitivity of this radio to
 *   clock frequency especially when using narrow bandwidths.
 *   Leaves the module in sleep mode.
 */
LOCAL void enableTCXO();			// <----------- trl
~~~
In RFM95.cpp at line: ~387
~~~
/* 
 * Enable TCXO mode
 *   Call this immediately after init(), to force your radio to use an external 
 *   frequency source, such as a Temperature Compensated Crystal Oscillator (TCXO).
 *   See the comments in the main documentation about the sensitivity of this radio to
 *   clock frequency especially when using narrow bandwidths.
 *   Leaves the module in sleep mode.
*/
LOCAL void enableTCXO()					// <------------- trl
{
    while ((RFM95_readReg(RFM95_REG_4B_TCXO) & RF95_TCXO_TCXO_INPUT_ON) != RF95_TCXO_TCXO_INPUT_ON)
    {
	RFM95_sleep();
	RFM95_writeReg(RFM95_REG_4B_TCXO, (RFM95_readReg(RFM95_REG_4B_TCXO) | RF95_TCXO_TCXO_INPUT_ON));
    } 
}
~~~

In MyTransportRFM95,cpp at line ~25 the transportInit() function was modifed:

~~~
bool transportInit(void)
{
	const bool result = RFM95_initialise(MY_RFM95_FREQUENCY);
	
// Only if we are using an RFM95 with a TCXO				// <------------ TRL
#if defined MY_RF95_TCXO
	enableTCXO();
#endif	
	
#if !defined(MY_GATEWAY_FEATURE) && !defined(MY_RFM95_ATC_MODE_DISABLED)
	// only enable ATC mode nodes
	RFM95_ATCmode(true, MY_RFM95_ATC_TARGET_RSSI);
#endif
	return result;
}
~~~

closed 2017-03-28 by henrikekblad

trlafleur, 2017-03-28:
> Closed #753 <https://github.com/mysensors/MySensors/issues/753> via #722

Is there info on how to use the new features like tx and rx rssi and
setting power levels ect...

in the ne 2.2b code released today

thanks

PR #722 (2017-03-28): Transport update
New features:
- Two RFM69 drivers: old (=default) driver, and refactored driver (not compatible with old driver, ATC & seq# implemented), enable using `#define MY_RFM69_NEW_DRIVER` on all nodes & GW.
- Transport power down during sleep if transport-specific power pin defined: `MY_XYZ_POWER_PIN`
- New mode: `MY_PASSIVE_NODE` - node reports passively, i.e. one-off transmission, no established uplink required, no resend, no ACK: transmissions can be undirected (no defined parent, but need for de-duplication as all repeaters will forward the message, not recommended - frequent rou [... cut]