mysensors/MySensors #765: ESP8266 AES example

anaisortn, 2017-02-01:
Hi, using your [example](https://github.com/mysensors/MySensors/blob/5c39f022c0db5e0280be1c1412f51be7d435cd60/libraries/MySensors/drivers/AES/examples/aes/aes.pde) on ESP8266, the decrypted output isAdd NodeAdd NodeAdd NodeAdd NodeAdd Node��N���0��%���렝N���0��%� instead of Add NodeAdd NodeAdd NodeAdd NodeAdd Node

```c
#include 
/* code was modified by george spanos 
* 16/12/14
*/
AES aes ;
byte *key = (unsigned char*)"0123456789010123";
byte plain[] = "Add NodeAdd NodeAdd NodeAdd NodeAdd Node";

//real iv = iv x2 ex: 01234567 = 0123456701234567
unsigned long long int my_iv = 36753562;
void setup ()
{
Serial.begin (115200) ;
delay(500);
printf("\n===testng mode\n") ;
//  otfly_test () ;
//  otfly_test256 () ;
}
void loop ()
{
prekey_test () ;
delay(2000);
}
void prekey (int bits)
{
aes.iv_inc();
byte iv [N_BLOCK] ;
byte plain_p[48];
byte cipher [48] ;
byte check [48] ;
unsigned long ms = micros ();
aes.set_IV(my_iv);
aes.get_IV(iv);
aes.do_aes_encrypt(plain,41,cipher,key,bits,iv);
aes.do_aes_decrypt(cipher,48,check,key,bits,iv);
printf("%s \n", check);
}
void prekey_test ()
{
prekey (128) ;
}
```
What am I missing?

henrikekblad, 2017-02-01:
The AES driver is a standalone library available here:
https://github.com/spaniakos/AES

So unless this is a MySensors problem I suggest you direct your questions to the maintainer.

closed 2017-02-01 by henrikekblad

anaisortn, 2017-02-01:
@henrikekblad already did that, but no answer.
@fallberg as I told, this is the example that is in this repository. If I knew why, I wouldn't ask..

fallberg, 2017-02-01:
Well, the example is from the maintainer, not from us (MySensors). I'd say the example is wrong, since we use encryption successfully so it is known to work. You could just try to read the same amount you write.

anaisortn, 2017-02-01:
I tried with `byte check [41]` and ` aes.do_aes_decrypt(cipher,41,check,key,bits,iv);`, with `40` as well, the garbage was there.. Thank you anyway..

fallberg, 2017-02-01:
Well, I don't know what the problem might be, but what you put in was decrypted properly as far as I can see. All the bytes in are present out (plus something else, but still there).
But as this is an external library, MySensors cannot provide support for it.