mysensors/MySensors #769: Tool bootstrap version check for git incorrectly fails on ubuntu and cygwin

bblacey, 2017-02-07:
from @tbowmo:
```
>.mystools/bootstrap-dev.sh
Checking operating system support: linux-gnu...
Checking github 'origin' & 'upstream' remotes...
Git "upstream" remote not found or incorrectly defined. Configuring remote upstream --> https://github.com/mysensors/MySensors.git...
Checking tool/utility prerequisites...
Found git  however, version 2.0 or greater is required...
Install git 2.0 or greater and re-run .mystools/bootstrap-dev.sh
thomas at thomas-ThinkPad-L450 in ~/tmp/MySensors git:(development) ✗
>git --version
git version 2.9.3
thomas at thomas-ThinkPad-L450 in ~/tmp/MySensors git:(development) ✗
```

from @mfalkvidd:

```
$ .mystools/bootstrap-dev.sh
git: 'mystoolspath' is not a git command. See 'git --help'.
Checking operating system support: cygwin...
Checking github 'origin' & 'upstream' remotes...
Git "origin" should point to your github fork, not the github MySensors repo
Checking tool/utility prerequisites...
Installing client-side git hooks...
Configuring git aliases for running mysensor tools...
Successfully configured your repo for MySensors development... Thanks for your support!
```

Most likely a regex portability issue.

cc: @fallberg

mfalkvidd, 2017-07-10:
The bootstrap check works on cygwin now. Not sure what has changed.

fallberg, 2018-02-18:
I tried to reproduce this on a virtual Kubuntu (17.10) but the bootstrap worked just fine. git version 2.14.1.

tbowmo, 2018-02-18:
Verified ok on plain ubuntu 17.10 as well.

closed 2018-02-18 by tbowmo