mysensors/MySensors #770: RPI gateway example should include #defines for controller INCLUSION_MODE

bblacey, 2017-02-07:
Per forum post https://forum.mysensors.org/topic/5795/solved-raspberry-pi-gateway-to-vera-ui7-not-adding-sensors/8, the examples_linux/mysgw.cpp example does not include the INCLUSION_MODE #defines found in the other gateway examples.  This caused thrashing when users tried to use the RPI gateway with the Vera Controller plugin.  While the #defines are in MyConfig.h, for parity with other gateway examples and to eliminate future confusion, we should add the following #defines to the mysgw.cpp example.

// Enable inclusion mode if your HA Controller supports it (e.g. Vera Controller)
#define MY_INCLUSION_MODE_FEATURE
// Enable Inclusion mode button on gateway
#define MY_INCLUSION_BUTTON_FEATURE

@marceloaqno - Does the RPI Gateway configuration support a button to enter inclusion mode?

cc/ @mfalkvidd

bblacey, 2017-02-07:
Note:  We may want to compare the other #defines between the gateway examples to ensure parity across the board.

marceloaqno, 2017-02-08:
@bblacey Yes (I've never tested it, but it should work).
We should also add:
#define MY_INCLUSION_MODE_BUTTON_PIN
to set the pin

PR #841 (2017-05-15): Linux: Add inclusion mode to gateway example
Fix #770
commit 2a7145cc (2017-05-15): Linux: Add inclusion mode to gateway example (#841)

Fix #770
closed 2017-05-15 by henrikekblad