mysensors/MySensors #773: The gateway should have a unique identifier

rakeshpai, 2017-02-14:
It's currently not possible to tell the difference between different gateways. This is useful if one wants to support multiple gateways on a single controller. This is also useful if the gateway is using client-mode and connecting to a cloud-based service, and the service needs to identify the gateway.

I'm guessing this could be implemented as an internal message in the serial API, and as a #define in the library API.

mfalkvidd, 2018-02-18:
On mqtt gateways, #define MY_MQTT_CLIENT_ID "mysensors-1" can be used. We should try to aligin solutions for ethernet and serial gateways (configuration-wise, implementattion probably needs to be different since the mqtt id concept doesn't exist for ethernet and serial)

MartinHjelmare, 2018-02-18:
It would be best, from a controller perspective, if this was part of the mysensors serial API, and not dependent on gateway type.