mysensors/MySensors #776: Using the list of available access points for ESP8266-based gateway.

myachniy, 2017-02-18:
Hello!

I have a large house and several different access points in it. My ESP8266-based sensors are located in different places all over my house. In these places they can't reach the same AP to connect to using MySensor library. Currently I'm using a bit modified lib that tries to connect to any AP listed in AP list. If one AP fails it tries to connect to next one using round-robin scheme.

Are there any plans of adding such sort of functionality into the library or I'm the only one nerd that needs it?

trlafleur, 2017-04-19:
This would be a very useful feature.....

Publish you code modification here, for others to test with....

myachniy, 2017-04-19:
Good evening.

Here is the file I've changed (and an example as well): 
[MultipleSSIDs.zip](https://github.com/mysensors/MySensors/files/937796/MultipleSSIDs.zip)

You can also find it in my fork of the repo: https://github.com/myachniy/MySensors