mysensors/MySensors #779: NODE_ID not saved when mode is AUTO.

LemarinelNet, 2017-03-02:
Hi,

It seems that node ID is never saved to EEPROM when MY_NODE_ID is set to AUTO.
I've discovered this while looking at gateway logs, since my nodes always request a new ID (and get a new one) at each reboot.

When looking at https://github.com/mysensors/MySensors/blob/development/core/MyTransport.cpp#L103 it seems that hwWriteConfig is only called when mode != AUTO, same at https://github.com/mysensors/MySensors/blob/development/core/MyTransport.cpp#L445 .

In the docs, I can read that when on AUTO mode, the node should save its ID in EEPROM : 
in https://www.mysensors.org/download/sensor_api_20 : "Once an id has been received from the controller it will be saved in the Arduino's static EEPROM memory. The id is automatically read from EEPROM each time the Sensor node resets or powers-up. "

Am I doing things wrong ?

tekka007, 2017-03-02:
Can you post your sketch and the full debug logs (GW and node)? Do you see the same issue when using different HW?

LemarinelNet, 2017-03-02:
Here is my GW and node sketchs : 

Gateway code : 
https://github.com/LemarinelNet/KicadDomo/blob/master/MySensors_Gateway/FIRMWARE/GateWay_RFM69.ino

Node code :
https://github.com/LemarinelNet/KicadDomo/blob/master/MultiUse_Battery_RFM69HCW/FIRMWARE/LMT85TemperatureSensor.ino

I don't have the node debug log : it runs wirelessly and doesn't has serial port.

The logs from the gateway will come soon.

tekka007, 2017-03-02:
Can you run the same sketch on a different HW with serial output - I'd like to see the node debug log to understand what's going on.

LemarinelNet, 2017-03-02:
OK I'll give a try this eneving.

LemarinelNet, 2017-03-04:
Hi,

Some feedback about this issue :+1: It's all my fault.
In facts, I was using an usbasp. Each time I do update my firmware with the usbasp, the EEPROM seems to be erased also !
I did backup the EEPROM before updating the program memory, all is fine !!

Sorry for the false warning !

tekka007, 2017-03-04:
Thanks for the feedback.

closed 2017-03-04 by tekka007