mysensors/MySensors #783: The example BinarySwitchSleepSensor doen't work anymore

mortommy, 2017-03-03:
Hi,
I have a sensor based on the example BinarySwitchSleepSensor. It was running for a year powered by a coin cell battery. Recently, before to replace the battery at the end of his cycle, I decided to upgrade it to the library 2.1.1 and I notice that when the signal at the interrupt pin is HIGH, is keeping waking up, when goes LOW stay in sleep mode. Going to the library 2.0.0 made it working again.
Thank you.

pgollor, 2017-03-11:
Could you post your sketch? For me it works with `2.1.1` and sleeping with an PIR sensor.

mortommy, 2017-03-14:
here it is:

```
//Enable debug prints to serial monitor
//#define MY_DEBUG

#define MY_RADIO_NRF24

#define MY_NODE_ID 4

#define PRIMARY_CHILD_ID 0
#define PRIMARY_BUTTON_PIN 3   //Arduino Digital I/O pin for button/reed switch

int BATTERY_SENSE_PIN = A0;
int oldBatteryPcnt = 0;
int buttonValue = 1; 
int buttonSentValue = 2;
int batteryValue = 0;
float batteryV = 0;
int batteryPcnt = 0;

#include <MyConfig.h>
#include <MySensors.h>

MyMessage msg(PRIMARY_CHILD_ID, V_TRIPPED);

void setup()  
{ 
  analogReference(INTERNAL);
  
  //Setup the buttons
  pinMode(PRIMARY_BUTTON_PIN, INPUT);

  //Activate internal pull-ups
  digitalWrite(PRIMARY_BUTTON_PIN, HIGH);
    
  //Short delay to allow buttons to properly settle
  sleep(5);
}

void presentation()  
{ 
  sendSketchInfo("BinarySensorIntBatt", "2.0");
  sleep(250);
  present(PRIMARY_CHILD_ID, S_DOOR);
}

// Loop will iterate on changes on the BUTTON_PINs
void loop() 
{  
  buttonValue = digitalRead(PRIMARY_BUTTON_PIN);
  
  if (buttonValue != buttonSentValue) {
     //Value has changed from last transmission, send the updated value
     send(msg.set(buttonValue==HIGH));
     buttonSentValue = buttonValue;
  }

  //sleep after radio communication before to read the battery value
  sleep(300);
   
  batteryValue = analogRead(BATTERY_SENSE_PIN);
  batteryV  = batteryValue * (3.44/1023);
  batteryPcnt = int(batteryValue / 10);
  
  if (oldBatteryPcnt != batteryPcnt) {
    sendBatteryLevel(batteryPcnt);
    oldBatteryPcnt = batteryPcnt;
 }

  //Sleep until something happens with the sensor
  sleep(digitalPinToInterrupt(PRIMARY_BUTTON_PIN), CHANGE, 0);
} 
```
I understood something was wrong when the new battery discharged after few days and the arduino's LED was blinking very quickly. The debug mode showed that the loop method was continuoisly executing when the PIN 3 was non connected to GND, and it was stopping only with PIN 3 connected to GND.

pgollor, 2017-03-15:
Try somethin linke this:
```
sleep(digitalPinToInterrupt(PRIMARY_BUTTON_PIN), CHANGE, 0, false);
```

So there is a bug with the overloaded functions i think and the smart sleep function doesn't work for me too. Without `false` it will try to use the smart sleep function because the compiler is using the wrong function i think.

mortommy, 2017-03-15:
Hi, thank you.
Did it work for you? I had no luck. It's a pity because the library 2.0.0 gives many fails tentative at startup:

```
Starting sensor (RNNNA-, 2.0.0)
TSM:INIT
TSM:RADIO:OK
TSP:ASSIGNID:OK (ID=4)
TSM:FPAR
TSP:MSG:SEND 4-4-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
TSM:FPAR
TSP:MSG:SEND 4-4-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
TSP:MSG:READ 0-0-4 s=255,c=3,t=8,pt=1,l=1,sg=0:0
TSP:MSG:FPAR RES (ID=0, dist=0)
TSP:MSG:PAR OK (ID=0, dist=1)
TSP:MSG:READ 3-3-4 s=255,c=3,t=8,pt=1,l=1,sg=0:1
TSP:MSG:FPAR RES (ID=3, dist=1)
TSM:FPAR:OK
TSM:ID
TSM:CHKID:OK (ID=4)
TSM:UPL
TSP:PING:SEND (dest=0)
!TSP:MSG:SEND 4-4-0-0 s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=fail:1
TSP:CHKUPL:FAIL (hops=255)
!TSM:UPL:FAIL
TSM:FPAR
TSP:MSG:SEND 4-4-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
TSM:FPAR
TSP:MSG:SEND 4-4-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
TSP:MSG:READ 0-0-4 s=255,c=3,t=8,pt=1,l=1,sg=0:0
TSP:MSG:FPAR RES (ID=0, dist=0)
TSP:MSG:PAR OK (ID=0, dist=1)
TSP:MSG:READ 3-3-4 s=255,c=3,t=8,pt=1,l=1,sg=0:1
TSP:MSG:FPAR RES (ID=3, dist=1)
TSM:FPAR:OK
TSM:ID
TSM:CHKID:OK (ID=4)
TSM:UPL
TSP:PING:SEND (dest=0)
!TSP:MSG:SEND 4-4-0-0 s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=fail:1
TSP:CHKUPL:FAIL (hops=255)
!TSM:UPL:FAIL
TSM:FPAR
TSP:MSG:SEND 4-4-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
TSM:FPAR
TSP:MSG:SEND 4-4-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
TSP:MSG:READ 0-0-4 s=255,c=3,t=8,pt=1,l=1,sg=0:0
TSP:MSG:FPAR RES (ID=0, dist=0)
TSP:MSG:PAR OK (ID=0, dist=1)
TSP:MSG:READ 3-3-4 s=255,c=3,t=8,pt=1,l=1,sg=0:1
TSP:MSG:FPAR RES (ID=3, dist=1)
TSM:FPAR:OK
TSM:ID
TSM:CHKID:OK (ID=4)
TSM:UPL
TSP:PING:SEND (dest=0)
!TSP:MSG:SEND 4-4-0-0 s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=fail:1
TSP:CHKUPL:FAIL (hops=255)
!TSM:UPL:FAIL
TSM:FPAR
TSP:MSG:SEND 4-4-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
TSM:FPAR
TSP:MSG:SEND 4-4-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
TSP:MSG:READ 0-0-4 s=255,c=3,t=8,pt=1,l=1,sg=0:0
TSP:MSG:FPAR RES (ID=0, dist=0)
TSP:MSG:PAR OK (ID=0, dist=1)
TSP:MSG:READ 3-3-4 s=255,c=3,t=8,pt=1,l=1,sg=0:1
TSP:MSG:FPAR RES (ID=3, dist=1)
TSM:FPAR:OK
TSM:ID
TSM:CHKID:OK (ID=4)
TSM:UPL
TSP:PING:SEND (dest=0)
!TSP:MSG:SEND 4-4-0-0 s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=fail:1
TSP:CHKUPL:FAIL (hops=255)
!TSM:UPL:FAIL
TSM:FPAR
TSP:MSG:SEND 4-4-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
TSM:FPAR
TSP:MSG:SEND 4-4-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
TSP:MSG:READ 0-0-4 s=255,c=3,t=8,pt=1,l=1,sg=0:0
TSP:MSG:FPAR RES (ID=0, dist=0)
TSP:MSG:PAR OK (ID=0, dist=1)
TSP:MSG:READ 3-3-4 s=255,c=3,t=8,pt=1,l=1,sg=0:1
TSP:MSG:FPAR RES (ID=3, dist=1)
TSM:FPAR:OK
TSM:ID
TSM:CHKID:OK (ID=4)
TSM:UPL
TSP:PING:SEND (dest=0)
!TSP:MSG:SEND 4-4-0-0 s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=fail:1
TSP:CHKUPL:FAIL (hops=255)
!TSM:UPL:FAIL
TSM:FPAR
TSP:MSG:SEND 4-4-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
TSM:FPAR
TSP:MSG:SEND 4-4-255-255 s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=bc:
TSP:MSG:READ 0-0-4 s=255,c=3,t=8,pt=1,l=1,sg=0:0
TSP:MSG:FPAR RES (ID=0, dist=0)
TSP:MSG:PAR OK (ID=0, dist=1)
TSP:MSG:READ 3-3-4 s=255,c=3,t=8,pt=1,l=1,sg=0:1
TSP:MSG:FPAR RES (ID=3, dist=1)
TSM:FPAR:OK
TSM:ID
TSM:CHKID:OK (ID=4)
TSM:UPL
TSP:PING:SEND (dest=0)
TSP:MSG:SEND 4-4-0-0 s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=ok:1
TSP:MSG:READ 0-0-4 s=255,c=3,t=25,pt=1,l=1,sg=0:1
TSP:MSG:PONG RECV (hops=1)
TSP:CHKUPL:OK
TSM:UPL:OK
TSM:READY
!TSP:MSG:SEND 4-4-0-0 s=255,c=3,t=15,pt=6,l=2,sg=0,ft=0,st=fail:0100
TSP:MSG:SEND 4-4-0-0 s=255,c=0,t=17,pt=0,l=5,sg=0,ft=1,st=ok:2.0.0
TSP:MSG:SEND 4-4-0-0 s=255,c=3,t=6,pt=1,l=1,sg=0,ft=0,st=ok:0
TSP:MSG:SEND 4-4-0-0 s=255,c=3,t=11,pt=0,l=19,sg=0,ft=0,st=ok:BinarySensorIntBatt
TSP:MSG:SEND 4-4-0-0 s=255,c=3,t=12,pt=0,l=3,sg=0,ft=0,st=ok:2.0
TSP:MSG:SEND 4-4-0-0 s=0,c=0,t=0,pt=0,l=0,sg=0,ft=0,st=ok:
Request registration...
!TSP:MSG:SEND 4-4-0-0 s=255,c=3,t=26,pt=1,l=1,sg=0,ft=0,st=fail:2
!TSP:MSG:SEND 4-4-0-0 s=255,c=3,t=26,pt=1,l=1,sg=0,ft=1,st=fail:2
!TSP:MSG:SEND 4-4-0-0 s=255,c=3,t=26,pt=1,l=1,sg=0,ft=2,st=fail:2
!TSP:MSG:SEND 4-4-0-0 s=255,c=3,t=26,pt=1,l=1,sg=0,ft=3,st=fail:2
Init complete, id=4, parent=0, distance=1, registration=1
TSP:MSG:SEND 4-4-0-0 s=0,c=1,t=16,pt=1,l=1,sg=0,ft=4,st=ok:1
TSP:MSG:SEND 4-4-0-0 s=255,c=3,t=0,pt=1,l=1,sg=0,ft=0,st=ok:54
```
The many tentative make the battery down soon. I don't see the fails using the library 2.1.1, but it wakes up continuously. Also tried 2.1.0.

mfalkvidd, 2017-07-10:
Could this be related to https://github.com/mysensors/MySensors/pull/813 ?

tekka007, 2019-01-27:
Possibly fixed by #813 and #1113 - closing issue for now.

closed 2019-01-27 by tekka007