mysensors/MySensors #788: Restart of Gateway on lost of WiFi on ESP8266

trlafleur, 2017-03-07:
I have noted that if the Access Point that the MQTT-ESP8266 gateway  is connect to is turned off and re powered on. IE: a power failure

That the gateway will not re connect to the AP with out a re-boot...

MySensor 2.1.1
IDE 1.8.1

pgollor, 2017-03-11:
I have a similar problem with my RaspberryPi Zero as MySensors Gateway with MQTT. If the Pi loss the WLAN signal for a few seconds, `mysgw` will not reconnect to the mqtt brocker on another Pi.

marceloaqno, 2017-05-15:
@pgollor Does mysgw print any error messages? Could you provide the debug information?

blacksheepinc, 2017-06-07:
Same for me, if there's a power loss, router restarts. While it's setting up,my gateway ESP tries to reconnect. When the router is up, both my ESP and raspberry pi with mosquitto broker connects to the wifi.
Although the ESP cannot connect to the MQTT, it just stuck at the reconnect state.
After I restart the ESP with the reset button, it works immediately.

micropet, 2017-07-11:
I also have the problem with my RPI gateways.
If the mqtt connection is interrupted, it will not be recovered.

Eorciu, 2017-11-26:
Same for me. 
MySensors gateway on ESP8266 nodemcu 0.9 without sensors  - version 2.2.0 - beta 
controller - Domoticz - newest beta. 

I have 3 gateways and 3 access point. One AP simetime "hangs" and after reboot it  gateway mysensors connected before to it  "not recconect" but EspEasy don't have problem with this.

gmarti, 2018-03-28:
Same problem on ESP8266 

When rebooting wifi router, esp8266 panic() in __yield

Logs attached, please ask if you need more information
[esp8266crash.txt](https://github.com/mysensors/MySensors/files/1856487/esp8266crash.txt)

devicter, 2018-10-16:
I can also confirm this problem.
My gateway works on RPI3
Library version 2.3.1-alpha

Yveaux, 2019-01-04:
Reproduced using:
- ESP8266: WeMos D1 MINI USB NodeMcu Lua V3
- nRF24L01+: E01-ML01D
- MySensors 2.3.1
- Arduino 1.8.8
- esp8266 2.4.2
- Default GatewayESP8266MQTTClient.ino sketch (note: MQTT in/out topics are swapped) using fixed IP, Wifi-AP: "test"

Steps to reproduce:
- After connecting to AP "test", the MQTT topic "mySensors/x/in/0/255/0/0/18" is set to "2.3.1".
- Powercycle AP
- ESP8266 crashes, reboot and tries to reconnect
- AP comes online and ESP8266 reconnects
- After connecting to AP "test", the MQTT topic "mySensors/x/in/0/255/0/0/18" is set to "2.3.1".

So it does crash, but recovers correctly; see attached log:

```
0$n⸮⸮`,lD⸮l`8⸮⸮⸮59 MCO:BGN:INIT GW,CP=RNNGE---,REL=255,VER=2.3.1
110 TSF:LRT:OK
126 TSM:INIT
140 TSF:WUR:MS=0
163 TSM:INIT:TSP OK
scandone
no test found, reconnect after 1s
229 TSM:INIT:GW MODE
251 TSM:READY:ID=0,PAR=0,DIS=0
reconnect
294 MCO:REG:NOT NEEDED
scandone
320 TSM:READY:NWD REQ
387 TSF:MSG:SEND,0-0-255-255,s=255,c=3,t=20,pt=0,l=0,sg=0,ft=0,st=OK:
820 GWT:TPC:CONNECTING...
1347 GWT:TPC:CONNECTING...
1875 GWT:TPC:CONNECTING...
2403 GWT:TPC:CONNECTING...
2931 GWT:TPC:CONNECTING...
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 
3459 GWT:TPC:CONNECTING...
3987 GWT:TPC:CONNECTING...

connected with test, channel 1
ip:192.168.1.98,mask:255.255.255.0,gw:192.168.1.1
ip:192.168.1.98,mask:255.255.255.0,gw:192.168.1.1
4515 GWT:TPC:CONNECTING...
4543 GWT:TPC:IP=192.168.1.98
4574 MCO:BGN:STP
4592 MCO:BGN:INIT OK,TSP=1
4620 GWT:TPC:IP=192.168.1.98
4650 GWT:RMQ:MQTT RECONNECT
5131 GWT:RMQ:MQTT CONNECTED
5161 GWT:TPS:TOPIC=mySensors/x/in/0/255/0/0/18,MSG SENT
pm open,type:2 0
bcn_timout,ap_probe_send_start
ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7

Panic c:\Users\Ivo\Documents\Arduino_MySensors2.3\libraries\MySensors/hal/architecture/ESP8266/MyMainESP8266.cpp:105 __yield

ctx: sys 
sp: 3fffedb0 end: 3fffffb0 offset: 01b0

[... 136 lines cut ...]
54334 GWT:TPC:CONNECTING...
54863 GWT:TPC:CONNECTING...
55392 GWT:TPC:CONNECTING...
55921 GWT:TPC:CONNECTING...
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with test, channel 1
ip:192.168.1.98,mask:255.255.255.0,gw:192.168.1.1
ip:192.168.1.98,mask:255.255.255.0,gw:192.168.1.1
56462 GWT:TPC:CONNECTING...
56624 GWT:TPC:IP=192.168.1.98
56655 MCO:BGN:STP
56674 MCO:BGN:INIT OK,TSP=1
56704 GWT:TPC:IP=192.168.1.98
56735 GWT:RMQ:MQTT RECONNECT
56984 GWT:RMQ:MQTT CONNECTED
57015 GWT:TPS:TOPIC=mySensors/x/in/0/255/0/0/18,MSG SENT

```

closed 2019-01-05 by tekka007

Yveaux, 2019-01-05:
BTW. RPi problems reported seem duplicates of #893

SjoerdNLD, 2019-01-12:
@Yveaux  Got exactly the same.
" bcn_timout,ap_probe_send_start "
" ap_probe_send over, rest wifi status to disassoc "

It is related to beacon timeout of your router.
In the router settings try to disable optimiziations like "wireless tv", "xbox", "coexistance" 
*terms may differ per router manufacturer. 

Worked for me with ESP8266 version 2.4.2

Version 2.5.0, no solution found, keeps hitting bcn_timout.
Seems to do with mysensors code, intergrated example from ESP8266 works normal.

PR #1248 (2019-01-05): Add MQTT Wifi connection status check
Fix #788, causing crash on ESP8266 when Wifi connection to AP fails with MQTT connected.