mysensors/MySensors #789: MY_MQTT_SUBSCRIBE_TOPIC_PREFIX as const char* ?

getlarge, 2017-03-08:
Hello, thanks for the MySensors team for this great project, it is an excellent and very complete library for some sweet home IOT! 

I didn't had any trouble until then but still i'm not sure if this is an issue or something misunderstood, but no info could be found on the forum or google.
Let me explain, when you define this prefix, the library adds "/+/+/+/+/+" before the subscribe function gets called. I'm trying to "replace" several #define (MQTT_USER, PASSWORD, CLIENTID) by some const char* to change these values with the library WifiManager by Tzapu it works for every variable, but the subscribe topic. 

- Here are the variable concerned :
#define MY_MQTT_PUBLISH_TOPIC_PREFIX mqttTopicOut
#define MY_MQTT_SUBSCRIBE_TOPIC_PREFIX mqttTopicIn 
#define MY_MQTT_CLIENT_ID mqttClient 
#define MY_MQTT_USER mqttUser
#define MY_MQTT_PASSWORD mqttPassword
//#define MY_CONTROLLER_IP_ADDRESS 164, 132, 103, 148
#define MY_CONTROLLER_URL_ADDRESS mqttServer // adresse du serveur MQTT
#define MY_PORT mqttPort
char mqtt_client[60], mqtt_user[20], mqtt_password[40], mqtt_server[40], mqtt_port[6];
char mqtt_topic_out[70], mqtt_topic_in[70], out[10]= "-out", in[10]= "-in" ; 
int mqttPort;
const char* mqttServer;
const char* mqttClient;
const char* mqttUser;
const char* mqttPassword;
const char* mqttTopicOut;
const char* mqttTopicIn;
strcpy(mqtt_topic_in,mqtt_client);
strcat(mqtt_topic_in,in);  
MY_MQTT_SUBSCRIBE_TOPIC_PREFIX = mqtt_topic_in;

- First time I just had "-in" to mqtt_client, but i had this error : " error: expected ')' before string constant
     _MQTT_client.subscribe(MY_MQTT_SUBSCRIBE_TOPIC_PREFIX "/+/+/+/+/+");"

- So I tried to concatenate to make the topic prefixes inherit from the MY_MQTT_CLIENT_ID :
char mqtt_topic_out[70], mqtt_topic_in[80], out[10]= "-out", in[20]= "-in/+/+/+/+/+" ; 
And also modified MyGatewayTransportMQTTClient.cpp on line 85 with 	    _MQTT_client.subscribe(MY_MQTT_SUBSCRIBE_TOPIC_PREFIX);

When doing so I encounter problems , the message gets received on the subscribe topic, but the gateway does not parse the message anymore...

Maybe the method is not right ? As someone have a suggestion that would not imply to mod the core library ?

henrikekblad, 2017-03-08:
You would probably also have to remove/modify the sanity check and prefix removal check here:
https://github.com/mysensors/MySensors/blob/37881d56babf058e37713ae99c8065eba64e0235/core/MyProtocolMySensors.cpp#L130-L136

With a combination of a "ugly fix" where you remove the appended subscribe suffix using something like this:
#define "/+/+/+/+/+" 

(not tested)

getlarge, 2017-03-09:
Thanks for your suggestion. I admit not knowing where to start with the sanity check, but i'll try few tests and come back here to report the results.

PR #1240 (2019-01-05): MQTT GW: Subscribe topic concatenation refactored
Refactor of MY_MQTT_SUBSCRIBE_TOPIC_PREFIX and "/+/+/+/+/+" to support dynamic declaration of MY_MQTT_SUBSCRIBE_TOPIC_PREFIX.

Fixes #789
closed 2019-01-05 by tekka007