mysensors/MySensors #790: raspberry Pi as MQTT gateway with software signing does not work

pgollor, 2017-03-10:
I'm using a NRF24L01 which is directly connected to a Raspberry Pi. I want to use a MQTT with software signing for the RaspberryPi as Gateway.
The software version is the git tag `2.1.1`.

The configure:
```
./configure --my-debug=enable --my-transport=nrf24 --my-rf24-irq-pin=18 --my-gateway=mqtt --my-controller-ip-address=127.0.0.1 --my-mqtt-publish-topic-prefix="/mysensors-out" --my-mqtt-subscribe-topic-prefix="/mysensors-in" --my-mqtt-client-id="MySysGateway" --my-signing=software --my-signing-request-signatures
```

After make I did the following things:
```
sudo ./mysgw --set-soft-serial-key=[key values]
sudo ./mysgw --set-soft-hmac-key=[key values]
sudo ./mysgw -d
```

The program halt after this without any failure:
```
mysgw: Starting gateway...
mysgw: Protocol version - 2.1.1
mysgw: MCO:BGN:INIT GW,CP=RNNG-SQ,VER=2.1.1
mysgw: TSF:LRT:OK
mysgw: TSM:INIT
mysgw: TSF:WUR:MS=0
mysgw: TSM:INIT:TSP OK
mysgw: TSM:INIT:GW MODE
mysgw: TSM:READY:ID=0,PAR=0,DIS=0
mysgw: MCO:REG:NOT NEEDED
mysgw: MCO:BGN:STP
mysgw: MCO:BGN:INIT OK,TSP=1
mysgw: Attempting MQTT connection...
mysgw: connected to 127.0.0.1
mysgw: MQTT connected
mysgw: Sending message on topic: /mysensors-out/0/255/0/0/18
```
My MQTT broker receives the message, but after that `mysgw` stops and I don't know why.

Without `--my-signing=software --my-signing-request-signatures` it works very good, but without signing. :-(
Could anyone help me?

PS: with version `2.1.0` it is the same.

pgollor, 2017-03-11:
Currently it works with the development branch (e7ae5d144e9a3355b2f035541f384bc912ab5c17).

fallberg, 2017-03-13:
So you mean it is not working on the latest release, but it is on the development branch (BETA)?

pgollor, 2017-03-13:
Yes exactly. So we could close this issue, or not?

closed 2017-03-13 by pgollor

fallberg, 2017-03-13:
Yes, I think so as it would be a fix on development anyway. Thanks for reporting and testing!