mysensors/MySensors #801: Prefered Parent Request

Akubi, 2017-03-26:
If for some reason a node lose it's prefered parent, no new I_FIND_PARENT_REQUEST is issued.
Ti's possible to add a I_FIND_PARENT_REQUEST on regular time bases if prefered parent defined by MY_PARENT_NODE_ID and current parentNodeId is different from this one ?

tekka007, 2017-03-27:
Hi Lorenzo,
Could you post your sketch along with a full debug log showing what you described?

Akubi, 2017-03-28:
Hi Olivier,
Ok I have to prepare a special Setup for that. I don't will do that on my "live" system.

I post as soon as I can.

Akubi, 2017-04-02:
Hi Olivier,
Finally, I was able to Setup the configuration.
The explanation is quite long, sorry, so I put it in a PDF file
[Find Parent issue.pdf](https://github.com/mysensors/MySensors/files/888378/Find.Parent.issue.pdf)

tekka007, 2017-04-02:
Hi Lorenzo, very well elaborated & explained.
What you describe is as intended in the code i.e. once the node is up and running it won't reconnect to a previously offline (& preferred) parent unless the uplink fails - this triggers a new parent search.
As an intermediate solution, you could set `#define MY_PARENT_NODE_IS_STATIC` - this will disable the parent search mechanism and statically route the communication via defined parent. 

However, enforcing communication via a predefined parent is a valid request, I'll hack something...

Akubi, 2017-04-02:
Hi Olivier,

Thank for the very quick answer.

Yes I take a look in the state machine and yes no parent search unless uplink fails. This is the "issue".

Periodically sending the state machine back to find parent state can probably lose some messages. But my knowledge about the state machine is poor.

I don't really like MY_PARENT_NODE_IS_STATIC. In case of real problem with the router, no way to communicate.

I will wait for your "I'll hack something...".  I'm sure you will get out with a nice hack.

For the moment I will check and manually reboot the nodes.