mysensors/MySensors #802: I2C EEPROM support for OTA using dual Optiboot

kisse66, 2017-03-28:
Enhancement suggestion:
I need alternate support for OTA using I2C EEPROM instead of the supported SPI Flash. Reason simply being I have boards with those (24C256/512).
I have already implemented experimental changes over development branch through not very well tested yet. Code for MySensors is almost publishable. Optiboot changes need more polishing, but work already.

So question is if there's interest in merging this feature? 

My approach was to make minimal changes to mysensors by implementing I2C EEPROM driver that looks like the SPI flash driver. There's just a few lines plus the new driver, which gets activated by a new compile time def. IF this sounds ok I can prepare it for PR.

tekka007, 2017-03-28:
@kisse66 Sounds good, please go ahead and submit your PR 👍

Tokay-tech, 2017-05-02:
Hello,
Are you still working on this enhancement?

kisse66, 2017-05-02:
Sorry for the delay. Got some personal issues and hobbies progressing slowly. I'll try to do it asap.
Code itself seems to work, but not very well tested so far. Also need to refresh the fork.

kisse66, 2017-05-02:
Updated the fork and re-tested, but now I get CRC failure after all blocks are received. Need to dig a bit deeper, can also be an issue with my board (cheap EEPROM, unfortunately soldered SOIC).
I'm testing it with arduino pro mini / nrf24. GW running on Raspberry, using MYScontroller 1.0.0.beta. This used to work. Well, tested like 3 or 4 times.

If someone wants to try the fork as-is it's now at https://github.com/kisse66/MySensors.git
Bootloader is not yet there, so one can only see if the EEPROM is handled ok. Need to define MY_OTA_USE_I2C_EEPROM, which disables SPI and uses I2C. HW I2C assumed.

kisse66, 2017-05-07:
OK, seems to work now. Just a stupid typo.
I modified the code to use extEEPROM instead of duplicating the functionality.

closed 2018-02-18 by tbowmo