mysensors/MySensors #818: New transport (clunet)

tonyich, 2017-04-09:
Is it possible to integrate new data transport into the project?
 I mean clunet or clunet 2.0:
https://github.com/ClusterM/clunet
https://github.com/dudanov/CLUNET-2.0

trlafleur, 2017-04-10:
After reviewing the CLUNET documentation, I would NOT recommend using this protocol in a network of more that a few feet in length, if that.

It connect the "network bus" directly to pin's on the processor, this will allow noise, common mode voltage and voltage spike to enter the processor and most likely they will exceed the specification of the processor I/O pin's. This over time will KILL the processor. Their are good reason that most "real network" use some type of transceiver that can buffer and isolate the processor from the real world environment. They are design to handle the noise, common voltages and voltage spikes that are real in the world, the processor is not.  As implemented, its not at all practical, especially at anywhere near it stated distance of 100 meters.

The protocol maybe good, but at best, the electrical interface stinks...

Maybe adding an RS485 transceiver to its simple interface may make this a viable protocol, but who knows.

just my two bits...

tonyich, 2017-04-11:
I pondered this problem...
I think that optocoupler solved it.
Sorry for my English.

dvv, 2017-04-12:
@trlafleur When I played with CLUNET ago it seemed like a 1-wire protocol made multi-master. And tons of projects bit-banging directly 1-wire networks do (rather) successfully exist in the wild.

trlafleur, 2017-04-17:
Yes, I'm sure it might works on a short network of a meter or so, but it will not survive in the real world at its stated range of 100 Meters...

henrikekblad, 2018-02-18:
CLUnet looks interesting. But we would need some help implement a PR (and maintain it) for this to happen. I'll close this issue for now.

closed 2018-02-18 by henrikekblad