mysensors/MySensors #822: STM32F1 OTA support?

mtiutiu, 2017-04-11:
Hi,

I know that maybe is to early to ask for this but can we have OTA implemented for STM32F1 in the future? It will be a really great addition. Maybe the existing stm32duino bootloader can be adapted to work with OTA(https://github.com/rogerclarkmelbourne/STM32duino-bootloader). I found here an example sketch for how to do flash write operations: https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/STM32F1/maple_mini_bootloaer_updater/maple_mini_bootloaer_updater.ino

I know that this may be not a very easy task to do and I will be glad to help but unfortunately my software skills are not so great at the moment...but willing to learn.
I wanted to know if this feature is feasible at least on the STM32 MCU's. Thanks.

Elektroman2, 2017-12-11:
Is there any progress in this topic? I am using Slober as IDE for mySensors development. Is that can work with OTA?
Thanks in advice, Marcell

tekka007, 2019-01-27:
Certainly of interest, but postponed indefinitly due to limited resources. PRs are welcome!

closed 2019-01-27 by tekka007