mysensors/MySensors #824: Poor range with the new RFM69 driver

user2684, 2017-04-12:
I was trying the new RFM69 driver from the dev branch but I get a shorter range compared to the old one and the result is very consistent after a few tries. To be honest the RSSI reported is very high so I can understand the signal get lost but still with the old one I could gain at least 10-15 meters more. Attaching below the logs of the gateway in case can be of any help (I restarted the node a few times while moving around, the first few lines were where I was close to the gateway, the other when moving away).
I've also tried to disable ATC and set tx power to 20dbm on both gw and node. Result is that a get as expected a much lower RSSI when close to the gw (60 instead of 80 like in the logs below) but still when moving away the situation is similar with ATC enabled. Btw if I read the code correctly, ATC is enabled on the node only so the gw will always run with 5 dbm set. Is my understanding correct?

Thanks

[rfm_logs.txt](https://github.com/mysensors/MySensors/files/916060/rfm_logs.txt)

tekka007, 2017-04-12:
A similar issue has been reported with the RFM95 driver (ref retries, see #810).
@user2684 Please describe your HW setup (MCU, RFM69 variant, power supply, etc.) and upload your sketch for testing purposes.

user2684, 2017-04-12:
Sure, my test setup is a serial gw with a RFM69HW attached on one side and a battery-powered pro mini board with another RFM69HW on the other. 
The sketch is quite complex but since all happens during presentation, an empty sketch with just a random presentation would be sufficient to reproduce. Those are the settings I used in the first test run (from which I took out the logs):
```
#define MY_RADIO_RFM69
#define MY_RFM69_FREQUENCY RFM69_868MHZ
#define MY_RFM69_NEW_DRIVER
#define MY_IS_RFM69HW
```
In the second test I mentioned above I've added on both gw and node the following even if the gw should not use ATC if I have understood correctly:
```
#define MY_RFM69_ATC_MODE_DISABLED
#define MY_RFM69_TX_POWER_DBM 20
```
I tried exactly the same hardware with the old driver a couple of times and I got consistent (different and better) results. Not to say the range is bad (I can get around 20m with two walls in between with the new driver) but at least 50% less than what I get with the old one.

tekka007, 2017-04-16:
@user2684 One of the main differences between the new and the old driver is the TX level at init (old driver uses max dBm, new driver defaults at 5dBm - applicable for non-ATC nodes & GW). However, this does not explain your second test scenario, where you set the TX level to 20dBm & disable ATC. Do you happen to have a full debug log from GW & node for both scenarios? Also, I assume you are using the latest dev code base, right?

user2684, 2017-04-21:
Thanks, will try the new driver and come back with the results. I only have the gw logs from my previous tests since to gather the logs from the node I would need to bring a laptop where the test takes place which is not handy but I'll do it and report back. Thanks again!

user2684, 2017-05-09:
Sorry for taking so long to complete this test :-/ 
Way better with https://github.com/tekka007/MySensors/tree/RFM69_Update from what I can understand from the logs. Looks like having the gw always at max tx power and the sensor starting with max power and than decreasing if needed is more effective than the other way around.
I've collected both the gw and the sensor's logs this time in two scenarios: 1- close one to each other 2- the sensor far away from the gw. In the first scenario I see the sensor adjusting the tx power. There is a single retry but nothing serious I think. In the second scenario the tx power remains at max the rssi stays around 90 and there is no re-transmission which didn't happen during my first test.

[close_gw.txt](https://github.com/mysensors/MySensors/files/986797/close_gw.txt)
[close_sensor.txt](https://github.com/mysensors/MySensors/files/986798/close_sensor.txt)
[away_gw.txt](https://github.com/mysensors/MySensors/files/986800/away_gw.txt)
[away_sensor.txt](https://github.com/mysensors/MySensors/files/986801/away_sensor.txt)

user2684, 2017-05-09:
Forgot to add both the gw and the sensor are running at 1Mhz in this test without a problem which was my original issue with the 2.1.1 driver

closed 2018-02-18 by tekka007