mysensors/MySensors #831: RPI gateway (GPIO) - Segmentation fault

bilbolodz, 2017-04-27:
I've playing with OTA using MYSController. I've noticed that sometime after OTA node upgrade or "reboot" command from MYSController node has problem with connecting to controller and finally gateway process on RPI got "Segmentation fault" and dies. Restarting gateway process won't fix it (still seg faults). Only way to get GW running is physical NODE reset. Logs from GW:

mysgw: Starting gateway...
mysgw: Protocol version - 2.1.1
mysgw: MCO:BGN:INIT GW,CP=RNNG-SQ,VER=2.1.1
mysgw: TSF:LRT:OK
mysgw: TSM:INIT
mysgw: TSF:WUR:MS=0
mysgw: TSM:INIT:TSP OK
mysgw: TSM:INIT:GW MODE
mysgw: TSM:READY:ID=0,PAR=0,DIS=0
mysgw: MCO:REG:NOT NEEDED
mysgw: Listening for connections on 0.0.0.0:5003
mysgw: MCO:BGN:STP
mysgw: MCO:BGN:INIT OK,TSP=1
mysgw: TSF:MSG:READ,1-1-255,s=255,c=3,t=7,pt=0,l=0,sg=0:
mysgw: TSF:MSG:BC
mysgw: TSF:MSG:FPAR REQ,ID=1
mysgw: TSF:CKU:OK,FCTRL
mysgw: TSF:MSG:GWL OK
mysgw: !TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=NACK:0
mysgw: TSF:MSG:READ,1-1-255,s=255,c=3,t=7,pt=0,l=0,sg=0:
mysgw: TSF:MSG:BC
mysgw: TSF:MSG:FPAR REQ,ID=1
mysgw: TSF:CKU:OK,FCTRL
mysgw: TSF:MSG:GWL OK
mysgw: TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=OK:0
mysgw: TSF:MSG:READ,1-1-255,s=255,c=3,t=7,pt=0,l=0,sg=0:
mysgw: TSF:MSG:BC
mysgw: TSF:MSG:FPAR REQ,ID=1
mysgw: TSF:CKU:OK,FCTRL
mysgw: TSF:MSG:GWL OK
mysgw: TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=OK:0
mysgw: TSF:MSG:READ,1-1-255,s=255,c=3,t=7,pt=0,l=0,sg=0:
mysgw: TSF:MSG:BC
mysgw: TSF:MSG:FPAR REQ,ID=1
mysgw: TSF:CKU:OK,FCTRL
mysgw: TSF:MSG:GWL OK
mysgw: TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=OK:0
mysgw: TSF:MSG:READ,1-1-255,s=255,c=3,t=7,pt=0,l=0,sg=0:
mysgw: TSF:MSG:BC
mysgw: TSF:MSG:FPAR REQ,ID=1
mysgw: TSF:CKU:OK,FCTRL
mysgw: TSF:MSG:GWL OK
mysgw: TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=OK:0
mysgw: TSF:MSG:READ,1-1-255,s=255,c=3,t=7,pt=0,l=0,sg=0:
mysgw: TSF:MSG:BC
mysgw: TSF:MSG:FPAR REQ,ID=1
mysgw: TSF:CKU:OK,FCTRL
mysgw: TSF:MSG:GWL OK
mysgw: !TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=NACK:0
mysgw: TSF:MSG:READ,1-1-255,s=255,c=3,t=7,pt=0,l=0,sg=0:
mysgw: TSF:MSG:BC
mysgw: TSF:MSG:FPAR REQ,ID=1
mysgw: TSF:CKU:OK,FCTRL
mysgw: TSF:MSG:GWL OK
mysgw: !TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=NACK:0
mysgw: TSF:MSG:READ,1-1-255,s=255,c=3,t=7,pt=0,l=0,sg=0:
mysgw: TSF:MSG:BC
mysgw: TSF:MSG:FPAR REQ,ID=1
mysgw: TSF:CKU:OK,FCTRL
mysgw: TSF:MSG:GWL OK
Segmentation fault

denissp, 2017-04-28:
Hi

I have the same problem
mysgw: !TSF:MSG:LEN,0!=7
mysgw: TSF:MSG:READ,1-1-1,s=1,c=1,t=1,pt=0,l=0,sg=0:
mysgw: !TSF:MSG:LEN,8!=7
mysgw: TSF:MSG:READ,1-1-1,s=1,c=1,t=1,pt=0,l=0,sg=0:
mysgw: !TSF:MSG:LEN,0!=7
mysgw: TSF:MSG:READ,255-255-255,s=255,c=7,t=255,pt=7,l=25,sg=1: 0
mysgw: !TSF:MSG:LEN,7!=32
mysgw: TSF:MSG:READ,255-21-2,s=0,c=7,t=255,pt=0,l=0,sg=0:
mysgw: !TSF:MSG:PVER,3=2
mysgw: TSF:MSG:READ,0-0-0,s=0,c=0,t=0,pt=0,l=0,sg=0:
mysgw: !TSF:MSG:PVER,0=2
mysgw: TSF:MSG:READ,0-0-0,s=0,c=0,t=0,pt=0,l=0,sg=0:
mysgw: !TSF:MSG:LEN,0!=7
Segmentation fault

marceloaqno, 2017-04-28:
What ./configure options are you using?
Could you test with the development branch to see if the segmentation fault also occurs?

denissp, 2017-04-28:
./configure --my-gateway=mqtt --my-controller-ip-address=127.0.0.1 --my-mqtt-publish-topic-prefix=mygateway1-out --my-mqtt-subscribe-topic-prefix=mygateway1-in --my-mqtt-client-id=mygateway1

I'll try development branch on next tuesday

bilbolodz, 2017-04-28:
My options are:

./configure --my-transport=nrf24 --my-leds-err-pin=29 --my-leds-rx-pin=7 --my-leds-tx-pin=11 --my-rf24-irq-pin=15 --my-rf24-pa-level=RF24_PA_MIN --my-signing=software

denissp, 2017-05-04:
Hi

Dev branch work fine for 2 days.

Regards

closed 2017-05-04 by henrikekblad