mysensors/MySensors #837: The Sensebender gateway freezes after a short time

micropet, 2017-05-12:
I got a Sensebender gateway. Mounted the W5100 Ethernet board and a RFM96HW 868 MHz Module. 
I use a sketch with the WizNET (W5100) Ethernet module, RFM96 and MQTT.

The connection to the MQTT server was established immediately. 
My Dallas Sensor Node sends the temperature via Sensebender Gateway to the MQTT Server. 
Everything was perfect. The gateway works as desired.

Unfortunately only for about 2 hours, then it freezes.
After a restart it runs again 2-3 hours and freezes again.
And so on.

The Sensebender gateway has only one DS18b20 node for testing.

What kind of information is needed?

Peter

trlafleur, 2017-05-12:
This is similar to what I'm seeing in issue 810 when using the new RFM95 driver... in my case, it can be a few hours, to a day or so....

You might want to turn on debug code and monitor it output....

micropet, 2017-05-12:
Thank you. This is the Output util it hangs:

0;255;3;0;9;2749 Attempting MQTT connection...
0;255;3;0;9;2762 MQTT connected
0;255;3;0;9;2772 Sending message on topic: mysensors2-out/0/255/0/0/18
0;255;3;0;9;36433 TSF:MSG:READ,101-101-255,s=255,c=3,t=7,pt=0,l=0,sg=0:
0;255;3;0;9;36443 TSF:MSG:BC
0;255;3;0;9;36453 TSF:MSG:FPAR REQ,ID=101
0;255;3;0;9;36463 TSF:PNG:SEND,TO=0
0;255;3;0;9;36473 TSF:CKU:OK
0;255;3;0;9;36483 TSF:MSG:GWL OK
0;255;3;0;9;37143 TSF:MSG:SEND,0-0-101-101,s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=OK:0
0;255;3;0;9;38509 TSF:MSG:READ,101-101-0,s=255,c=3,t=24,pt=1,l=1,sg=0:1
0;255;3;0;9;38519 TSF:MSG:PINGED,ID=101,HP=1
0;255;3;0;9;38547 TSF:MSG:SEND,0-0-101-101,s=255,c=3,t=25,pt=1,l=1,sg=0,ft=0,st=OK:1
0;255;3;0;9;51551 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:21.2
0;255;3;0;9;51561 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;71552 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:21.2
0;255;3;0;9;71562 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;91557 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:21.2
0;255;3;0;9;91567 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;111557 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:21.2
0;255;3;0;9;111567 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;131566 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:21.2
0;255;3;0;9;131576 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;151562 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:21.2
0;255;3;0;9;151572 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;171564 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:21.2
0;255;3;0;9;171574 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;191566 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:21.2
0;255;3;0;9;191576 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;211568 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:21.2
0;255;3;0;9;211578 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;231571 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:21.2
0;255;3;0;9;231581 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;251580 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:21.2
0;255;3;0;9;251590 Sending message on topic: mysensors2-out/101/0/1/0/0

trlafleur, 2017-05-12:
~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

trlafleur, 2017-05-12:
Add this to your code to enable more debug info

~~~
// Enable debug prints
#define MY_DEBUG
#define MY_DEBUG_VERBOSE_RFM95
~~~

micropet, 2017-05-13:
Here the Log with #define MY_DEBUG_VERBOSE_RFM69 until it freezes:

0;255;3;0;9;37878 TSF:MSG:READ,101-101-255,s=255,c=3,t=7,pt=0,l=0,sg=0:
0;255;3;0;9;37888 TSF:MSG:BC
0;255;3;0;9;37898 TSF:MSG:FPAR REQ,ID=101
0;255;3;0;9;37908 TSF:PNG:SEND,TO=0
0;255;3;0;9;37918 TSF:CKU:OK
0;255;3;0;9;37928 TSF:MSG:GWL OK
0;255;3;0;9;37978 RFM69:SWR:SEND,TO=101,RETRY=0
0;255;3;0;9;37996 RFM69:SWR:ACK,FROM=101,SEQ=1,RSSI=-31
0;255;3;0;9;38006 TSF:MSG:SEND,0-0-101-101,s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=OK:0
0;255;3;0;9;44793 TSF:MSG:READ,101-101-255,s=255,c=3,t=7,pt=0,l=0,sg=0:
0;255;3;0;9;44803 TSF:MSG:BC
0;255;3;0;9;44813 TSF:MSG:FPAR REQ,ID=101
0;255;3;0;9;44823 TSF:CKU:OK,FCTRL
0;255;3;0;9;44833 TSF:MSG:GWL OK
0;255;3;0;9;45644 RFM69:SWR:SEND,TO=101,RETRY=0
0;255;3;0;9;45662 RFM69:SWR:ACK,FROM=101,SEQ=2,RSSI=-32
0;255;3;0;9;45672 TSF:MSG:SEND,0-0-101-101,s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=OK:0
0;255;3;0;9;46861 RFM69:SAC:SEND ACK,TO=101,RSSI=-37
0;255;3;0;9;46874 TSF:MSG:READ,101-101-0,s=255,c=3,t=24,pt=1,l=1,sg=0:1
0;255;3;0;9;46884 TSF:MSG:PINGED,ID=101,HP=1
0;255;3;0;9;46899 RFM69:SWR:SEND,TO=101,RETRY=0
0;255;3;0;9;46915 RFM69:SWR:ACK,FROM=101,SEQ=4,RSSI=-35
0;255;3;0;9;46925 TSF:MSG:SEND,0-0-101-101,s=255,c=3,t=25,pt=1,l=1,sg=0,ft=0,st=OK:1
0;255;3;0;9;46988 RFM69:SAC:SEND ACK,TO=101,RSSI=-37
0;255;3;0;9;47001 TSF:MSG:READ,101-101-0,s=255,c=3,t=15,pt=6,l=2,sg=0:0100
0;255;3;0;9;47011 RFM69:SWR:SEND,TO=101,RETRY=0
0;255;3;0;9;47029 RFM69:SWR:ACK,FROM=101,SEQ=6,RSSI=-34
0;255;3;0;9;47039 TSF:MSG:SEND,0-0-101-101,s=255,c=3,t=15,pt=6,l=2,sg=0,ft=0,st=OK:0100
0;255;3;0;9;47099 RFM69:SAC:SEND ACK,TO=101,RSSI=-38
0;255;3;0;9;47112 TSF:MSG:READ,101-101-0,s=255,c=0,t=17,pt=0,l=10,sg=0:2.2.0-beta
0;255;3;0;9;47122 Sending message on topic: mysensors2-out/101/255/0/0/17
0;255;3;0;9;47183 RFM69:SAC:SEND ACK,TO=101,RSSI=-37
0;255;3;0;9;47196 TSF:MSG:READ,101-101-0,s=255,c=3,t=6,pt=1,l=1,sg=0:0
0;255;3;0;9;47206 Sending message on topic: mysensors2-out/101/255/3/0/6
0;255;3;0;9;49260 RFM69:SAC:SEND ACK,TO=101,RSSI=-33
0;255;3;0;9;49273 TSF:MSG:READ,101-101-0,s=255,c=3,t=11,pt=0,l=18,sg=0:Temperature Sensor
0;255;3;0;9;49283 Sending message on topic: mysensors2-out/101/255/3/0/11
0;255;3;0;9;49344 RFM69:SAC:SEND ACK,TO=101,RSSI=-35
0;255;3;0;9;49357 TSF:MSG:READ,101-101-0,s=255,c=3,t=12,pt=0,l=3,sg=0:1.2
0;255;3;0;9;49367 Sending message on topic: mysensors2-out/101/255/3/0/12
0;255;3;0;9;49428 RFM69:SAC:SEND ACK,TO=101,RSSI=-36
0;255;3;0;9;49442 TSF:MSG:READ,101-101-0,s=0,c=0,t=6,pt=0,l=0,sg=0:
0;255;3;0;9;49452 Sending message on topic: mysensors2-out/101/0/0/0/6
0;255;3;0;9;49513 RFM69:SAC:SEND ACK,TO=101,RSSI=-37
0;255;3;0;9;49526 TSF:MSG:READ,101-101-0,s=255,c=3,t=26,pt=1,l=1,sg=0:2
0;255;3;0;9;49541 RFM69:SWR:SEND,TO=101,RETRY=0
0;255;3;0;9;49561 RFM69:SWR:ACK,FROM=101,SEQ=13,RSSI=-34
0;255;3;0;9;49571 TSF:MSG:SEND,0-0-101-101,s=255,c=3,t=27,pt=1,l=1,sg=0,ft=0,st=OK:1
0;255;3;0;9;65502 RFM69:SAC:SEND ACK,TO=101,RSSI=-39
0;255;3;0;9;65665 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:23.2
0;255;3;0;9;65675 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;85495 RFM69:SAC:SEND ACK,TO=101,RSSI=-39
0;255;3;0;9;85509 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:23.2
0;255;3;0;9;85519 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;105494 RFM69:SAC:SEND ACK,TO=101,RSSI=-40
0;255;3;0;9;105507 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:23.2
0;255;3;0;9;105517 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;125493 RFM69:SAC:SEND ACK,TO=101,RSSI=-42
0;255;3;0;9;125507 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:23.2
0;255;3;0;9;125517 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;145493 RFM69:SAC:SEND ACK,TO=101,RSSI=-42
0;255;3;0;9;145507 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:23.2
0;255;3;0;9;145517 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;165492 RFM69:SAC:SEND ACK,TO=101,RSSI=-44
0;255;3;0;9;165506 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:23.2
0;255;3;0;9;165516 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;185492 RFM69:SAC:SEND ACK,TO=101,RSSI=-45
0;255;3;0;9;185506 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:23.2
[... 349 lines cut ...]
0;255;3;0;9;1625734 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;1645714 RFM69:SAC:SEND ACK,TO=101,RSSI=-51
0;255;3;0;9;1645727 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:23.4
0;255;3;0;9;1645737 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;1665716 RFM69:SAC:SEND ACK,TO=101,RSSI=-51
0;255;3;0;9;1665729 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:23.4
0;255;3;0;9;1665739 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;1685719 RFM69:SAC:SEND ACK,TO=101,RSSI=-51
0;255;3;0;9;1685732 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:23.4
0;255;3;0;9;1685742 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;1705720 RFM69:SAC:SEND ACK,TO=101,RSSI=-51
0;255;3;0;9;1705734 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:23.4
0;255;3;0;9;1705744 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;1725721 RFM69:SAC:SEND ACK,TO=101,RSSI=-51
0;255;3;0;9;1725734 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:23.4
0;255;3;0;9;1725744 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;1745723 RFM69:SAC:SEND ACK,TO=101,RSSI=-52
0;255;3;0;9;1745737 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:23.4
0;255;3;0;9;1745747 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;1765726 RFM69:SAC:SEND ACK,TO=101,RSSI=-51
0;255;3;0;9;1765739 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:23.3
0;255;3;0;9;1765749 Sending message on topic: mysensors2-out/101/0/1/0/0
0;255;3;0;9;1785727 RFM69:SAC:SEND ACK,TO=101,RSSI=-50
0;255;3;0;9;1785741 TSF:MSG:READ,101-101-0,s=0,c=1,t=0,pt=7,l=5,sg=0:23.4
0;255;3;0;9;1785751 Sending message on topic: mysensors2-out/101/0/1/0/0

trlafleur, 2017-05-13:
Its interesting to note, that re-try's are always 5(the max) and never 1 to 4... same as I was seeing with the RFM95 radio...

Looks like the retry logic need some work...

tbowmo, 2017-05-14:
@micropet are you using the GatewayW5100MQTT example sketch as is, or a modified one? If modified, then please post your sketch as well..

micropet, 2017-05-14:
Here is the Sketch i use:

`
#define MY_DEBUG
#define SKETCH_VERSION "0.2"
#define MY_RADIO_RFM69
#define MY_RFM69_NEW_DRIVER
#define MY_IS_RFM69HW                         // mit RFM69HCW
#define MY_RFM69_FREQUENCY  (868200000ul)     // default  #define RFM69_868MHZ            (868000000ul)
#define MY_RFM69_TX_POWER_DBM          1      // TX power level     default 5dBm (overriden if ATC mode enabled) (0..31)
#define MY_DEBUG_VERBOSE_RFM69
// ---------------------------------
// Enable inclusion mode
#define MY_INCLUSION_MODE_FEATURE
// Enable Inclusion mode button on gateway
#define MY_INCLUSION_BUTTON_FEATURE

// Inverses behavior of inclusion button (if using external pullup)
//#define MY_INCLUSION_BUTTON_EXTERNAL_PULLUP

// Set inclusion mode duration (in seconds)
#define MY_INCLUSION_MODE_DURATION 60
// Digital pin used for inclusion mode button
//#define MY_INCLUSION_MODE_BUTTON_PIN  3

#define MY_DEFAULT_LED_BLINK_PERIOD 300
#define MY_DEFAULT_LED_BLINK_PERIOD   400
#define MY_DEFAULT_RX_LED_PIN         MYSX_D3_INT      // gruen  Receive led pin
#define MY_DEFAULT_TX_LED_PIN         MYSX_D4_INT      // orange Transmit led pin
#define MY_DEFAULT_ERR_LED_PIN        MYSX_D5_PWM      // rot    Error led pin
// ---------------------------------

#define MY_GATEWAY_MQTT_CLIENT
#define MY_MQTT_PUBLISH_TOPIC_PREFIX   "mysensors2-out"
#define MY_MQTT_SUBSCRIBE_TOPIC_PREFIX "mysensors2-in"
#define MY_MQTT_CLIENT_ID              "SAMD21-01"

//#define MY_GATEWAY_CLIENT_MODE
#define MY_IP_ADDRESS 192,168,0,103
#define MY_MAC_ADDRESS 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED
#define MY_IP_GATEWAY_ADDRESS 192,168,0,3
#define MY_IP_SUBNET_ADDRESS 255,255,255,0
#define MY_CONTROLLER_IP_ADDRESS 192,168,0,23
#define MY_PORT 1883

// ---------------------------------
#include <Ethernet.h>
#include <MySensors.h>
// ---------------------------------

#define EEPROM_VERIFICATION_ADDRESS 0x01

static uint8_t num_of_leds = 5;
static uint8_t leds[] = {LED_BLUE, LED_RED, LED_GREEN, LED_YELLOW, LED_ORANGE};

void setup()
{
    // Setup locally attached sensors
}

void presentation()
{
    // Present locally attached sensors
}

void loop()
{
    // Send locally attached sensor data here
}

void preHwInit()
{

    pinMode(MY_SWC1, INPUT_PULLUP);
    pinMode(MY_SWC2, INPUT_PULLUP);
    if (digitalRead(MY_SWC1) && digitalRead(MY_SWC2)) {
        return;
    }

    uint8_t tests = 0;

    for (int i=0; i< num_of_leds; i++) {
        pinMode(leds[i], OUTPUT);
    }
    uint8_t led_state = 0;
    if (digitalRead(MY_SWC1)) {
        while (!Serial) {
            digitalWrite(LED_BLUE, led_state);
            led_state ^= 0x01;
            delay(500);
        } // Wait for USB to be connected, before spewing out data.
    }
    digitalWrite(LED_BLUE, LOW);
    if (Serial) {
        Serial.println("Sensebender GateWay test routine");
        Serial.print("Mysensors core version : ");
        Serial.println(MYSENSORS_LIBRARY_VERSION);
        Serial.print("GateWay sketch version : ");
        Serial.println(SKETCH_VERSION);
        Serial.println("----------------------------------");
        Serial.println();
    }
//    if (testSha204()) {
        digitalWrite(LED_GREEN, HIGH);
        tests++;
//    }
//    if (testSDCard()) {
        digitalWrite(LED_YELLOW, HIGH);
        tests++;
//    }

    if (testEEProm()) {
        digitalWrite(LED_ORANGE, HIGH);
        tests++;
    }
    if (testAnalog()) {
        digitalWrite(LED_BLUE, HIGH);
        tests++;
    }
    if (tests == 4) {
        while(1) {
            for (int i=0; i<num_of_leds; i++) {
                digitalWrite(leds[i], HIGH);
                delay(200);
                digitalWrite(leds[i], LOW);
            }
        }
    } else {
        while (1) {
            digitalWrite(LED_RED, HIGH);
            delay(200);
            digitalWrite(LED_RED, LOW);
            delay(200);
        }
    }

}

bool testSha204()
{
    uint8_t rx_buffer[SHA204_RSP_SIZE_MAX];
    uint8_t ret_code;
    if (Serial) {
        Serial.print("- > SHA204 ");
    }
    atsha204_init(MY_SIGNING_ATSHA204_PIN);
    ret_code = atsha204_wakeup(rx_buffer);

    if (ret_code == SHA204_SUCCESS) {
        ret_code = atsha204_getSerialNumber(rx_buffer);
        if (ret_code != SHA204_SUCCESS) {
            if (Serial) {
                Serial.println(F("Failed to obtain device serial number. Response: "));
            }
            Serial.println(ret_code, HEX);
        } else {
            if (Serial) {
                Serial.print(F("Ok (serial : "));
                for (int i=0; i<9; i++) {
                    if (rx_buffer[i] < 0x10) {
                        Serial.print('0'); // Because Serial.print does not 0-pad HEX
                    }
                    Serial.print(rx_buffer[i], HEX);
                }
                Serial.println(")");
            }
            return true;
        }
    } else {
        if (Serial) {
            Serial.println(F("Failed to wakeup SHA204"));
        }
    }
    return false;
}

bool testEEProm()
{
    uint8_t eeprom_d1, eeprom_d2;
    SerialUSB.print(" -> EEPROM ");
    eeprom_d1 = hwReadConfig(EEPROM_VERIFICATION_ADDRESS);
    delay(500);
    eeprom_d1 = ~eeprom_d1; // invert the bits
    hwWriteConfig(EEPROM_VERIFICATION_ADDRESS, eeprom_d1);
    delay(500);
    eeprom_d2 = hwReadConfig(EEPROM_VERIFICATION_ADDRESS);
    if (eeprom_d1 == eeprom_d2) {
        SerialUSB.println("PASSED");
        hwWriteConfig(EEPROM_VERIFICATION_ADDRESS, ~eeprom_d1);
        return true;
    }
    SerialUSB.println("FAILED!");
    return false;
}

bool testAnalog()
{
    int bat_detect = analogRead(MY_BAT_DETECT);
    Serial.print("-> analog : ");
    Serial.print(bat_detect);
    if (bat_detect < 400 || bat_detect > 650) {
        Serial.println(" Failed");
        return false;
    }
    Serial.println(" Passed");
    return true;
}`

bblacey, 2017-05-27:
I think we have narrowed down the root cause and fixed it in #866 that was just merged.  Give it a test and report back.

micropet, 2017-05-29:
Thank you - testing since 1h. Be back soon.

micropet, 2017-05-30:
It looks good. The gateway has been running for 12 hours.

Now only my RFM95 gateway should work, then I am satisfied.
This also seems to be a software bug.

micropet, 2017-05-31:
Almost 2 days and it still works.
I think it's fine.
Many Thanks.

closed 2017-05-31 by micropet