mysensors/MySensors #844: MQTT Auto node id

MPParsley, 2017-05-14:
When using several hundred nodes that can be relocated it would be nice to have the auto node id feature to work.

henrikekblad, 2017-05-14:
Not sure what you mean here.. It is up to your controller to hand out ids. Works exactly the same over MQTT as the MySensors protocol.

MPParsley, 2017-05-14:
Handing out the ID is indeed up to the controller but the documentation is somewhat unclear on this.

According to [this documentation](https://www.mysensors.org/build/mqtt_gateway) Auto ID is not supported through MQTT:

> NOTE: No controller supports dynamic ID assignment through MQTT. All nodes must have MY_NODE_ID defined in the sketch to work with MQTT. If you don't set MY_NODE_ID, nodes will complain with the message "!TSM:ID:FAIL".

Yet, the [documentation here](https://www.mysensors.org/controller) states that "node ID generation" & "MQTT gateway" it is supported by these controllers:
* Domoticz
* HomeGenie
* MyController.org
* OpenHAB 2.x
* MyHouse

So as it is supported by these controllers I assumed some code would be needed on the Mysensors side for MQTT?

MPParsley, 2017-05-16:
Closing this as it seems to be implemented on the MySensors side.

FYI: I tested this with mycontroller and as of the latest (build) version it supports auto node id generation with MQTT.

trlafleur, 2017-05-17:
Can you please state how you tested this, as I'm looking at the code, I see no logic for storing or giving out of client ID in the MQTT examples or MQTT transport code.

Thanks

MPParsley, 2017-06-30:
Hi @trlafleur,

In the node sketch I put `#define MY_NODE_ID AUTO`.

Note that you'll need to build the latest version of MyController (https://github.com/mycontroller-org/mycontroller) or you'll have to ask its maintainer to create a new release.

closed 2017-06-30 by MPParsley