mysensors/MySensors #854: There ist an Error compiling  with Linux for RFM95 and RPI

micropet, 2017-05-21:
I tried to compile my RFM95 Gateway with:
./configure --my-gateway=mqtt --my-controller-ip-address=192.168.0.23 --my-mqtt-publish-topic-prefix=mysensors2-out --my-mqtt-subscribe-topic-prefix=mysensors2-in --my-mqtt-client-id=mygateway5 --my-transport=rfm95 --my-leds-err-pin=12 --my-leds-rx-pin=16 --my-leds-tx-pin=18

I get:

g++ -MT build/examples_linux/mysgw.o -MMD -MP -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -DMY_RADIO_RFM95 -DMY_GATEWAY_LINUX -DMY_GATEWAY_MQTT_CLIENT -DMY_DEBUG -DLINUX_SPI_BCM -DLINUX_ARCH_RASPBERRYPI -DRPI_TYPE=\"rpi2\" -DMY_DEFAULT_TX_LED_PIN=18 -DMY_DEFAULT_RX_LED_PIN=16 -DMY_DEFAULT_ERR_LED_PIN=12 -DMY_MQTT_CLIENT_ID=\"mygateway5\" -DMY_MQTT_SUBSCRIBE_TOPIC_PREFIX=\"mysensors2-in\" -DMY_MQTT_PUBLISH_TOPIC_PREFIX=\"mysensors2-out\" -DMY_CONTROLLER_IP_ADDRESS=192,168,0,23  -Ofast -g -Wall -Wextra  -I. -I./core -I./drivers/Linux -I./drivers/BCM -c examples_linux/mysgw.cpp -o build/examples_linux/mysgw.o
In file included from ./MySensors.h:109:0,
                 from examples_linux/mysgw.cpp:83:
./hal/architecture/MyHwLinuxGeneric.cpp: In function ‘ssize_t hwGetentropy(void*, size_t)’:
./hal/architecture/MyHwLinuxGeneric.cpp:79:17: error: ‘SYS_getrandom’ was not declared in this scope
  return syscall(SYS_getrandom, __buffer, __length, GRND_NONBLOCK);
                 ^
./hal/architecture/MyHwLinuxGeneric.cpp:79:52: error: ‘GRND_NONBLOCK’ was not declared in this scope
  return syscall(SYS_getrandom, __buffer, __length, GRND_NONBLOCK);
                                                    ^
./hal/architecture/MyHwLinuxGeneric.cpp:80:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
Makefile:98: recipe for target 'build/examples_linux/mysgw.o' failed
make: *** [build/examples_linux/mysgw.o] Error 1

tekka007, 2017-05-21:
@d00616 Can you have a look at this as it seems related to #845?

d00616, 2017-05-21:
I have created https://github.com/mysensors/MySensors/pull/856
This PR replaces getrandom() with reading from /dev/urandom

john4242, 2017-05-21:
I had the same error, #856 fixes it, but I think #856 should use a define, just in case some systems do have SYS_getrandom

marceloaqno, 2017-05-21:
What about getrandom() ?
"getrandom() was introduced in version 3.17 of the Linux kernel."
http://manpages.ubuntu.com/manpages/xenial/man2/getrandom.2.html

marceloaqno, 2017-05-22:
getrandom() won't help, it seems to need sys/random.h and again, not every system has it.

closed 2017-05-23 by marceloaqno