mysensors/MySensors #862: SecurityPersonalizer: 'EEPROM_PERSONALIZATION_CHECKSUM_ADDRESS' was not declared in this scope

SergioRius, 2017-05-24:
Compiling SecurityPersonalizer.ino throws: 'EEPROM_PERSONALIZATION_CHECKSUM_ADDRESS' was not declared in this scope At line 467.

Relevant dump:
`...\SecurityPersonalizer\SecurityPersonalizer.ino: In function 'void write_eeprom_checksum()':

SecurityPersonalizer:695: error: 'EEPROM_PERSONALIZATION_CHECKSUM_ADDRESS' was not declared in this scope

  hwWriteConfigBlock((void*)&hash[0], (void*)EEPROM_PERSONALIZATION_CHECKSUM_ADDRESS, 1);`

Seems that this file is in a pre-release state, but it's the one referenced at the docs in the mysensors website, and the recommended at the forum.

mfalkvidd, 2017-05-25:
@SergioRius which version/branch are you using?

fallberg, 2017-06-06:
This is not a problem on the development branch at least. Please use that one for beta-firmware.

closed 2017-06-06 by fallberg

SergioRius, 2017-06-07:
Sorry guys for the late answer, I've been very ill (still do).
Go to https://www.mysensors.org/about/signing#relevant-sketches and copy the SecurityPersonalizer sketch there.
Then fix the new lib error by replacing (at 40): "#include "sha204_lib_return_codes.h"" for "#include "sha204_includes/sha204_lib_return_codes.h"" as this is how the library is distributed.

Then just compile to get:
[image]

Perhaps an include is missing? <EEPROM.h>?
I've tried to fix it but still I don't understand well what's going on there.

fallberg, 2017-06-08:
You can't just pick any ino file and compile it. You need to make sure you have the necessary libraries. And that the libraries are up2date and in sync with the expectations of the ino file.
In the case of EEPROM_PERSONALIZATION_CHECKSUM_ADDRESS, it is available on the development branch of the library, and it is defined [here](https://github.com/mysensors/MySensors/blob/development/core/MyEepromAddresses.h#L66).

fallberg, 2017-06-08:
Regarding the references on the page, you are correct, and I have updated the references there.

SergioRius, 2017-06-08:
I see we finally understood each other. Thanks. And sorry for the unconvenience.