mysensors/MySensors #863: RPI Gateway with RFM95 does not receive any nodes

micropet, 2017-05-26:
I have built many gateways with an RPI2 and the RFM96. All work well. Now I wanted to test an RFM95 gateway.
The software compiles and starts properly.
My 3 RFM95 Nodes get but no connection.
Has anyone ever run an RPI gateway with RFM95?

Compiled with:

.myconfigure --my-gateway = mqtt --my-controller-ip-address = 192.168.0.23 --my-mqtt-publish-topic-prefix = mysensors2-out --my-mqtt-subscribe-topic-prefix = I have the same problem with mymy-leds -my-leds-err-pin = 12 - my-leds-rx-pin = 16 --my-leds-tx -pin = 18

The wiring:

Raspberry	RFM69	
GND	GND	
3.3V	        VCC
24		NSS	
23		SCK	
19		MOSI	
21		MISO	
22		DI00	

18		TX	Green
16		RX	Yellow
12		ERR	Red

mfalkvidd, 2017-05-26:
@micropet could you post the debug output from one of the nodes?

micropet, 2017-05-26:
On a Teensy, which runs on the ttn Network the last weeks i got:
Transport device initialization failed.

I have:
#define MY_RFM95_CS_PIN                 10     // NSS
#define MY_RFM95_IRQ_PIN                 6     // IRQ DIO0
#define MY_RFM95_RST_PIN                 4     // Reset

Will try it on a Arduino Pro. 

1401 MCO:BGN:INIT NODE,CP=RLNN----,VER=2.2.0-beta
1401 TSM:INIT
1401 TSF:WUR:MS=0
1401 RFM95:INIT
1406 RFM95:INIT:PIN,CS=10,IQP=6,IQN=8,RST=4
1416 !TSM:INIT:TSP FAIL
1416 TSM:FAIL:CNT=1
1416 TSM:FAIL:DIS
1416 TSF:TDI:TSL
1416 RFM95:RSL
11417 TSM:FAIL:RE-INIT
11417 TSM:INIT
11417 RFM95:INIT
11422 RFM95:INIT:PIN,CS=10,IQP=6,IQN=8,RST=4
11432 !TSM:INIT:TSP FAIL
11432 TSM:FAIL:CNT=2
11432 TSM:FAIL:DIS
11432 TSF:TDI:TSL
11432 RFM95:RSL
21433 TSM:FAIL:RE-INIT
21433 TSM:INIT
21433 RFM95:INIT
21438 RFM95:INIT:PIN,CS=10,IQP=6,IQN=8,RST=4
21448 !TSM:INIT:TSP FAIL
21448 TSM:FAIL:CNT=3
21448 TSM:FAIL:DIS
21448 TSF:TDI:TSL
21448 RFM95:RSL
31449 TSM:FAIL:RE-INIT
31449 TSM:INIT
31449 RFM95:INIT
31454 RFM95:INIT:PIN,CS=10,IQP=6,IQN=8,RST=4
31464 !TSM:INIT:TSP FAIL
31464 TSM:FAIL:CNT=4
31464 TSM:FAIL:DIS
31464 TSF:TDI:TSL
31464 RFM95:RSL

micropet, 2017-05-26:
with an Arduino Pro a little better - but no connect:

0 MCO:BGN:INIT NODE,CP=RLNNA---,VER=2.2.0-beta
49 TSM:INIT
51 TSF:WUR:MS=0
51 RFM95:INIT
53 RFM95:INIT:PIN,CS=10,IQP=4,IQN=0
67 RFM95:PTX:LEVEL=5
69 TSM:INIT:TSP OK
71 TSM:INIT:STATID=150
77 TSF:SID:OK,ID=150
79 TSM:FPAR
79 RFM95:SWR:SEND,TO=255,SEQ=0,RETRY=0
2084 TSF:MSG:SEND,150-150-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
4096 !TSM:FPAR:NO REPLY
4098 TSM:FPAR
4100 RFM95:SWR:SEND,TO=255,SEQ=0,RETRY=0
6105 TSF:MSG:SEND,150-150-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
8114 !TSM:FPAR:NO REPLY
8116 TSM:FPAR
8118 RFM95:SWR:SEND,TO=255,SEQ=0,RETRY=0
10123 TSF:MSG:SEND,150-150-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
12132 !TSM:FPAR:NO REPLY
12134 TSM:FPAR
12136 RFM95:SWR:SEND,TO=255,SEQ=0,RETRY=0
14141 TSF:MSG:SEND,150-150-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
16150 !TSM:FPAR:FAIL
16152 TSM:FAIL:CNT=1
16154 TSM:FAIL:DIS
16156 TSF:TDI:TSL
16158 RFM95:RSL

micropet, 2017-06-05:
Can no one help?

Does not someone use an RFM95?
Also the developers do not?

mfalkvidd, 2017-07-06:
The rfm95 gateway works for me, see https://forum.mysensors.org/topic/7022/rpi-gateway-with-rfm95-does-not-receive-any-nodes/ for details.

Could you check your configuration settings and post the gateway log in the forum?

mfalkvidd, 2017-07-08:
Problem located. There was no problem with the gateway. See the linked thread for details.

Closing.

closed 2017-07-08 by mfalkvidd