mysensors/MySensors #870: RPi: Stops receiving messages after some time when using nrf24 interrupts

marceloaqno, 2017-05-31:
This problem should have been fixed with #835, but my gateway still suffers from it although less frequent.
Related to #798.

@bathoryr Have you had any problems recently?

mfalkvidd, 2017-06-01:
Could this be the same problem as https://github.com/mysensors/MySensors/pull/866 or are you using a version where that fix is included?

marceloaqno, 2017-06-01:
@mfalkvidd extEEPROM is not used by the RPi, it must be something else.

bathoryr, 2017-06-05:
@marceloaqno, I have no problems anymore. It works well.

kisse66, 2017-08-07:
Anything new on this?   I just noticed something similar after updating to latest development git yesterday.
RPi running nrf24 GW just stopped receiving anything after about 10h. GW connectivity still ok. Restart and it worked. Let's see if it happens again.
Before the update it was running without issues. One time does not mean much, but someone else still seeing this?
Unfortunately I did not write down the previous commit.

./configure --my-radio=nrf24 --my-rf24-irq-pin=15 --my-leds-err-pin=12 --my-leds-rx-pin=16 --my-leds-tx-pin=18

marceloaqno, 2017-08-09:
I'm still having this problem with a RPi3 I use as a gateway, I couldn't identify the root of the problem, I was suspecting that it could be my power supply or the long wires I'm using to connect the nrf24 module.
For now, I'm using this workaround:
https://github.com/marceloaqno/MySensors/commit/83767b2a962791f3ead073cb4505f7f95ecbe67c

In which it prints to the system logs messages such as:
`Aug 08 20:36:18 rpi3 mysgw[20666]: BAD INTERRUPT TRIGGER`
when the problem occurs and the gateway continues to receive messages.

@kisse66 How is your nrf24 module connected to the RPi? Did you try using another power supply?

kisse66, 2017-08-09:
Thanks, if it crashes again I'll try. Was fine yesterday and today.

I have PA+LNA version (cheap unshielded one) mounted on a small PCB plugged right into RPi connector. Wires soldered and quite short. Power comes from 1117 linear regulator (5V->3.3) with several caps.
This problem appeared after updating SW. Before that it ran fine for months. Doesn't proove anything, but kind of points more towards SW, or new SW just brings a HW issue more visible. OR it was just one incident...
RF interference could be one thing as I observed with RFM69H, that corrupted RPI serial data before adding more caps and shielding.
I'll hook up my scope and look a bit how the power feed looks like, currently I have max RF power.

ace-at-git, 2017-08-17:
Hi, I can confirm the occurrence of "BAD INTERRUPT TRIGGER".

I have a RaspberryPi zero w. I attached several NRF24 modules to it, in the latest hardware I use also a PA-LNA version as kisse66 describes. It has now a ready made adapter board with AMS1117 Step-Down and various capacitors.

All the modules experienced a loss of communication (some hours after start) and a restart of the service helped. After all the experiment with hardware I decided to dig into the software issue descriptions...

I use git development commit 463358209da0f2bcdfe62e5a9c7f683363e21a1e
(13. August) and I added marceloaqno's workaround and observed one "BAD INTERRUPT TRIGGER" after 4h.

I am not sure if we see a problem with crappy hardware or if there is a software issue in the interrupt handling. I have no clue what goes wrong...

I now think about the possibility to use a small capacitor to smoothen the IRQ line a bit - what do you think?

Anyhow I can beta test patches on that issue I some programmer has one ;-)

kisse66, 2017-08-17:
A small capacitor to filter glitches out was what I used for the RFM69 & serial port issue earlier. I used 10nF to ground for RX as close to the connector on RPi as possible. Checked the signal with scope and edges looked perfect (i.e. not too large cap to delay it too much). No issue seen after that, but I also shielded it a bit.
My nrf24 GW has now been running fine, so looks like single incident.

kisse66, 2017-08-30:
worked for a while but it did it again just now. I'll add the patch above  to see next time is it the same thing.

ace-at-git, 2017-08-31:
I experimented with some capacitors values around 10nF. The number of BAD INTERRUPTS varied in all cases. At least one occurrence remained in about 3 hours...

So I decided to organize a Oscilloscope, assuming that I have very bad interrupt signal quality - Surprisingly the negative interrupt pulse looked nicely and sharp, even when I removed the best working capacitor.

I see typically interrupt pulses of about 200us to 300us. The time from high to low is about 0.6us, the time from low back to high is about 0.4us. I did not observe interrupt pulses smaller than 200us, but when I started to look for longer interrupt pulses I measured a pulse above 1400us (1.4ms). In the log was one BAD INTERRUPT. I can't tell if the long pulse caused the BAD INTERRUPT log - but I would bet...

I code points out that an execution of several 100us is expected, but factor 14 is more than "several" - I think. So maybe it is a normal situation that extreme long interrupts occur and therefor possibly trigger it twice...
Anyhow I start to think that the patch needs to go into the regular release to assure that the gateway continues to work after a seldom event...

kisse66, 2017-09-01:
I also agree, GW should not lock up on that.
One thing I've learned about scopes is that things are not always exactly as they seem on the screen. Scope will filter fast things, modify waveforms and the ground lead especially near RF signals works as antenna adding some extra (which many filter away by limiting b/w, which then hides more).

Do you have the radio(antenna) very close to RPi? I have.  My issue with RFM69 and serial port disappeared after adding the cap, but I also moved the antenna a bit further away from RPi and added shield (grounded aluminum tape) between. It's not impossible that RF causes interference somewhere else on the RPi board.

kisse66, 2017-09-01:
One thing about the long interrupt pulse - Linux is not real-time system. It's not unusual to get long responses here and there. Haven't looked at how the driver works (ack by userspace task ?), but it may not be unusual.

ace-at-git, 2017-09-01:
You are right not blindly trust measurements ;-)

Thank you for the point with the HF antenna. I use a RaspberryPi zero w - it has a WLAN antenna on the board (printed structure). This structure is about 5cm away from the NRF24 board and its antenna. I wrapped the NRF24 in aluminum foil when I was "dinging in the dark". I was already wondering about the coexistence of both radio techniques in the same band (well I choose different channels). Maybe I experiment with my hardware build a bit to avoid this HF "voodoo".

What can we do to get the patch officially in the release since it makes the system more robust and also brings hints about the existence of potential HW problems?

If the NRF24 keeps the interrupt line low until it is cleared by the software long pulses are explainable (last night I measured up to 3ms!) - I have not studied the NRF24 data sheets yet, nor have I gone into the software depth - still adjudicated guesses.

manutremo, 2017-10-25:
I've been having a similar issue with my gw with a nodemcu and rfm69hw where the gw stopped receiving any communications sometimes after some hours, sometimes after 2-3 weeks. I've been blaming the nodemcu and I've been trying all sorts of sw and hw solutions with no success, but I never thought on the possibility of a noisy interrupt line. I have now fitted a 10nF cap between int and gnd and have also shielded the rfm69hw with plastic and then aluminium foil. For the moment it seems it's working well, will report back if the problem reappears.

ghvader, 2017-12-24:
I was running 2.0 for a long time without issue. recently changed my HA platform and combined my gateway with it. i'm using 2.2.0 rc2 with interrupt (i was not using irq in 2.0) and now i'm seeing this were it stops responding. if i service mysgw restart it works again.

PR #1094 (2018-03-27): Linux: RF24 interrupt trigger workaround
Fix #870
commit b5a2fc22 (2018-03-27): Linux: RF24 interrupt trigger workaround (#1094)

Fix #870
closed 2018-03-27 by mfalkvidd